dresden-elektronik/deconz-rest-plugin #1718: Raspbee: BCM 18 (PWM0)

flobernd, 2019-07-30:
Hi there,

it seems like the Raspbee is using pin BCM 18 (PWM0) for a reset functionality. I've tried to stack the Rasbpee on top of another module - which failed because of the Raspbee using the above port.

As I usually don't need the reset functionality, I'm wondering if it is possible to disable usage of the pin in some kind of config file?

Best regards

manup, 2019-08-01:
The pin is not used in normal operation. Only by GCFFlasher and here only if no recent firmware is installed. With the current deCONZ firmware the reset is issued via serial command.

However the show stopper is likely that the reset pin is hardware to the micro controller and not handled by the controller firmware at all (works like the reset button on a PC).

So you would need to physically "disconnect" the pin from the RaspBee.

flobernd, 2019-08-03:
Thanks for your response :)

closed 2019-12-08 by stale[bot]