dresden-elektronik/deconz-rest-plugin #1719: Unable to connect raspbee

AshokkumarDeva-Emis, 2019-07-31:
Hi, 
I am using raspberry pi 3 B+ and raspbee premium. I have flashed Raspbian Stretch Desktop(2.05.63) SD card image and tried to boot Raspberry pi.
On booting, deconz app is getting launched and i could see "Raspbee" under connect to device option but the connect button is disabled and i am unable to connect.  Also i could see join network status also as "Not connected"

The LED light in raspbee glows as red which confirms that zigbee firmware is connected as per the document. In network settings i see everything is empty. Is there something i should do  specific after flashing sdcard image? 

I tried to use  HassoS and deconz addon, but still i get firmware not connected and channel as 0 in phoscon app

Can you please help me in sorting out this issue. Also, I have attached the images for reference

[image]
[image]

manup, 2019-08-08:
Can you please check with the official docs of installation on Raspbian.
Here the step 1) should be verified to check that the serial interface is configured properly:

https://phoscon.de/en/raspbee/install#raspbian

If it still shows not connected the system needs to be checked that not multiple deCONZ processes are running which are trying to access the serial interface:

    ps ax | grep deconz

Since the LED is steady red I assume a firmware is already running. So missing firmware is likely not the problem here, but for reference to update the firmware manually the following guide can be used:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

AshokkumarDeva-Emis, 2019-08-08:
Thanks for the response @manup 
Yes I tried installing Raspbian, deconz and configured serial interface properly as required. Also, I tried with SD-Card image from
https://phoscon.de/en/raspbee/sdcard
I face the same problem

Only one process of deconz is running, kindly refer the attachment
[image]

When i tried to update the firmware, i get "Could not sync with bootloader" error

The issue which I am  facing now has been already reported by another user under below ticket,
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1473

I tried everything that you suggested in the above ticket and i am getting the same output as that user
Please let me know if there is something else i can try

manup, 2019-08-08:
Ok that brings more insights. Notes on the screenshot, I see two problems:
* `GCFFlasher_internal` command needs to be used instead of `GCFFlasher`
* deCONZ must be closed before updating the firmware

To update the firmware manually please use the exact commands and steps as described here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

AshokkumarDeva-Emis, 2019-08-08:
Yes, I tried closing deCONZ and I tried to update the firmware but no luck

Also, I tired sudo GCFFlasher_internal -r and the LED did not blink and it was always red
[image]

manup, 2019-08-08:
Strange this needs to work, running out of options, two more things to check:
* Which power-supply are you using, how many Ampere does it provide?
* You may check if the RaspBee module gets rather hot, this would indicate that it was wrongly connected to the header at some time. See: https://phoscon.de/en/raspbee/install#connection
I hope this is not the case, since this would have damaged the RaspBee completely.

AshokkumarDeva-Emis, 2019-08-08:
Ahh, it is 2.4A and raspbee does not get hot when connected 
[image]

Does 0.1 A influence this ?
Anyhow I will get 2.5A power supply to it and update you on the same

manup, 2019-08-08:
>Does 0.1 A influence this ?

Unlikely, but we have seen many strange things with power-supplies, mostly we recommend the official power adapter.

AshokkumarDeva-Emis, 2019-08-12:
@manup - Any update on this? As you have suggested in the other thread, I am planning to go for a new raspbee

manup, 2019-08-16:
> @manup - Any update on this? As you have suggested in the other thread, I am planning to go for a new raspbee

Since the configuration and power-supply looks valid, I can only conclude that the device is broken. But as mentioned above, this would normally result in the RaspBee getting rather hot and can only be caused by when the module is plugged on the wrong pins resulting in a short-circuit.

Do you remember if the RaspBee package had a proper security patch? We had rare cases where on Amazon damaged RaspBee's were returned and shipped again.

AshokkumarDeva-Emis, 2019-08-16:
Thanks for this @manup 
I am not sure about the package received from Amazon, since I had received this from my UK counterpart ordered from Amazon. I will check with a new device and update you on the same thread

lhoracek, 2019-12-19:
@AshokkumarDeva-Emis  any news on this one .. looks like I'm facing similar issue

AshokkumarDeva-Emis, 2019-12-19:
> @AshokkumarDeva-Emis any news on this one .. looks like I'm facing similar issue
This happened due to the damaged raspbee. Newly purchased raspbee is working fine for me

closed 2019-12-19 by AshokkumarDeva-Emis

henkkakay, 2021-01-23:
I experience a very similar issue that almost felt random - upon boot-up of the Rasp-Pi, the RaspBee is occasionally connected, occasionally not. I _may_ have spotted the pattern here. The symptoms point to CPU load during boot-up.

System info:
- RaspPi 2B v1.1 (revision a01041 from cpuinfo)
- RaspBee (not the RaspBee II) current firmware 26390500 (similar may have been experienced in other/older firmwares as well)
- power supply: Internet router's USB-C port (meant to be max 3Amp at 5V) connected via a USB-C to micro-B cable
- Setup loads up X11, which I can VNC into
- System loads deCONZ on boot

Observations:
1. When I am working on changes to the system, I often want to login to system asap after reboots. So I load vncviewer + connect maybe 20 seconds after power-up/boot-up. VNC successfully connects because X11/vncserver are both up. Eventually deCONZ loads but fails to find/connect to Raspbee
2. Once in a while I decides to wait a bit after reboot (grab a coffee, do something else), and only connects to VNC few minutes later (I wait until the heavy blinking of the green LEDs on the Pi stops), deCONZ seem to have a higher chance of successfully connected to raspbee with all the Zigbee connections up (it is not 100% still, but it is enough that I can feel the improvement compared to the first case)

From this - it _appears_ the CPU load (or memory use?) associated with loading up VNC has an impact on initialisation of the Raspbee. (recognising that my Pi2B is probably a bit low on computing power but it really works quite ok 99% of the time). I wonder if there _could_ be something that can be improved on the initialisation code.