dresden-elektronik/deconz-rest-plugin #1725: Battery values virtually recharged up to 100% (after resetting gateway)

enr0c, 2019-08-01:
Hello, 
i was wondering how the battery status for an Xiaomi Multisensor is being retrieved or how its being influenced?
The following two Rest Responses are from March 2019 (Battery 91%) and August 2019 (100%) for the same device. I did not change any battery in the device. The only difference is the swversion attribute which is now being send by the API.

- March 2019 - I dont know which Firmware was installed back than
- Firmware as of now, installed in May 2019:
```
'fwversion': '0x26330500',
 'swupdate2': {'autoinstall': {'on': False, 'updatetime': ''},
  'bridge': {'lastinstall': '2019-05-24T12:27:10',
   'state': 'allreadytoinstall'}
```

- REST Response - Battery = 100% **TODAY**

 ```{'config': {'battery': 100, 'on': True, 'reachable': True},
 'manufacturername': 'LUMI',
 'modelid': 'lumi.weather',
 'name': 'Temperature Sensor Aqara',
 'state': {'lastupdated': '2019-08-01T16:51:52', 'pressure': 997},
 'swversion': '20161129',
 'type': 'ZHAPressure'}
```
- REST Response - Battery = 100% **March 2019**
```{'config': {'battery': 91, 'on': True, 'reachable': True},
  'manufacturername': 'LUMI',
  'modelid': 'lumi.weather',
  'name': 'Temperature Sensor Aqara',
  'state': {'lastupdated': '2019-03-10T22:00:10', 'pressure': 998},
  'type': 'ZHAPressure'}
```

Thanks:)

ebaauw, 2019-08-01:
> i was wondering how the battery status for an Xiaomi Multisensor is being retrieved

Not.  Xiaomi sensors don’t support querying the battery.

The sensor sends an hourly special report that includes the battery voltage.  After starting deCONZ it might take a couple of hours, before the default value of 100% is updated.

The REST API plugin translates the battery voltage to a percentage.  Technically, this doesn’t reflect the battery capacity - button cell batteries report a high / normal voltage for a very long time, and then voltage drops quickly when the battery is almost depleted.  It’s not uncommon for sensors to report 100% for over a year of usage.

enr0c, 2019-08-01:
Thank you for the explanation, helped a lot!

> The sensor sends an hourly special report that includes the battery voltage. After starting deCONZ it might take a couple of hours, before the default value of 100% is updated.

I will wait for a couple of hours and close the issue!

closed 2019-12-06 by stale[bot]