dresden-elektronik/deconz-rest-plugin #1727: Not showing the Phoscon App -> Menu -> Settings -> Gateway

Zjaaspoer, 2019-08-03:
As described in the title, it's not showing the page. As in: A blank page, with no content. The Conbee 2 stick seems to be working fine for the rest. In the deConz app I can inspect the whole network which is operating fine:

[image]

The Conbee 2 stick is running the latest firmware, updated manually just now to be sure:

[image]

Only when I go to Menu -> Settings -> Gateway I get a grey screen:

[image]

I want to change the channel, so any idea what can be causing this issue?

TilBumble, 2019-08-04:
Seems to be the problem discussed in #1630 .
A workaround (deleting the "2" after "...-gateway" in the browser address-field) is described in #1572 .

My system (Microsoft Edge on Windows 10 1809) is affected too. (Gateway-page completely empty except for frames like in Your screenshot)

I am waiting patiently for 2.05.67.
Learning about the workaround was very helpful and is ok for the moment for me.

closed 2019-12-09 by stale[bot]

Ergerbolt, 2019-12-23:
Hi, ive got the same problem. Gateway is not showing up. Anything else is working. Im using a RPi 4.

Smanar, 2019-12-23:
Hello,
Have you tried with another browser ? (To check cache problem)

markus-fischbacher, 2020-02-01:
Still not fixed with version 2.05.72 / 12/12/2019 in Home Assistant deConz Add-on

Krocko, 2020-02-17:
I have the same problem with the safari browser. The chrome browser is working. 
Is there a fix already available?

chadderksen, 2020-03-03:
Same here, safari on iOS ans MacOS is not working...

MickL, 2020-04-13:
Same here. Why is this issue closed? :(

markus-fischbacher, 2020-04-13:
It‘s really annoying.

enno42, 2020-04-27:
same issue.. tried with different browsers

bramschats, 2020-05-16:
Had the same problem with Safari but Chrome worked as should

crixyd, 2020-06-17:
Same problem occurs for me in Safari, however Chrome is fine

Raniz85, 2020-06-27:
I had the same problem in Firefox but fixed it by clearing all data for the phoscon site.

In Firefox:
1. Click the lock next to the URL in the address bar
2. Click "Clear Cookies and Site Data"

arevinje, 2020-06-30:
Same fix for me; flushed cache in Chrome and it's now back in business.

PointingP, 2020-12-16:
Having same issue. Emptying cache does not help. Tried 3 different browsers on macOS and 2 on iOS, it's always empty.

Smanar, 2020-12-16:
Can you check in deconz GUI if all seem fine ?
Can you take a look on the browser on the "developper console" ?

PointingP, 2020-12-17:
Can't seem to open deconz GUI with VNC to RPi - nothing happens when trying to open. I think it did work at some point, but I don't use it.

Data from developer console:

group.js?fd8dfa:3 Uncaught SyntaxError: Unexpected token function
version-compare.js?b3947f:1 Uncaught SyntaxError: Unexpected token ?
moment.js?69ab16:1 Uncaught SyntaxError: Invalid or unexpected token
moment-timezone-with-data.js?e60811:1 Moment Timezone requires Moment.js. See https://momentjs.com/timezone/docs/#/use-it/browser/s @ moment-timezone-with-data.js?e60811:1(anonymous function) @ moment-timezone-with-data.js?e60811:1(anonymous function) @ moment-timezone-with-data.js?e60811:1(anonymous function) @ moment-timezone-with-data.js?e60811:1
moment-timezone-with-data.js?e60811:1 Uncaught TypeError: Cannot read property 'version' of undefined(…)(anonymous function) @ moment-timezone-with-data.js?e60811:1(anonymous function) @ moment-timezone-with-data.js?e60811:1(anonymous function) @ moment-timezone-with-data.js?e60811:1
xstate.js?218fe0:1 Uncaught SyntaxError: Unexpected number
xstate.interpreter.js?567382:1 Uncaught SyntaxError: missing ) after argument list
languages.js?9559e1:2257 Object {txt_close: "closed", text_selectBlind: "Select blind", btn_newBlind: "Add new blind", text_blindEmptyInfo: "No blinds were connected yet."}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {txt_change_pw_login_success: "The password was changed successfully.", txt_zigbee_channel_fail: "The channel could not be changed.", txt_zigbee_channel_success: "The channel change was successfull.", txt_zigbee_channel_change: "Changing the gateway Zigbee channel"..."", homebridge_pin: "Homebridge pin"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {sel_steps: "Increment", sel_select: "select", sel_cros: "Crossing", sel_sat: "Saturation", sel_step_num: "number of levels"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {help: "help", link_manuals: "Manuals", link_onlinehelp: "Online help", label_documentation: "Documentation", label_contact: "Contact"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {txt_noScenes: "No scenes created", txt_addScene: "Create scenes in scene editor", txt_noLights: "No lights assigned", txt_addLight: "Assign lights via", btn_color: "Color"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {account_settings: "Account manager", btn_new_Account: "New account", text_allOnOff: "All lights on/off", text_control: "Control", text_manage: "Manage"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {nav_btn_LinkSwitch: "Link switch", nav_head_switchControl: "Switch control of the group", nav_dialog_head: "Key assignment changed", btn_new_Group: "Add group", btn_all_Group: "All off"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {text_clientNotConnectedInfo: "Currently not connected", text_new_conf_client: "Connect WiFi", text_enterPassword: "Enter password", text_DeactivateNow: "Deactivating Access Point. <br> This takes a few seconds.", text_DeactivateClientNow: "Deactivating the WiFi connection. <br> This takes a few seconds."…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {func_onOffOptions: "On / off", func_callScene: "Call scene", func_callSensor: "Call sensor", func_effect: "Effect", func_dimming: "Change brightness"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {create_sensor_control_info: "Create automatic light controls with motion and brightness sensors.", h_delete_sensor_control: "Delete sensor control", q_delete_sensor_control: "Delete sensor control <b>{1}</b>?", text_rename_sensor_control: "Rename sensor control", h_pir_settings: "Settings"…}
languages.js?9559e1:2266 undefined
languages.js?9559e1:2257 Object {create_switch_control_info: "Link switch to group to assign keys freely", txt_noSwitch: "No switches linked to the group yet.", txt_scene_delete_info: "was deleted", txt_push: "slide", txt_drop: "drop"…}
languages.js?9559e1:2266 undefined

I'm coding illiterate, so unfortunately that doesn't say anything to me.

the Weird thing is, that the Gateway page did load yesterday when checking on the phone - once - now it won't work again.