dresden-elektronik/deconz-rest-plugin #1728: Alexa Discovery

jonb81, 2019-08-03:
I am having a persistent issue discovering deconz devices from alexa.

In desperation I have disconnected everything from the network except:
Conbee stick with Raspberry Pi 3 Model B Rev 1.2 running Raspian Buster OS
Phoscon Version 2.05.65 (firmware 26330500) on port 80
Amazon Echo Dot v2
(and my phone .... sometimes)

The pi is connected directly to my isp provided router via ethernet and the echo dot  is connected  to the router wifi.  Additionally I have tried to complete discovery using the alexa.amazon website from the pi.

I have only 1 sensor connected to the conbee at the minute and that is a Xiaomi Aura Temperature Humidity Sensor since I reset everything.

I have looked at #759 and a few others but still no joy.

I can see the unpn advertisement over the network  (http://192.168.1.7/description.xml in my case) and I have checked the xml file to make sure the manufacturer has been updated to Royal Philips Electronics.

From a tcp dump on the pi during the discovery process I can see what appears to be the devices talking to each other which is encouraging (GET request with a 200 response)?

14:19:59.015295 IP amazon-**sensored**.50000 > 239.255.255.250.1900: UDP, length 107
14:19:59.122927 IP amazon-**sensored**.40667 > raspberrypi.http: Flags [P.], seq 2211542329:2211542394, ack 1284411059, win 1369, options [nop,nop,TS val 60317550 ecr 1774811257], length 65: HTTP: GET /description.xml HTTP/1.1
14:19:59.123006 IP amazon-**sensored**.40668 > raspberrypi.http: Flags [P.], seq 4191560008:4191560073, ack 1930741251, win 1369, options [nop,nop,TS val 60317550 ecr 1774811257], length 65: HTTP: GET /description.xml HTTP/1.1
14:19:59.123747 IP raspberrypi.http > amazon-**sensored**.40667: Flags [P.], seq 1:1184, ack 65, win 227, options [nop,nop,TS val 1774890094 ecr 60317550], length 1183: HTTP: HTTP/1.1 200 OK
14:19:59.124174 IP raspberrypi.http > amazon-**sensored**.40668: Flags [P.], seq 1:1184, ack 65, win 227, options [nop,nop,TS val 1774890095 ecr 60317550], length 1183: HTTP: HTTP/1.1 200 OK
14:19:59.126312 IP amazon-**sensored**.40667 > raspberrypi.http: Flags [.], ack 1184, win 1406, options [nop,nop,TS val 60317550 ecr 1774890094], length 0
14:19:59.126513 IP amazon-**sensored**.40668 > raspberrypi.http: Flags [.], ack 1184, win 1406, options [nop,nop,TS val 60317550 ecr 1774890095], length 0
14:19:59.126531 IP amazon-**sensored**.40667 > raspberrypi.http: Flags [F.], seq 65, ack 1184, win 1406, options [nop,nop,TS val 60317550 ecr 1774890094], length 0
14:19:59.126694 IP raspberrypi.http > amazon-**sensored**.40667: Flags [F.], seq 1184, ack 66, win 227, options [nop,nop,TS val 1774890097 ecr 60317550], length 0
14:19:59.129076 IP amazon-**sensored**.40667 > raspberrypi.http: Flags [.], ack 1185, win 1406, options [nop,nop,TS val 60317550 ecr 1774890097], length 0
14:19:59.140760 IP amazon-**sensored**.40668 > raspberrypi.http: Flags [F.], seq 65, ack 1184, win 1406, options [nop,nop,TS val 60317551 ecr 1774890095], length 0
14:19:59.141050 IP raspberrypi.http > amazon-**sensored**.40668: Flags [F.], seq 1184, ack 66, win 227, options [nop,nop,TS val 1774890112 ecr 60317551], length 0
14:19:59.143284 IP amazon-**sensored**.40668 > raspberrypi.http: Flags [.], ack 1185, win 1406, options [nop,nop,TS val 60317552 ecr 1774890112], length 0
14:19:59.192716 ARP, Request who-has 192.168.1.51 tell amazon-**sensored**, length 46
14:19:59.194585 ARP, Request who-has 192.168.1.28 tell amazon-**sensored**, length 46
14:19:59.195013 IP raspberrypi.35161 > SkyRouter.Home.domain: 14598+ PTR? 28.1.168.192.in-addr.arpa. (43)
14:19:59.207517 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [S], seq 3205127706, win 65535, options [mss 1460,sackOK,TS val 60317558 ecr 0,nop,wscale 6], length 0
14:19:59.207787 IP raspberrypi.http > amazon-**sensored**.40684: Flags [S.], seq 2251246337, ack 3205127707, win 28960, options [mss 1460,sackOK,TS val 1774890178 ecr 60317558,nop,wscale 7], length 0
14:19:59.208124 IP SkyRouter.Home.domain > raspberrypi.35161: 14598 NXDomain* 0/1/0 (102)
14:19:59.209997 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [.], ack 1, win 1369, options [nop,nop,TS val 60317558 ecr 1774890178], length 0
14:19:59.210436 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [P.], seq 1:72, ack 1, win 1369, options [nop,nop,TS val 60317558 ecr 1774890178], length 71: HTTP: GET /api/**sensored**/lights HTTP/1.1
14:19:59.210545 IP raspberrypi.http > amazon-**sensored**.40684: Flags [.], ack 72, win 227, options [nop,nop,TS val 1774890181 ecr 60317558], length 0
14:19:59.212510 IP raspberrypi.http > amazon-**sensored**.40684: Flags [P.], seq 1:160, ack 72, win 227, options [nop,nop,TS val 1774890183 ecr 60317558], length 159: HTTP: HTTP/1.1 200 OK
14:19:59.214861 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [.], ack 160, win 1369, options [nop,nop,TS val 60317559 ecr 1774890183], length 0
14:19:59.216522 ARP, Request who-has raspberrypi tell SkyRouter.Home, length 46
14:19:59.216705 ARP, Reply raspberrypi is-at **sensored** (oui Unknown), length 28
14:19:59.252334 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [P.], seq 72:143, ack 160, win 1369, options [nop,nop,TS val 60317563 ecr 1774890183], length 71: HTTP: GET /api/**sensored**/lights HTTP/1.1
14:19:59.253680 IP raspberrypi.http > amazon-**sensored**.40684: Flags [P.], seq 160:319, ack 143, win 227, options [nop,nop,TS val 1774890224 ecr 60317563], length 159: HTTP: HTTP/1.1 200 OK
14:19:59.258493 IP amazon-**sensored**.40684 > raspberrypi.http: Flags [.], ack 319, win 1369, options [nop,nop,TS val 60317563 ecr 1774890224], length 0

Does anybody have any ideas?

jonb81, 2019-08-04:
I've had another fiddle today but still no joy.  I have dug a little deeper into the tcp dump and I may be off track with this because I'm not entirely confident with what I am looking at really and iv'e nothing to compare it with but it does look as though an api key is given with a POST request during discovery which is then used to probe  /api/**randomapikey**/lights with a GET request.

I can open this as a web page during discovery and its empty because I have no lights.  I can manually open /api/**randomapikey**/sensors from a web browser and that shows the sensor information but I can't see that this is being called during the discovery as it does not show up in the tcp dump.

I'm a bit confused why during discovery /api/**randomapikey**/lights would be repeatedly probed but then never progress to /api/**randomapikey**/sensors??  Please can someone let me know if the alexa integration is just for lights and that is why I'm not finding any sensors or whether I need at least one light connected. Thanks

KodeCR, 2019-08-09:
The native build-in Hue support of Alexa is very basic and does only lights, and not even colour lights I believe. So yes, it is expected to not find any sensors.

The way around is to either develop a deCONZ alexa skill yourself or use homebridge-hue in combination with homebridge-alexa (which I use and works very well).

closed 2019-12-14 by stale[bot]