dresden-elektronik/deconz-rest-plugin #1729: Xiaomi Aqara Multisensor

dhartmann84, 2019-08-03:
Recently I own 5 pieces of the Xiaomi Multisensor. (Humidity, temperature, pressure)
If I want to add this now in the Phoscom app, this only goes up to the second sensor from the third is simply overwritten again one of the already configured.

Is there a limit to the number of units? What can I do that the sensors are detected individually?
The incorporation of Hue lamps works perfectly.

Thank you for your help.

System:
Raspberry Pi 4 with Conbee II USB Stick
Raspbian Buster 10 (Kernel 4.19.58-V71+)
Deconz 2.05.65 Buster Stable

Smanar, 2019-08-04:
No there is no limit (there is one in reality, but network depend).
Deconz create 1 node by MAC adress, even the adress network change. I don't think it's possible having 2 sensors with the same mac adress.

Can you take a look in same time in deconz (if you have it with GUI) ?

dhartmann84, 2019-08-04:
Thank you.  
My knowledge of linux is not very good. 
What I already tried: 
My system is without a GUI so I tried to start the GUI via Xming.  This worked too.
But if I select the stick and click on "connect", nothing happens. I've come that far.

Maybe I'll do something wrong at this point.

Smanar, 2019-08-04:
You can have only 1 deconz running, the headless or with the GUI, so it is more complicated to switch.
But if you haven't too much devices, you can take a look too in the api with something like

`http://192.168.1.1:80/api/36E138295E/sensors`

With your ip/port/api_key

This command will list all your sensors, you can check if there is something strange during yours manipulations, like if a device is realy overwritted by another. All sensors have 1 uniqueid, it's the mac adress.

dhartmann84, 2019-08-05:
> You can have only 1 deconz running, the headless or with the GUI, so it is more complicated to switch.
> But if you haven't too much devices, you can take a look too in the api with something like [...]

I could check the data about api. They are all different MAC addresses.

Every time I connected a new sensor, I checked the address.
Now everything has worked. Why can not I explain at all.

 - Does it matter how fast the sensors are trained one after the other?
- Is it possible that the sensors were too close to each other when learning?

I have no idea why it worked now.

Thanks for your help!

closed 2019-08-05 by dhartmann84