dresden-elektronik/deconz-rest-plugin #1730: Eria Adurosmart Wireless Dimming Switch - Request Support

jbkuma, 2019-08-04:
[image]
[image]

The Eria wireless dimmer has 4 buttons similar to the Hue wireless dimmer, but it's smaller and fits in a decora style wall plate.

I've tried various methods of pairing it, but I'm not getting any thing from the API  when press the buttons.

skluthe, 2019-10-19:
I would very much like this. How were you able to pair it? Mine wont connect even in paring mode.

jbkuma, 2019-10-21:
> I would very much like this. How were you able to pair it? Mine wont connect even in paring mode.

There are two ways to pair it: press and hold the top button, press and hold the bottom one.  I can't remember which way I got it to pair.  I either used generic or Ikea as the profile.

skluthe, 2019-10-21:
Wasn't able to get it to pair with Deconz but it paired with home assistant
no problem (holding the bottom button).

mdolnik, 2020-10-10:
> I would very much like this. How were you able to pair it? Mine wont connect even in paring mode.

I was able to get this paired in deCONZ via phoscon, but it seems very hit and miss. I am fairly certain I chose `Other` in the `Add new switch` window and holding either the down or up button for 5 or more seconds.

This doesn't always work, I would say 1 out of 20 tries this will work for seemingly random reasons, and most times I would try to "factory reset" the remote by holding both the up and down buttons at the same time until the green LED flashes on the remote.

Once it is "paired":

- It does not get successfully added to Phoscon, it only shows up in deCONZ. Phoscon doesn't register that anything was ever found.
- Most of the remote's values in deCONZ matches the screenshots above, but I also have:
    - `Model Identifier` : Adurolight_NCC
    - `Date Code` : `20180402`
- Every button press will flash the green LED on the remote. When it's not paired, the LED does not light up on every button press.
- When deCONZ is in debugging mode I get the following output on any button press:

```
05:37:48:426 MAC Poll 0x02 0x4B71
05:37:48:426 verify 0x00158d000198d590 is child node after 809128 s
05:37:48:436 verify neighbor status: APP_SUCCESS (0x00)
05:37:48:436 MAC Poll 0x02 0x4B71
05:37:48:441 Node 0x00158D000198D590 is known by 1 neighbors, last seen 0 s
05:37:48:507 MAC Poll 0x02 0x4B71
05:37:48:515 MAC Poll 0x02 0x4B71
05:37:48:522 MAC Poll 0x02 0x4B71
05:37:48:532 APS-DATA.indication from child 0x4B71
05:37:48:679 MAC Poll 0x02 0x4B71
05:37:48:688 MAC Poll 0x02 0x4B71
05:37:48:750 MAC Poll 0x02 0x4B71
05:37:48:752 MAC Poll 0x02 0x4B71
05:37:48:755 MAC Poll 0x02 0x4B71
05:37:48:766 poll node ec:1b:bd:ff:fe:7f:58:ee-01-0b04
05:37:48:766 Poll ZHAPower sensor node Power 3
05:37:48:767 APS-DATA.indication from child 0x4B71
05:37:48:770 Idle timer triggered
05:37:48:770 binding for attribute reporting of ep: 0x01 cluster 0x0B04 seems to be active
05:37:48:770 Force binding of attribute reporting for node Power 5
05:37:48:920 Node 0xEC1BBDFFFE7F58EE is known by 2 neighbors, last seen 2 s
05:37:49:028 Poll APS request to 0xEC1BBDFFFE7F58EE cluster: 0x0B04 dropped, values are fresh enough
```

`0x4B71` being the ID of the remote in deCONZ.

Since it does not get picked up in Phoscon I am unable to use it for anything.

mdolnik, 2020-10-10:
> Wasn't able to get it to pair with Deconz but it paired with home assistant no problem (holding the bottom button).

How did you pair it in home assistant without using deCONZ?

mdolnik, 2020-10-10:
Is it possible to re-open this issue?

What can I try changing in the source code to get this working in deCONZ/Phoscon?

Mimiix, 2020-10-12:
@SwoopX Can you put this on top of your list when your other PR's are resolved?

SwoopX, 2020-10-13:
No, because that list is long.

skank01, 2020-11-04:
Interested in this too. !

mdolnik, 2020-11-29:
Any updates on this?

evanhemmen, 2021-01-02:
Hi @Mimiix, a friendly reminder (as suggested on GoT): Could you please check the status of this device? Thanks!

Mimiix, 2021-01-02:
@SwoopX how's the list?

evanhemmen, 2021-01-07:
Hi @Mimiix & @SwoopX, any news on this dimmer? I'm curious to find out of it is supported. If it would help I could buy a unit and provide you with the required info, but I think that has already been done by the initial requester.

Thanks in advance!

Mimiix, 2021-01-07:
Just spoke with Swoop. He is still waiting on some PR's to get merged. I'll try to push Manuel to give those priority. If those happen in the next few days, he'll check on this one.

evanhemmen, 2021-01-12:
@SwoopX & @Mimiix I just received the dimmer I bought and it is working out-of-the-box! Is there anything I could do for you, is there information missing that I could deliver, or can this issue be closed?

mdolnik, 2021-01-12:
@evanhemmen can you speak to [my previous comment above](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1730#issuecomment-706621272) in regards to:

- What buttons on the switch are pressed to successfully pair?
- What type of switch do you choose in the Phoscon UI?
- Does it show up in both deCONZ AND Phoscon once paired?
- Does EVERYTHING work for you? Are you able to assign button commands in Phoscon, or is it just "pairing"?

Also what version of deCONZ are you running?

Regardless I will see if I have time in the next few days to test with my Eria Adurosmart switch to see if some newer deCONZ release since October happened to fix my issues.

evanhemmen, 2021-01-12:
Oh my god, I'm feeling really really stupid at the moment.... I'm sorry for the confusion, which is al created by me....

I made a mistake. I was looking for support for the Eria Adurosmart Dimmer **plug**, not the Eria Adurosmart Dimmer Switch. I completely missed that this particular issue was about the switch/remote. In my search for support for this plug on the www, I was linked to this issue and did not read it thoroughly. Or at least I did read it, but in my "wishfull thinking mode" I missed that little, but crucial detail.

Again, sorry for the false hopes for you and anyone else waiting for a solution and for asking attention from @SwoopX and @Mimiix for the wrong reason.

mdolnik, 2021-01-12:
> I'm sorry for the confusion

No worries at all, I had a glimmer of hope that maybe some irrelevant update to deCONZ fixed this issue, but it's all good, simple mistake.

I wasn't aware that there are zigbee plugs that are dimmable, TIL.

closed 2021-02-22 by manup

mdolnik, 2021-03-01:
The previous Pull Request did not resolve this issue, refer to new Pull Request #4495

mdolnik, 2021-03-01:
Is it possible to re-open this issue?

mdolnik, 2021-03-13:
Pull Request #4495 has been closed and will not be merged.

Pull Request #4580 replaces #4495

switchboy, 2022-01-15:
It still is not fixed despite Pull Request #4580 being merged a long time ago...