dresden-elektronik/deconz-rest-plugin #1731: CentraLite 3326 Motion Sensors

DrJeff, 2019-08-05:
Having some trouble adding 3326-L Motion Sensors (Iris Motion Gen2) and 3320-L Door Sensors (Iris Gen2). Is the only way to get these working still to follow this guide https://github.com/dresden-elektronik/deconz-rest-plugin/issues/623#issuecomment-437649093 and compile myself? After compiling and adding can I revert to using the Docker version? Or do i have to continue to use the compiled version running on Ubuntu?

dale3h, 2019-08-31:
I am having similar issues as well. Have you found a solution to this yet?

nphil, 2019-10-24:
I would like to know this too, I tried adding these via Phoscon, but they would not pair.

mike3y, 2019-10-25:
Love to know if this gets fixed. I bought the dongle, but just cancelled my order until I hear more.

nphil, 2019-10-25:
Attached screenshot with information
[image]

dbmet, 2019-10-28:
Just curious if there has been any movement on getting these working?

arraylabs, 2019-11-21:
Curios as well. I have a pile of these.

Hackshaq, 2020-01-16:
I am having a similar issue. I have been able to pair these and have them show up in the Wired view but not the web app. How do I fully add them to the mesh?

[image]

rmotapar, 2020-02-20:
Any updates on this issue? I am a newbie trying to move from hubitat to hassio/Deconz setup for my ZigBee devices. I have a bunch of Centralite 3326 motion sensors. Wish I knew about this issue before I started the project.

manup, 2020-02-20:
Interesting, the manufacturer code seems to be different.
Here is mine (working):

[image]

[image]

Can you please also provide a screenshot of the basic cluster after reading the attributes. I'll add the sensor to the list then.

rmotapar, 2020-02-20:
> Interesting, the manufacturer code seems to be different.
> Here is mine (working):
> 
> [image]
> 
> [image]
> 
> Can you please also provide a screenshot of the basic cluster after reading the attributes. I'll add the sensor to the list then.

Thank you for the quick response. Assuming you were asking me, I am not yet sure how to get you the screenshots you asked for but I should be able to setup whatever is necessary. My setup until now involved adding the deconz add-on to hassio which has access to conbee ii. I was almost about to return the Conbee ii but this gives me hope. I will paste the screenshots soon. Thanks again @manup

manup, 2020-02-20:
Thanks, want I'd like to check is the basic cluster "Model Id" attribute of the sensor version with the manufacturer code 0xc2df from above.

dbmet, 2020-02-20:
> Thanks, want I'd like to check is the basic cluster "Model Id" attribute of the sensor version with the manufacturer code 0xc2df from above.

Let see If this works. 
[image]

[image]

rmotapar, 2020-02-21:
@manup I have two different kinds of lowes motion sensors. One is 3326-L and another is iL07_1 which measures himidity as well (this was the latest lowes version). I tried to add two 3326-L and one iL07_1 units and all three don't appear in the deconz UI. Attached some screenshots. Please let me know if you need anything else.

[image]

[image]
[image]

manup, 2020-02-24:
Thanks I'll check the respective code sections if there is something missing for these.

soydeboca, 2020-02-25:
I'm also trying to pair these gen2 contact sensors from Lowe's that I was using flawlessly in Smartthings, but it appears that the pairing process does not complete (the green dot is blinking permanently). 

[image]

rmotapar, 2020-03-02:
@manup @SwoopX Would you please consider adding support for these sensors in the next release if at all possible?

SwoopX, 2020-03-02:
@manup are you already into this? Otherwise, I'd put it on my list ;)

SwoopX, 2020-03-03:
Ok, let me take care of that. Manup has already made some preparations. Can any of you guys please poste a screenshot of the power configuration cluster with the attributes being read?

rmotapar, 2020-03-03:
@SwoopX  Thank you for looking into this. Is this what you were asking?

[image]

SwoopX, 2020-03-03:
Jep, thanks, Wanted to be sure about the battery voltage.

soydeboca, 2020-03-03:
Mine looks like this
[image]

Thanks SwoopX!

Hackshaq, 2020-03-04:
> Mine looks like this
> [image: 2020 03 03_19 59 12 -
> HA__deCONZ_RPI4_(Home_Assistant_-deCONZ)-_VNC_V]
> <https://user-images.githubusercontent.com/27639753/75828014-acd89800-5d89-11ea-8648-212b4a82c3d3.png>
>
> Thanks SwoopX!

@manup <https://github.com/manup> @SwoopX <https://github.com/SwoopX> While
we are talking about these Centralite sensors, can you guys work to
officially support as many of their devices as possible? Now that the
Lowes/Iris platform is dead, ebay is flooded with Centralite stuff, and it
is cheap. Iris had a bunch of useful stuff I would love to add to my Deconz
zigbee, such as the 3460 Smart Button. In the US at least, the Deconz
platform would be a lot more useful if you had more complete support for
this Centralite stuff, just based on the amount of devices available. I
dont want to go get a bunch of these, only to find out they are not working
with Deconz at the moment.

Thanks. David

SwoopX, 2020-03-05:
Well, that's easier said than done. Each and every device may have it's quirks which need to be known and sorted out. With a detailed technical documentation, that would be far easier, as it usually provides the necessary details.

A good example for that is cluster FC0F from the screenshot above. We don't know nothing about it unless we do some reverse engineering and/or sniff the traffic.

DrJeff, 2020-03-09:
@SwoopX What else do ya need more power attributes?

soydeboca, 2020-03-30:
Sorry if I missunderstood, it's supposed to be working for both sensors mentioned on the first post (3326 and 3320)?

I'm using deconz v2.05.75 as an Home Assistant add-on and cannot add my Iris contact door sensors that DrJeff also mentioned on the first post (CentraLite 3320-L). This issue is only about the CentraLite 3326 and I have to open a new one? Thanks

SwoopX, 2020-03-30:
Please open a new request for that. It has not been integrated yet.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

dmike3, 2020-07-07:
Any update on this? Is it working?

dbmet, 2020-07-07:
> Any update on this? Is it working?

Yes they work. I have a couple added to my system