dresden-elektronik/deconz-rest-plugin #1733: Xiaomi WXKG01LM switch/button creates bogus light switch

xbmcnut, 2019-08-06:
I have two WXKG01LM buttons in deCONZ under Hassio and one created a battery level sensor in the add-ons integrations page along with a bogus light switch that responds to button pushes. Light is always on and goes off when button pushed.

[image]

[image]

The other switch/button for the master bedroom works and creates deCONZ events in Hass but does not show any sensors in the add-on configuration page.

Is there a fix for that?

Conbee II 26490700
Hassio add-on version 3.1
deCONZ version 2.05.66

Smanar, 2019-08-13:
Hello, have you tried to delete then and re-pair them ?
It's not normal having 2 differents reactions for same sensor. All the "magic" is done during the pairing process, but it can be defective.

xbmcnut, 2019-08-21:
That worked, thanks.

closed 2019-08-21 by xbmcnut