dresden-elektronik/deconz-rest-plugin #1735: Hue tap is not displayed under deconz-gui and problem with rename coordinator

Nastras, 2019-08-06:
Hello everyone, can anyone tell me if it is normal that the Hue Tap is not displayed in the deconz-gui? I have to test a hue lamp and a hue tap integrated but only the lamp is displayed.

[image]

Is it because the hue tap is not a battery powered device?

Another thing still, if I change the name of the coordinator in the Node Info window this is not saved. After a restart it disappeared again. 

Thanks for the help.

King Regards

Nastra

ebaauw, 2019-08-06:
ZGP switches don’t show in the GUI.

The possibility to change the coordinator name was lost several versions ago, when the database was extended to persist devices.

Nastras, 2019-08-06:
@ebaauw Thank a lot for the answer.

closed 2019-08-06 by Nastras