dresden-elektronik/deconz-rest-plugin #1736: Phoscon App does not connect to ConBee 2 stick

Mikha495, 2019-08-06:
Hi!

have an issue with the ConBee 2 stick. Installed everything according to the manual on my Raspberry Pi with HOOBS as platform (Homebridge). After logging into the Phoscon App, the stick seems to be not recognized. 
Running lsusb in terminal, shows me a Dresden Electronics device connected. 
Can you help, please?

[image]

Mikha495, 2019-08-06:
Have the Conbee 2 directly connected to the Raspberry, as no USB extension cable available at the moment. Can this cause this issue?

Mikha495, 2019-08-07:
> Have the Conbee 2 directly connected to the Raspberry, as no USB extension cable available at the moment. Can this cause this issue?

Have now connected a 1,8m USB extension cable and restarted Homebridge. No result, no change...
Please help!

Smanar, 2019-08-07:
Hello, you have another problem, your conbee is detected as raspbee.
Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1433

Mikha495, 2019-08-07:
Result is:
hoobs@hoobs:/var/lib/homebridge $ sudo GCFFlasher_internal -l
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path | Vendor | Product | Serial | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0 | 0x0000 | 0x0000 | | RaspBee
/dev/ttyACM0 | 0x1CF1 | 0x0030 | DE1995980 | ConBee II

Unfortunately, can't follow the solution in #1433. If this would solve the issue, could you be so kind and describe me the necessary steps to move the device to a different location?

manup, 2019-08-08:
Hi, can you please check the basic troubleshooting for ConBee II on Linux:

http://phoscon.de/en/support#conbee2-connection-issues

Smanar, 2019-08-08:
Hello, why in some case the conbee is see at /dev/ttyACM0 and for some other config as /dev/ttyUSB0 ?
With a simple raspberry ?

manup, 2019-08-08:
>it's an hardware difference between conbee 1 and 2 ?

This.
* The ConBee I has a FTDI USB chip and enumerates as /dev/ttyUSBx.
* The ConBee II has native USB support and enumerates as /dev/ttyACMx.

Mikha495, 2019-08-08:
It's working now. I believe the issue was the missing
sudo gpasswd -a $USER dialout

However, I believe to remember, that I did entered it during the initial installation.

Mikha495, 2019-08-10:
After trying to add a new light, Phoscon again lost the connection to the ConBee II and I'm back, where I've been before. 
Any suggestions?

Mikha495, 2019-08-10:
Now I deleted flashed the ConBee, deleted and reinstalled deconz. Now, Phoscon doesn't find the ConBee during the search sequence at all. 
I'm lost...

Smanar, 2019-08-10:
Restart all checks from the start from @manup link.

> GCFFlasher_internal -l

Unplug/replug the conbee to restart the gateway

Mikha495, 2019-08-10:
Did everything from the troubleshooting page. No results

Smanar, 2019-08-10:
And the result for "GCFFlasher_internal -l" ?

Mikha495, 2019-08-10:
CFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE1995980  | ConBee II

Mikha495, 2019-08-10:
Is it what it should show?

Smanar, 2019-08-11:
Unfortunately yes, she is correctly detected.
Have you tried to force a firmware update with the GCFFlasher_internal  command ?
You config is still like this one (I think yes, if she have suddenly stopped working) ?
https://user-images.githubusercontent.com/1177810/44988558-66835c80-af8b-11e8-9cc1-751f60c2b439.png
I realy don't have more ideas, I think  the next step will be logs, but I can't help you a lot for that.

Mikha495, 2019-08-11:
I've reinstalled deconz as root. No changes. 
I'm using hoobs for Homebridge, so as my understanding the default username is hoobs. Hence I'm putting in 
sudo gpasswd -a hoobs dialout
Is that correct?

Smanar, 2019-08-11:
Yep, you can use 
`whoami`
To be sure having the good user name, then your command (and restart)

You have no error message when trying to update your firmware whith command ?

Mikha495, 2019-08-11:
Have flashed the firmware with 
sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26490700.bin.GCF
and it ran well. Now, Phoscon lets me login and I see a device, which is "not connected". Basically, I'm there, where I've been before. 
Going crazy...
[image]

Mikha495, 2019-08-11:
After several attempts successfully installed deCONZ in debugg mode. 
Now, everything runs, as it should. Will see for how long...

Smanar, 2019-08-11:
By curiosity, you don't have special hardware ? External drive, PI4 with usb 3.0, or other ?

Mikha495, 2019-08-11:
Nope, nothing else connected to the Raspbi

Mikha495, 2019-08-13:
After taking all my network devices incl. Raspberry from power, the ConBee II again can’t be connected. 
Same issue as before... 
anyone an idea what is going wrong with my setup???

jose-d, 2019-08-22:
1) try to change your USB power source incl. possible cabling
2) try to go with vanilla Raspbian + standard clean installation of deCONZ

Mikha495, 2019-08-22:
Found out, that deCONZ is somehow not booting (or not booting correctly) after unplugging from power. 
I’ve to manually stop and disable deCONZ, restart Homebridge and start and enable deCONZ. 
Then it runs. 

Any idea how to set it up correctly?

ebaauw, 2019-08-22:
deCONZ crashes (SIGABRT) when started in non-minimal mode before the graphical environment (X11 server) has been started.  During boot, this happens a few times in quick succession, and sometimes `systemctl` gives up on the `deconz-gui` service.

We haven’t been able to find the proper dependencies for the `deconz-gui` service to prevent this, but using a timer to  delay starting the service works, see #1535.

Mikha495, 2019-08-22:
Ok, but deconz-gui is disabled in my setup. Is what you are writing also valid for deconz (headless)?

ebaauw, 2019-08-22:
I don’t run headless, but I don’t think so.  You should be able see in the syslog.

zladukas, 2019-08-22:
@Mikha, 
I'm using 3minutes timer for homebridge, just try and You'll see.

Nastras, 2019-08-26:
@Mikha495 You can set it headless exactly as described in #1535 only that the service is not called deconz-gui.service but is it only deconz.service

By the way. Despite the delay of the service I often had the case that he didn't recognize the conbee stick and the firmware update screen appeared.

I also created a systemd timer which performs a soft reset (sudo GCFFlasher_internal -r) of the conbee shortly after boot. Now it is 100% reliable when recognizing the conbee.

Here my timer and the service as someone would like to use it:

sudo nano /etc/systemd/system/deconz-conbee-reset.timer
`[Unit]
Description=deconz-conbee-reset.timer

[Timer]
OnBootSec=20
Unit=deconz-conbee-reset.service

[Install]
WantedBy=timers.target	
`

then create the service with:
sudo nano /etc/systemd/system/deconz-conbee-reset.service

`[Unit]
Description=deconz-conbee-reset.service
Wants=network.target

[Service]
Type=oneshot
User=dannypiper
ExecStart=/usr/bin/sudo GCFFlasher_internal -r

[Install]
WantedBy=multi-user.target`

now activate the timer with :
sudo systemctl enable deconz-conbee-reset.timer

Mikha495, 2019-08-26:
@Nastras thank you! Set up the timer for deconz.service.

Regarding the second timer (sudo GCFFlasher_internal -r):
Do I have first to disable the deconz.timer by `sudo systemctl diable deconz-gui.timer`? Or do both timers run parallel?

Nastras, 2019-08-28:
Both timers run in parallel. Both timers start once after a reboot.

The deconz-conbee-reset.timer starts a command which performs a soft reset on the conbee. The deconz-gui.timer starts the service deconz-gui. This service will be active as long as deconz is running.

Both timers are in standby after execution until the boot event occurs again.

Mikha495, 2019-08-30:
So today I unplugged my Raspberry in order to do a backup. After restarting, I receive following error:
[image]
Already disabled both timers, enabled decent.service, restarted everything, re-installed the Homebridge.hue plug-in. Nothing helps. Somehow the REST-API cannot connect. Phoscon is is running well and displays all connected devices. 
How can I solve this?

Mikha495, 2019-08-30:
Ok, this was strange.
Have ran the  GET request to https://dresden-light.appspot.com/discover in the REST plug-in for Firefox and it showed me the Phoscon-GW is now running on a different port, than before. I included the new port to the "host" in the Homebridge-hue entry of the config.json and now it runs. 

Maybe I should now reactivate the timers... but what would prevent the change of the ports? Any setting in my router? Don't think it's possible to set a fixed port for a service, isn't it?

ebaauw, 2019-08-30:
The ports are specified as command line parameters when starting deCONZ.  If you hard-code these in `deconz.service` or `deconz-gui.service`, they will be overwritten with each deCONZ upgrade.  You need to use an override instead.

I use `/etc/systemd/system/deconz-gui.service.d/override.conf`:
```
[Service]
EnvironmentFile=/etc/default/deconz
ExecStart=
ExecStart=/usr/bin/deCONZ $DECONZ_OPTS
```
And then specify the options in `/etc/defaults/deconz`:
```
# DECONZ_OPTS="--http-port=80 --upnp=0 --dbg-info=2 --dbg-aps=2 --dbg-zcl=1 --dbg-zdp=1 --dbg-http=1 --dbg-error=1"

DECONZ_OPTS="--http-port=80 --dbg-info=2 --dbg-aps=2 --dbg-error=1"
```

closed 2020-01-04 by stale[bot]