dresden-elektronik/deconz-rest-plugin #1738: Device "wlan0" does not exist.

patrickjane, 2019-08-07:
I have a working installation of deconz (conbee stick + installed debian package https://www.dresden-elektronik.de/deconz/raspbian/deconz-latest.deb), but I've recently turned off wifi on my raspberry pi (it is not needed since its connected via ethernet).

However ever since I get spammed by those errors:

```
Aug  7 18:16:15 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:17 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:18 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:20 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:21 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:23 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:24 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:25 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:27 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
Aug  7 18:16:28 homepi deCONZ-WIFI2.sh[342]: Device "wlan0" does not exist.
```

Automation and access to zigbee devices still works. So would there be a way to turn off those messages?

dhartaifb, 2019-08-13:
I got a similar error and just disabled the wifi component. 

    sudo systemctl disable deconz-wifi.service

patrickjane, 2019-08-13:
Thanks, worked.

patrickjane, 2019-08-23:
Actually, after every reboot, the service was started again, even though I disabled it. I have yet to find a way to disable this permanently.

dhartaifb, 2019-08-28:
Saw this, too. Thought it would be off for good.

Smanar, 2019-08-28:
In the last version  V2_05_67 you have
"Don't spam logs with wlan0 messages"
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_67
Make a try.

patrickjane, 2019-08-29:
I can confirm the message went away with the latest version, which is also available as deb package:

```
pi@homepi:~ $ wget https://www.dresden-elektronik.de/deconz/raspbian/stable/deconz-latest.deb
--2019-08-29 08:47:26--  https://www.dresden-elektronik.de/deconz/raspbian/stable/deconz-latest.deb
Resolving www.dresden-elektronik.de (www.dresden-elektronik.de)... 84.19.195.242
Connecting to www.dresden-elektronik.de (www.dresden-elektronik.de)|84.19.195.242|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5118914 (4.9M) [application/x-debian-package]
Saving to: ‘deconz-latest.deb’

deconz-latest.deb                                                     100%[=========================================================================================================================================================================>]   4.88M  1.98MB/s    in 2.5s

2019-08-29 08:47:29 (1.98 MB/s) - ‘deconz-latest.deb’ saved [5118914/5118914]
```

```
pi@homepi:~ $ sudo dpkg -i deconz-latest.deb
(Reading database ... 55143 files and directories currently installed.)
Preparing to unpack deconz-latest.deb ...
Important: /boot/cmdline must not contain console=serial0

use raspi-config to configure serial port access as described in user manual
Unpacking deconz (2.05.67-raspbian-jessie-beta) over (2.05.66-raspbian-jessie-beta) ...
Setting up deconz (2.05.67-raspbian-jessie-beta) ...
enable systemd deconz-update service
Important: /boot/cmdline must not contain console=serial0

use raspi-config to configure serial port as described in documentation:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements
Processing triggers for mime-support (3.62) ...
```

The Wifi-Service *still* gets started (even though it was `systemctl disable`'d before, but the error message is gone.

closed 2020-01-03 by stale[bot]

SwoopX, 2020-07-05:
You can ultimately mute the service with `sudo systemctl mask deconz-wifi`. However, the messages should be gone with the next release. Requested the necessary changes...