dresden-elektronik/deconz-rest-plugin #1739: Raspbee was working, now nothing

redlegoman, 2019-08-07:
I have a Raspbee, connected to a Pi 3B+ and it was working fine for about 4 weeks, I had bulbs, switch and sensors all working fine. However, sometime today everything in Home Assistant became "Unavailable" and all my devices in Phoscon are greyed out. I have tried powering off the RPi, I have tried using another Pi, but now the Phoscon app will just not find any bulbs or switches. The Phoscon app seems to work OK, it can see the Raspbee and report it's firmware version (26330500), but it ust won't find anything. Tried GCFFlasher_internal -r, leaving and joining the network, I even flashed the latest firmware again (I was already on the same version).
Any ideas on how I can troubleshoot this? Or is my Raspbee kaput?

redlegoman, 2019-08-07:
I just tried using the "change channel" option in Phoscon and it said "The channel could not be changed." Might this indicate a hardware fault?
I also tried to create a backup, even though I have nothing to backup and that also failed with: "Creating backup failed".

manup, 2019-08-08:
Hi, which version of deCONZ are you using?

Can you please check that the serial interface is still correctly configured as described here:

http://phoscon.de/en/raspbee/install#raspbian

When all is configured properly you may check the hidden "Zigbee configuration" page if the configuration parameters have changed, here you can also revert to a previous working configuration.

Usage of Zigbee configuration page:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

redlegoman, 2019-08-08:
The deCONZ version I have is 2.05.66; it is the latest version available fron the site. I downloaded it yesterday to install it on a different Pi. The serial is working, like I say, the Phoscon app can see the Raspbee and can report its serial number, and deconz itself shows it and I can go to configure network, but it just doesn't do anything anymore.

manup, 2019-08-08:
> I can go to configure network, but it just doesn't do anything anymore.

Do you mean the Zigbee configuration page; does it show anything?

redlegoman, 2019-08-08:
What's the Zigbee configuration page? I use the Phoscon webapp and the deconz GUI. They both show the Raspbee device. In my last comment I was talking about the F9 "Network Settings" window:
[image]

[image]

redlegoman, 2019-08-08:
[image]

[image]

manup, 2019-08-08:
Ah looks like the configuration is pretty messed up, that is also the reason the backup can't be created.

>What's the Zigbee configuration page? 

It's a hidden page in the Phoscon App. With that page you can restore a previous configuration where the parameters were working (given that the gateway wasn't reset in between).

Usage of Zigbee configuration page:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Can you please show a screenshot of that page.

redlegoman, 2019-08-08:
Do I have to install that software? If I alt-click the advanced icon with the version I have right now, it just changes to an empty screen:
[image]

redlegoman, 2019-08-08:
OK, now I see this screen, but when I click "Load" I get a segmentation fault:
[image]

[image]

manup, 2019-08-08:
Something is _really_ really strange here :/

Can you please send the file `/home/pi/.local/share/dresden-elektronik/deCONZ/zll.db` to support@dresden-elektronik.de so I can check if the configuration is still in there.

Earlier in this thread you said that:

>The serial is working, like I say, the Phoscon app can see the Raspbee and can report its serial number

But I've noticed that /dev/ttyAMA0 is shown instead of /dev/ttyS0 therefore can you please double check that the serial interface is correctly configured as described here:

http://phoscon.de/en/raspbee/install#raspbian

redlegoman, 2019-08-08:
I've done all the steps in that link, a couple of times, because like i said, I have tried this on a fresh Raspberry Pi as well as the Pi that this was working OK on. The same things happen on the fresh Pi as the Pi it was working on.

I have actually clicked the "Reset Gateway" on the Gateway page now, and obviously that has erased all my lights and switches (which were greyed out and unavbailable in home assistant), so when I get that chance a little later, I will try to pair the bulbs and sockets again.

redlegoman, 2019-08-11:
Well.  I don't know hat went wrong, but it's all working again now. I did all sorts of things, re-flashed the firmware, restored from back, reset the gateway, all seemingly to no avail. Except I gave up on Friday, was out all day Saturday and now it's gone back to working. All the devices are no longer greyed out and home assistant can see and operate the devices too. Strange.
I don't really know which of the things I tried made the difference, but something did. Or maybe none of them.

closed 2019-08-11 by redlegoman

manup, 2019-08-15:
>I've done all the steps in that link, a couple of times, because like i said, I have tried this on a fresh Raspberry Pi as well as the Pi that this was working OK on.

Hmmm, I don't understand why the  /dev/ttyAMA0 is shown, this shouldn't be the case after the serial port is configured via sudo raspi-config.

Can you please provide the output of the following commands:

    cat /etc/os-release

and

    uname -a

>I have actually clicked the "Reset Gateway" on the Gateway page now, and obviously that has erased all my lights and switches (which were greyed out and unavbailable in home assistant), so when I get that chance a little later, I will try to pair the bulbs and sockets again.

Sadly the zll.db file gets lost by this, it would have helped to investigate the root problem.