dresden-elektronik/deconz-rest-plugin #1740: Philips Hue StripLight (7190155PH)

P2mb2, 2019-08-07:
Hi everyone, 

I'm trying to pair a lightstrip from Philips. THe model is this:

Model Nr | 915005106701
Supplier Nr | 7190155PH

https://www.amazon.es/gp/product/B0148NMVQA/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Althought the light flashes 10 times... meaning its reseting... i'm unable to add it to deconz on Home Assistant. I'm really new with to this... so... can you please help me move forward with this investigation? 

Thanks.

P2mb2, 2019-09-07:
Anyone successfully paired a philips striplight withdeconz ?

Smanar, 2019-09-08:
Hello, some devices need a powercycle after the factory reset.

P2mb2, 2019-09-08:
Thanks Smanar, 

how could i perform that powercycle ? Power on/off multiple times ?

Smanar, 2019-09-08:
Just one time. Make a factory reset, and the device make zigbee announce after the next power-cycle.

On Phoscon the detection can last some minuts, enought time to make a factory reset, them a power cycle.

BTW I haven't find the factory reset procedure on internet for this device, it works with remote ? flash 10 times ?

P2mb2, 2019-09-08:
I used the ikea tradfri command to reset the philips light strip. Reseting the tradfri command closer to the striplight battery would make the light reset and flash 10 times.

https://www.ikea.com/pt/pt/catalog/products/10460751/

P2mb2, 2019-09-08:
No luck with the cycle power :( I've done the reset, and power off/Power on the strip... with deconz in search mode... nothing happens.

closed 2020-01-13 by stale[bot]

flavio, 2020-01-14:
To be honest I'm experiencing the same issue. I tried to do the cycle power trick and also to remove the led strip from the official philips hub app, no luck...

flavio, 2020-01-26:
Can we please re-open this issue?

I've tried several times to register the lightstrip against the philips hue bridge and then remove them from it. Unfortunately deconz never finds them.

I've also tried to perform touchlink, not sure it's supposed to work with lightstrips, but it didn't help.

The bridge is on a different channel compared to the one used by conbee II, I've tried to move it to the same channel and then de-register the lightstrip from the hue app. Again, no luck

rjhilgefort, 2020-06-04:
I'm having issues with this as well. Anyone had success with this?

Goldwing1973, 2020-07-18:
I am also having problems with pairing the lightstrips to deconz.

IT-Economy, 2021-06-02:
I tried to reset the light strip with the dimmer switch. It did not help. I still can't bind the strip. Is there a solution to this problem?

Goldwing1973, 2021-06-02:
> I tried to reset the light strip with the dimmer switch. It did not help. I still can't bind the strip. Is there a solution to this problem?

Did the light strip flash white after resetting it with the dimmer switch?

IT-Economy, 2021-06-02:
Yes.

I also did reset it with the hue app over bluetooth.

Goldwing1973, 2021-06-02:
The v1  lightstrip doesn’t have Bluetooth.

also, did you powercycle the strip after resetting it? Do the powercycle AFTER you start the search for new lights in Deconz.

Alternative what you can try (if you still have a Hue Bridge) is first try to pair it with the Hue bridge, after pairing, remove it from the bridge using the Hue App, powercycle the lightstrip and retry to pair with the Conbee 2

IT-Economy, 2021-06-02:
It's a new light strip.

Yes I did power cycle it several times.

I tried starting the search, resetting the strip with the remote, waited for it to blink and go back to normal, then power cycled it. 

Still no reaction in Phoscon.

I can pair it via bluetooth and then reset the light in the app. But that did not help either.

Goldwing1973, 2021-06-02:
The discussion here is about the 7190155PH (first gen) lightstrip.
I don’t have any experience with the Plus or Outdoor.

IT-Economy, 2021-06-02:
The current version apparently has the same problems with binding to deconz. So there was no fix for this yet?

Goldwing1973, 2021-06-02:
> The current version apparently has the same problems with binding to deconz. So there was no fix for this yet?

I have 5 7190155PH lightstrips working with Deconz.
The reset with the Dimmer switch worked fine on all of them.

mgrn0, 2021-08-08:
> The current version apparently has the same problems with binding to deconz. So there was no fix for this yet?

Did you succeed with using the new, updated light strip in deconz? Having the same troubles - cannot pair the light strip at all, even though it is using zigbee.

mgrn0, 2021-08-08:
After a dozen of resets I got the light to pair, however i cannot configure nor turn the light on and off through phoscon.

robinp, 2024-11-27:
Note: the reset procedure with the Dimmer switch as described in https://www.reddit.com/r/Hue/comments/lrzcis/new_hue_dimmer_swich_v2_bulb_reset/ worked. Though I have to note, the reception and/or the buttons are flaky, I needed ~10 attempts for the strip to start flashing. After flashing, ZHA picked it up fine.

(Were no need to use the Hue Bridge. Reset the dimmer; Pair with dimmer; Powercycle; Switch off with dimmer; Powercycle; Press dimmer against control box and keep buttons pressed -> Flash (eventually)).