dresden-elektronik/deconz-rest-plugin #1741: Experience report conbee II with deconz and homebridge-hue

Nastras, 2019-08-10:
Hello, everybody,

i tried deconz for the first time about two years ago. At that time the thing was still too unstable for me and I switched back to the native integration in HomeKit of my Philips and Ikea lamps and devices.

After having seen that there is a new Conbee stick and I would like to use deconz to manage various lamps and devices in a network, I have now re-learned everything in my home of lamps and devices from Ikea and Philips to deconz and integrated it into HomeKit via homebridge-hue.

There are currently 111 devices in my network. 
A lot of work -.-

I have installed hue taps as wall switches in the Gira frame at home everywhere and in HomeKit these are linked to the lights with triggers.

Unfortunately, I have to admit that everything is very unreliable at the moment.

Example if I switch on or off 16 lamps in a room via a switch then I can sometimes watch how one lamp is switched after the other. It is often very delayed but not always.

In addition, often individual lamps or devices are suddenly no longer reachable. Although I have a very tight network.

[image]

Is there a way to reactivate them via the deconz-gui? Sometimes it doesn't always work with reset select node.

Where I integrated the lamps and devices natively in HomeKit that was not the case.

Currently I am using deconz 2.05.66 with conbee II with firmware 26490700.

I'm afraid that if I can't get a grip on it my wife will force me to change everything I want to avoid. 

If someone has an idea what I can do with it I would be very grateful.

Best regards
Nastra

ebaauw, 2019-08-10:
> I have installed hue taps as wall switches in the Gira frame at home everywhere and in HomeKit these are linked to the lights with triggers.

That’s very resource intensive and generally a bad idea.  Using HomeKit scenes and automations, each light is sent a separate unicast command, leading to the delays you experience.  Also, unicast commands are vulnerable to routing issues in deCONZ.  Basically, there’s two alternatives to prevent this:
- Group your lights in deCONZ, and expose the group to HomeKit.  Only include the group in the HomeKit scene and/or automation.  This works well if you want to set the same state to all lights in the group;
- if you need different states for different lights, create a scene in deCONZ.  Homebridge-hue doesn’t yet expose scenes (see https://github.com/ebaauw/homebridge-hue/issues/475), but you can expose a CLIP flag and create a deCONZ rule to recall the scene, when the flag is changed.

I would create deCONZ rules to handle the Hue taps, so they work when HomeKit doesn’t.  I actually use Hue dimmer switches that control the lights (in the associated group) directly, so I can still control my lights when deCONZ is down.  In parallel, the dimmers drive deCONZ rules that amongst other set a flag.  The flag is exposed to HomeKit and drives the HomeKit automation to control my Sonos speakers.  This way, a single Hue dimmer press turn on or off both the lights and the music.

Nastras, 2019-08-11:
Very interesting what you write might be the reason for the bad performance of deconz.

I have to test the triggers in HomeKit for all hue taps disabled and set them up directly in Phoscon. The hue taps are now linked to their own scenes in Phoscon.

The result is that the performance when switching in rooms with few lamps has improved a bit. In my room where 19 lamps are switched the switching is unfortunately still very delayed. You can still watch how lamp by lamp is switched on.

If it's really the separate unicast command then it should work better in my test?

ebaauw, 2019-08-11:
I don’t use Phoscon, but, afaik, the REST API plugin doesn’t associate a group with the Hue tap.  I think Phoscon creates rules for the tap’s buttonevents.  These rules should contain a single action, recalling a scene by PUTing a `/groups` resource.  If you see lights turning on one by one, there’s still a PUTs per `/lights` resource.

Nastras, 2019-08-11:
I assumed that phoscon will group the lights in deconz via the REST API as described above and no single commands will be sent anymore.

I will try again to create the groups directly from deconz.

deconz scenes in Homekit as switch I want to use as last variant. Great that you implemented it yesterday.

It's a pity that it's so difficult to switch some lights without delay. I thought deconz would be more user friendly 😞 by now.

ebaauw, 2019-08-11:
> I assumed that phoscon will group the lights in deconz via the REST API as described above and no single commands will be sent anymore.

So would I.

Did you double-check that the HomeKit automations were actually disabled, before starting the configuration in Phoscon?  It might take up to 30 min for HomeKit to sync the changed configuration from your iPhone to the HomeKit hub over iCloud.  When doing multiple changes at once, sometimes some of them get lost or reverted.

Did you check what rules Phoscon created?   If you attach homebridge-hue's debug dump file, I could have a look.  And did you double-check that these rules are actually triggered when pressing the tap's button?  If you run `homebridge -D`, you can see in the homebridge log which changes are done on the gateway (using deCONZ rules) vs which changes are initiated from HomeKit.

> It's a pity that it's so difficult to switch some lights without delay.

The issue is switching a lot of lights _at the same time_ without delay.  The lights share a low-bandwidth ZigBee network that can only handle so many messages at a time.  deCONZ (like any decent ZigBee gateway) throttles the ZigBee requests to the lights, not to overload the ZigBee network.  homebridge-hue throttles the (REST API) requests to the gateway, not to overload the gateway (although this is more an issue with the Hue bridge than with deCONZ).  Each light actually switches on without delay, once the request is made.

> I thought deconz would be more user friendly 😞 by now.

I don't use Phoscon; for user friendliness, I use HomeKit ;-)

Nastras, 2019-08-11:
> Did you double-check that the HomeKit automations were actually disabled, before starting the configuration in Phoscon? It might take up to 30 min for HomeKit to sync the changed configuration from [...]

Yes, I have. -)

> Did you check what rules Phoscon created? If you attach homebridge-hue's debug dump file, I could have a look. And did you double-check that these rules are actually triggered when pressing the [...]

I have started the debug mode. Can't but not really see what in the log. I'm sure it's the scenes from Phoscon because I've added some color changes for testing.

> The issue is switching a lot of lights at the same time without delay. The lights share a low-bandwidth ZigBee network that can only handle so many messages at a time. deCONZ (like any decent ZigBee [...]

That's good to know..
It is advertised by up to 200 nodes. What would happen if I put all lamps in one trigger for my alarm system? Then the burglar could watch a nice play of light as one after the other lamp goes on.

> I don't use Phoscon; for user friendliness, I use HomeKit ;-)
😂😂😂😂😂😂😂👌

I once put my homebridge-hue.json on you. Maybe you see something?
[homebridge-hue.json.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3490010/homebridge-hue.json.gz)

What I also noticed is that if I operate the switch quite fast one after the other, the lamps are not switched at all although an entry is made in the log of homebridge-hue.

I have to wait about 10 - 15 sec before I press the switch again so that the state of the lamps is taken over.

Is it throttles the ZigBee requests?

I'm currently thinking about putting my hue-bridge and the Ikea gateway back into operation and creating three Zigbee networks and connecting them in HomeKit.

Network 1. Ikea Lamps
Network 2. Philips lamps
Network 3. deconz with Ikea outlets as repeater and Philips switches for more switching states

How do you feel about that? Then there should be no performance problems anymore.

Even if I am then far away from my goal to cover everything with a gateway. Then I even have one more than before 🥴

ebaauw, 2019-08-11:
> I once put my homebridge-hue.json on you. Maybe you see something?

Looks like Phoson made rules for two switches:
- Schalter 1 Badezimmer, group 10, Badezimmer (5 lights), scenes: 1, 2, 3, 4;  Only button 3 was pressed once.
- Schalter 1 Wohnzimmer, group 2, Wohnzimmer (20 lights), scenes: 1, 2, 3 and off.  Buttons pressed several times.

> Is it throttles the ZigBee requests?

No.  The rules look OK: group commands alright.  The other taps shouldn't do anything, or are still handled by HomeKit automations.

> What I also noticed is that if I operate the switch quite fast one after the other, the lamps are not switched at all although an entry is made in the log of homebridge-hue.
> I have to wait about 10 - 15 sec before I press the switch again so that the state of the lamps is taken over.

How did you create the scenes?  Could it be that deCONZ is still storing the scenes onto the lights?  This would explain the lights turning on one by one for the first time the scene is recalled.  The delays should go away once the scenes have been stored on the lights.

> I'm currently thinking about putting my hue-bridge and the Ikea gateway back into operation and creating three Zigbee networks and connecting them in HomeKit.
> How do you feel about that?

I was very close to doing that, when my deCONZ network kept losing Hue motion sensors and Eurotronic radiator valves (see #849).  Luckily, my network runs stable now.  Note that I am using group commands in all my rules.  I only set CLIP flags/statusses from HomeKit automations and use deCONZ rules to interact with the ZigBee devices.

ZigBee networks should become more stable with more devices, so splitting the network generally speaking is a bad idea.  Also, deCONZ runs more stable than HomeKit, so I would prefer not to depend on HomeKit to handle ZigBee devices.  I only use HomeKit to integrate and automate my non-ZigBee devices.  Make sure to use different ZigBee channels on each of the networks.

Different vendors seem to interpretate the ZigBee standard differently, however, so different networks per vendor makes some sense.  Hue doesn't scale well, as the lights don't support attribute reporting.  They need to be polled by the bridge/gateway; the more lights, the longer the delay between to polls of the same light.  As long as the lights are only controlled through the Hue bridge, this doesn't matter too much.  This becomes a pain when you use other means to control the lights (20th century wall switches, smart switches bound directly to the lights).
I would definitely pair the Hue taps with the Hue bridge in a multi-gateway setup.  deCONZ still has limited support for ZGP devices, in that they are all handled by the RaspBee/ConBee.  It still doens't configure the lights to translate ZGP commands.  This also means that the switches need to be in direct reach of the gateway.

IKEA does attribute reporting, and doesn't seem to like being polled, also for the discovery of the network.  I think deCONZ no longer polls them, but I'm not sure (I use mostly Hue lights).  I have no clue how to set the channel on the Trådfri hub.  I'm not even sure if it forms a single network, or one per room.  Just got one this weekend.  It doesn't expose switches nor sensors to HomeKit, so you'd need to connect these to deCONZ.  Also, no CLIP sensor equivalent, so no way to integrate it efficiently with HomeKit.  I think there is a homebridge plugin for it, but I never looked at it.  I don't think I would include the Trådfri hub in a multi-gateway setup.

Nastras, 2019-08-11:
> Looks like Phoson made rules for two switches:
Schalter 1 Badezimmer, group 10, Badezimmer (5 lights), scenes: 1, 2, 3, 4; Only button 3 was pressed once.
Schalter 1 Wohnzimmer, group 2, Wohnzimmer (20 lights), scenes: 1, 2, 3 and off. Buttons pressed several times.

Yes, it is. I have additionally set up a switch with a few lamps in phoscon for testing. I forgot to tell you.

> No. The rules look OK: group commands alright. The other taps shouldn't do anything, or are still handled by HomeKit automations.

Right the other switches are still furnished in homekit.

> How did you create the scenes? Could it be that deCONZ is still storing the scenes onto the lights? This would explain the lights turning on one by one for the first time the scene is recalled. The delays should go away once the scenes have been stored on the lights.

I created the scenes in phoscon with the scene editor. I don't know if the scene will be saved on the lamp or wherever. I don't know how to tell either?

[image]

> ZigBee networks should become more stable with more devices, so splitting the network generally speaking is a bad idea. Also, deCONZ runs more stable than HomeKit, so I would prefer not to depend on [...]

> ZigBee networks should become more stable with more devices, so splitting the network generally speaking is a bad idea. Also, deCONZ runs more stable than HomeKit, so I would prefer not to depend on [...]
I would definitely pair the Hue taps with the Hue bridge in a multi-gateway setup. deCONZ still has limited support for ZGP devices, in that they are all handled by the RaspBee/ConBee. It still doens't configure the lights to translate ZGP commands. This also means that the switches need to be in direct reach of the gateway.
IKEA does attribute reporting, and doesn't seem to like being polled, also for the discovery of the network. I think deCONZ no longer polls them, but I'm not sure (I use mostly Hue lights). I have no clue how to set the channel on the Trådfri hub. I'm not even sure if it forms a single network, or one per room. Just got one this weekend. It doesn't expose switches nor sensors to HomeKit, so you'd need to connect these to deCONZ. Also, no CLIP sensor equivalent, so no way to integrate it efficiently with HomeKit. I think there is a homebridge plugin for it, but I never looked at it. I don't think I would include the Trådfri hub in a multi-gateway setup.

If I understand you correctly would you in my case set up the hue taps, philips lamps and ikea lamps on two hue bridges as multi networks to achieve the best result?

ebaauw, 2019-08-11:
First, second, and third, I would try a single network.  If that doesn't work and a split is absolutely needed, I'd connect all Hue and Friends of Hue lights, sensors, and switches to a Hue bridge (or two, if needed for scaling), and everything else to deCONZ (and don't use the Trådfri hub).  I would definitely not connect Trådfri lights to a Hue bridge.

Nastras, 2019-08-12:
I'm gonna try it that way, see if it gets any better.

From deconz I am a little disappointed. I hoped for a better performance. Especially because Dresden Elekronik advertises with the support of up to 200 devices from different manufacturers.

And I use only two manufacturers philips and ikea and had only 96 devices trained.

Learning 200 devices alone doesn't help if the power is not sufficient.

Erik i thank you first and will report how the performance is after switching to the hue bridge.

Nastras, 2019-08-14:
Hello @ebaauw 

I've moved my network for testing now. Something different than you suggested. I connected all phillips lamps to the hue bridge and all Ikea lamps to the Ikea gateway. 

With deconz I have all philips switches and all Ikea outlets connected so I have the extra buttons to program. 

I programmed the switches as before in homekit because I thought there should be no power problems with 3 gateways even if each lamp is controlled individually with unix commands. 

Result was that yesterday everything reacted super fast. So I expected it from deconz where all devices were registered there.

Unfortunately I have a new problem with deconz after a reboot since this morning. 
Since the reboot deconz only connects to three devices of 31 devices. All other devices are no longer available and I have no idea why. I am really disappointed. deconz still makes me crazy. 

[image]

I have tried the following solutions: 

- reset the affected nodes 
- conbee II soft reset with GCFFlasher_internal -r 
- conbee II removed from pi and reconnected 
- Zigbee configuration checks if panid etc. has changed, here I have only two entries. (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1226) 
- one of the affected sockets is disconnected from the power supply and plugged in again. 

I had nothing changed to deconz since last night where everything works and I have no idea where the problem is now?

ebaauw, 2019-08-14:
> Since the reboot deconz only connects to four devices of 31 devices.

Did you actually try and control the devices, or did you only look at the lines in the GUI?  What happens if you select a node in the GUI and press `0`?  What if you press `L`?

Smanar, 2019-08-14:
Hello, what does the "L" key do ?

ebaauw, 2019-08-14:
It sends a _Leave with automatic rejoin_ request (or something like that) to the device.  I think this is supposed to reset the device, except for the network settings.  Sometimes it helps to revive frozen devices.

Nastras, 2019-08-15:
I reset the network yesterday just before you replied. Couldn't test your tips anymore. 

To your inquiry, the devices were really not available. What still worked were the hue taps which are not shown in the gui.

The new network is still running and has survived a reboot.

It is now my last attempt with deconz should the network destroy itself again I give up. And finally hook up deconz.

I will report again at 😉

Again thank you for your effort.

Nastras, 2019-08-18:
Hall Erik,

I'd like to report again. Everything has been running very stable and without problems for a few days now.

Even though I now have three zigbee gateways in use instead of one as planned, I am enthusiastic about the performance.

I hope it stays that way.

 greetings Nastra

closed 2019-08-18 by Nastras