dresden-elektronik/deconz-rest-plugin #1742: Ikea "neuer" Bewegungsmelder von Trådfri spinnt

Michaelnorge, 2019-08-10:
Ich habe mir heute ein Bewegungsmelder von Trådfri gekauft und eingebunden.
Allerdings schalten bei Bewegung ALLE angeschlossenen Trådfri-Steckdosen (sind bei mir 23).

Ich benutze die neuere Version des Bewegungsmelders. Vielleicht nicht 100% kompatibel?

Sogar als ich den Trådfri Bewegungsmelder aus der Phoscon-App entfernt hatte haben die Trådfri-Steckdosen noch immer reagiert. Erst nach einem Reset des Bewegungsmelders (4x drücken) war endlich schluß.

Smanar, 2019-08-10:
I m not sure (I have used google translate) But I think you will have somes information here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676
But no solutions yet.

Michaelnorge, 2019-08-11:
Yes Smanar you are right. it´s the same problem 
Thx for information!

closed 2019-08-11 by Michaelnorge