dresden-elektronik/deconz-rest-plugin #1743: how to delete phantom switch in the API?

clementx2006, 2019-08-11:
Absolutely confused. Saw there are "sensors" and "lights" but not "switches"? 

I tried to pair an IKEA remote control and nothing showed up in the Web App but there is a node in the GUI that looks like the remote controller (and named as such) and that was not there before. So wanted to delete that but found no way... must have missed a simple trick? 

Btw pairing IKEA bulbs and steering devices is such a task...sigh...

Smanar, 2019-08-11:
In the GUI, just select it, and press "delete" key on the keyboard.

clementx2006, 2019-08-11:
Deleting that way doesn't really permanently delete the node... 

In my case it was one of the E27s...

Smanar, 2019-08-11:
Ha ok so you have a "ghost" device ? In this situation you have 2 situations, A device with a special MAC address, totally unknow, for this one I don't know how to delete it, or a ghost make by real device, here it's  just some data in a "neighbour table", you just need to wait for some days, and retry.

But it's not a real problem, just delete your device in the api (you can keep it in deconz), and repair your remote using "switch" option, the api will create again the device (and configure it) with better options.

clementx2006, 2019-08-12:
Thanks - the IKEA remote is now fixed!

but I really want to delete another phantom node as I have sold the device... kind of weird if the switch node cannot be retrieved in the API whereas the lights and the sensors can...

Smanar, 2019-08-12:
Don't worry, if you are in the second situation, just wait for some days, and you will be able to delete it with "delete" key on deconz.
And in the api you can retreive switch node info too.

closed 2019-12-17 by stale[bot]