dresden-elektronik/deconz-rest-plugin #1744: deconz binding - how to use switch when deconz is offline, but keeping some automations

sr-1989, 2019-08-11:
Hi everyone !
i'm discovering deconz a little bit more every day! :)
i'm using deconz (with a conbee 2) with hassio. i succeed to make some automations, but now i'm asking me if it's possbile to bind my switchs with lights when deconz is off ? (and keep my deconz events for my other automations with hassio). 
I've read the doc, looked for examples, i tried to bind but with a lot of difficulties, each time i got a timeout error... few time i succeed was when i reset the unit or did a ping... maybe do you have a better way to do this ?

When i had a success message to bind my switch and light or plug,, nothing worked... i try to bind IEEE and try to bind group. But when i disconnect deconz it's not working... any idea ?
i don't see the black wire like in the doc to show the binding...

any ideas?

Thanks in advance for your help! 

Seb

Smanar, 2019-08-12:
I have only done that with ikea remote, never suceed with switch, perhaps a cluster story problem, but you have some tips here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/431

sr-1989, 2019-08-12:
Hi!
thanks for you help ! 
if i correctly understand the other post i can only bind in deconz the switch which are shown in the phoscon app in the menu... (only ikea and philips hue switch are shown...) i've done the test with my ikea switch and it's working when conbee is off. This is exactly what i want. (and i can keep some automation with hassio when it's on)
but a little disappointed to can't do this with the xiaomi switch :-s ... i've 3 switch "transmitter 2 gang" and i'd like to do the same thing... maybe there is a way ?

thanks !

Seb

ebaauw, 2019-08-12:
> if i correctly understand the other post i can only bind in deconz the switch which are shown in the phoscon app in the menu

Whether you can bind a switch depends on the switch itself.  Some switches (IKEA, iCasa, innr, ...) send commands directly to (a group of) lights.  You use bindings to configure which groups and/or lights.  deCONZ eavesdrops on these commands to reverse-engineer the button events.  For these kind of switches, the API exposes `config.group` in the `/sensors` resource.  Other switches (notably Xiaomi) use attribute reporting to report button events to the gateway.  These switches cannot control (a group of) lights directly.  Afaik the Hue dimmer is the only switch that does both.

We should probably mark in the Wiki how each switch works - it should be a major consideration when deciding what switches to get.

zladukas, 2019-08-13:
Erik, 
could You clarify, if IKEA Tradfri on/off has config.group pointing to group "X" and I will include some lights into  group "X" "lights", eg   "lights": [ "1", "2", "3" ], switch will control these lights, even gateway down?

ebaauw, 2019-08-13:
Yes.  Same as for any switch which includes `config.group`.  Make sure any automations don't contradict the group commands, or you'll get funny behaviour.

Smanar, 2019-08-28:
Reponse here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1137#issuecomment-525588380

zladukas, 2019-09-17:
I'm successfully using config.group on several remotes.

Is it possible to setup this group, that one button remote, Ikea TRADFRI remote control center button, (using scenes, rules, clip, etc.) would switch light 100% brightness or off, since other sensors, e.g. motion sensor, are using different same lamp brightness settings?

If yes, what is best way to achieve this?

dakipro, 2019-09-20:
Hi, I am planing to into buying deconz to use it with homeseer and ikea (and some xiaomi sensors). 
I would like to confirm that I understand how this all is done :) 
I have 5 non-color bulbs and 3 round switches from Ikea. I want three lights on one switch, and other two lights on their own switch. Do I need to pair all these 8 devices with decons, and then assign their grouping from UI and they will just work the way I assign them even if I unplug deconz stick from the server?
Or do I need to manually pair each switch to each light, and then include only switches into deconz plugin for homeseer?
Or something else entirely?

closed 2020-01-29 by stale[bot]