dresden-elektronik/deconz-rest-plugin #1745: Downgrade to 2.05.65

bdickhaus, 2019-08-12:
Version 2.05.66 was released almost two months ago. And it broke a couple of things for me. I have been patiently waiting for 2.05.67 which is supposed to fix these issues, but it has not appeared.

I've moved to the apt based updates because it is easier to do updates. However there is no path to go back to a previous release. Before I would just apply the previous .deb and that was it.

So, my assumption is that I go back to the manual process of updates to downgrade. But do I also have to downgrade the firmware? I've never done that before, but I know there is a documented process.

peer69, 2019-08-12:
You can go back using the "old" way by installing via dpkg -i. It will overwrite the existing version. I have done the same to get rid of the .66 bugs. Just dont upgrade deconz via apt afterwards until the new version is released.

TroLoos, 2019-08-12:
Yeah, v66 seems like a misunderstanding :-( Unfortunately I already moved almost 30 devices from Xiaomi Gateway, don't want to go back, will try v65 as well.

To give you perspective how bad it is - one of my Aqara switches (wired) converted itself in Phoscon to Multi Sensor :-) It also started to report low battery warnings!! What a mess... not to mention that some Phoscon devices (Aqara Multi Sensor, Aqara Wireless Button) are visible in Phoscon, but Deconz GUI is not aware of their existence.

Larsn1, 2019-08-12:
Instead of downgrading I would like to upgrade to 67.

peer69, 2019-08-12:
You should be able to compile the current master yourself but I’ve never done that.

Larsn1, 2019-08-12:
I’ll try that later

bdickhaus, 2019-08-14:
I did the old dpkg install and all seems well. I don't want to compile .67 because there is likely a good reason that they have not released it yet and I didn't want to introduce other problems. My main issue was with the Hue LWB bulbs not reporting current dim level.

closed 2019-08-16 by bdickhaus