dresden-elektronik/deconz-rest-plugin #1746: Development Stalled?

gmitch64, 2019-08-12:
Looking at the Code page, the Latest commit (254f1a8) was 18 days ago, which seems to be an awful long time ago, particularly with the issues with the 66 release. Does anyone know what's going on?

G

00lex, 2019-08-12:
its holiday time in germany

alex0411, 2019-08-13:
But there was the same situiation before the last releases. There are still many things unsolved (sensor get lost, first push to a button is not recognized, etc.) and new promised features (rework of sensor, etc.) never arrived. At the moment I am thinking about switching back to CC2531.

jo-me, 2019-08-13:
@alex0411 is there already a ticket for the "first push to a button is not recognized" issue?
I'm seeing this for Osram Smart Plugs at least.

Hypfer, 2019-08-28:
Hotline told me that the workload is just too high for the available employees. Development hasn't stalled. They're just over capacity atm.

GottZ, 2019-08-28:
sounds like a good reason to go open source then or to let some independents in under nda

closed 2020-01-02 by stale[bot]