dresden-elektronik/deconz-rest-plugin #1747: Sensor working but missing in Phoscon sensor list

udoewich, 2019-08-12:
I'm working on adding support for a motion sensor and have it paired, working and visible via the API:
```
{
    "config": {
        "battery": 100,
        "duration": 90,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "27282d54132d9a50ead664fbd9ba950f",
    "manufacturername": "ORVIBO",
    "modelid": "895a2d80097f4ae2b2d40500d5e03dcc",
    "name": "Motion1",
    "state": {
        "lastupdated": "2019-08-12T21:55:33",
        "lowbattery": false,
        "presence": false,
        "tampered": false
    },
    "type": "ZHAPresence",
    "uniqueid": "00:15:8d:00:01:c1:24:8d-01-0500"
}
```
However, it is not shown in the Connected sensors list of Phoscon.  I was able to assign it to a group via Sensor control and it works fine triggering events when motion is detected.

What determines whether or not a sensor is listed under Sensors?

Ronoon, 2019-08-15:
the phoscon app is not open source therefor you have to ask @manup to integrate the motion sensor to the app

udoewich, 2019-08-15:
Right, forgot about that.  I'll look into getting my changes into shape for a pull request.

liorm0505, 2019-08-19:
Hi udoewich, 

May i know how did you able to pair the device and get the info through the API?
When i pair my device the node has no name but it still visible in the deconz app. 
Also, Requesting the api for all sensors does not respond with this sensor.  

I am running the app on Rp3 with HASSIO the deconz addon.

udoewich, 2019-08-20:
Sensors that are not listed as already supported (see the Supported Devices Wiki page for that) have to be added to the source code before they are recognized by the API and Phoscon.  deCONZ will show devices even if they have not been added to the REST API yet.

I had to make changes to the code before the ORVIBO sensors worked for me.  I'm prepping a pull request right now so hopefully there will be official support for these soon.

liorm0505, 2019-08-21:
That's great news :)
I would love to have support for this sensor in the next release.

udoewich, 2019-08-21:
Turns out that it _does_ list the sensor on Phoscon. A second sensor I added showed up just fine on the sensor listing. The first still not being listed might be due to the incomplete support on the API side at the time I added it. I'll eventually remove it from the database to make sure.

closed 2019-08-21 by udoewich