dresden-elektronik/deconz-rest-plugin #1748: Aqara temp sensor reports -100 degrees

Schlbuditz, 2019-08-13:
Using Conbee 1 and Deconz V2_05_66, but the issue has also been there in previous versions.

Sometimes a couple of my Aqara temp/hum/press-sensors report -100°C but it is not even close to that temperature where it is located.

This is a picture from my Home Assistent history from one of the sensors that is reporting faulty values. 
[image]

When using the Xiaomi gateway instead of Conbee and Deconz this is never an issue.

bdickhaus, 2019-08-14:
I have also seen this. My application has a way to throw away values outside the norm, and that is what I have done. I thought it was a sensor problem, but I don't have the Xiaomi gateway to compare to.

Smanar, 2019-08-14:
You are using too Home assitant @bdickhaus ?
Because I haven't this problem on my side but on another application.

bdickhaus, 2019-08-14:
No, I use HomeSeer with the JowiHue plugin to interface with deCONZ and the Data History plugin to collect the data.

Smanar, 2019-08-15:
"Data History plugin" ? it' a logger ? So you have perhaps logged the bad web socket return ?

bdickhaus, 2019-08-16:
It is a plugin for HomeSeer. The JowiHue plugin collects the data using the deCONZ REST API and then the Data History plugin logs the data in a database on the HomeSeer system.

Hypfer, 2019-08-28:
Can confirm.

This seems to be related to the battery level of the temperature sensor. The frequency of these false readings increases with the battery voltage drop.

Smanar, 2019-08-28:
Yep, but if someone know how bad data looks, it will be possible to ignore bad data in code, but without that ...
No-one have some log, at least the websocket return to check the bad data, to reconize them ?

jo-me, 2019-09-26:
I have the same issue on deconz 2.05.69 with Firmware 26330500 on Conbee (1).
I'm seeing this for 2 out of my 3 Aqara Temp Sensors but only for temperature values and not for humidity/pressure.

Xiaomi Temp Sensor also seem to work fine.

Smanar, 2019-10-03:
Have you tried to change battery ? cf > https://www.domoticz.com/forum/viewtopic.php?f=68&t=29370

NigelRook, 2019-10-15:
Have also had this. Changing the battery has worked in one case, and the device did credibly require a new battery (the temperature readings were flat). Have had another device just start doing it and haven't tried this yet.

It's worth investigating where the -100 readings are coming from though, they're obviously not accurate. If they're due to deconz mishandling communications errors caused by the low battery (big if), then that wants fixing.

NigelRook, 2019-10-15:
Confirmed that changing the battery works for one of my devices. That domoticz post suggests that the issue is present when using the official xiaomi gateway, which suggests the error is coming from the sensor itself rather than an deconz bug.

ebaauw, 2019-10-18:
I'm seeing the same.  As far as I can tell, the -100°C is reported by the Xioami special report; the regular attribute reports for the _Measured Value_ of the _Temperature Measurement_ cluster seem OK.  The sensor reports 30.05V for the battery, which is still mapped to 100%.  Apart from the odd temperature value, the sensor seems to function normally, sending both regular and the Xiaomi special attribute reports.

ebaauw, 2019-10-19:
Damn, the -100°C is also reported by the regular attribute report for 0x0402/0x0000.

alex3305, 2019-11-25:
I also had this issue (#1708) and for me the fix was very trivial, as my battery was just low. Also a given solution by @Smanar.

Although I think invalid reading handling can also be improved in deconz (for these sensors).

jo-me, 2019-11-26:
@ebaauw 's fix should be in the latest version. I'm not seeing the -100° since I upgraded yesterday afternoon. Do you?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

hille721, 2022-09-22:
> It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

Currently facing this issues with ony of my Aqara sensors. Showing up since Tueasday, at now happens 2 to 3 times a day. 
Conbee 2 with Firmware 26720700 and Phoscon v2.17.01

Mimiix, 2022-09-22:
Hi there, 

This is a very old issue.

Please check your battery (replace it) and update conbee firmware.

If that doesn't fix it, please open an topic on the forums :)