dresden-elektronik/deconz-rest-plugin #1751: Ikea dimmer does not show up in Switch editor

Asbjoern, 2019-08-14:
I have added an Ikea dimmer to Deconz but it only works partially.

- It does show up as a group and adding it to Openhab works great and I get alle the different key events when I turn it.
- Adding the dimmer to a group using the Switch editor does not work. It does not appear on the list of available switches.
- Adding lights to the dimmer does also not work. None of the ligts respond to the dimmer.

I am running v. 2.05.65. I have tried updating to v. 66 but the problem remains.

I read that other peaple have the remote working. What is your secret?

madshell, 2019-08-31:
I currently have the Ikea dimmer and I get the exact same behaviour. The dimmer can produce 4 event types:
- slow rotate clockwise
- fast rotate clockwise
- slow rotate anti-clockwise
- fast rotate anti-clockwise

The only workaround I found is setting 2 events on 'turn off the lights' and 2 events on 'turn on the lights' on the Apple Home app (or Eve) to get it act as a simple switch. You can also set 0%, 25%, 50%, 100% on the 4 events but it is not very accurate so it always feel glitchy. :(

closed 2020-01-05 by stale[bot]

kzaoaai, 2020-06-12:
same issue with Home Assistant as the only information exposed is the event type, no value for the rotation angle. It's only useful if paired as a direct light link with a compatible light.
You can pair it using the old deconz GUI, it does not show as a switch option in the new GUI.