dresden-elektronik/deconz-rest-plugin #1752: Xiaomi Pairing Problems

OliverGeist, 2019-08-14:
With these types of Xiaomi Aqara devices is no pairing possible (using Phoscon App) although listed on the [Supported Devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices) list:

Xiaomi Aqara door/window sensor
Type: Mi Windows and Door Sensor
Model: MCCGQ01LM

Xiaomi Aqara Smart Cube

In opposite these items from the same set worked directly/fine:
Xiaomi Aqara Mi Wireless Switch
Model: WXKG01LM

Xiaomi Aqara Mi Motion Sensor
Model: RTCGQ01LM

Smanar, 2019-08-16:
Some sensors can be annoying, but from my memory the MCCGQ01LM is working >https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1412

Smanar, 2019-08-16:
Have you tried this procedure ?

> Press and hold the reset button on the device for +- 5 seconds (until the blue light starts blinking).
> IMPORTANT: Release and start pressing the reset button every second. This keeps the device awake, otherwise pairing will fail!.

OliverGeist, 2019-08-18:
The optically the sensor seems finishing the pairing process (fast blinking stops followed by a single blinking) but the Phoscon App does not show the device.

AutoStierna, 2019-08-18:
I might have the same issue. What I’ve noticed is even if the device is not showing in Phoscon app after pairing, it’s visible in hassio under integrations. Is it the same for you?

OliverGeist, 2019-08-18:
I use the Phoscon App just to register the devices and as an Middleware for ioBroker. But they are not being imported as known devices to ioBroker

fullcavediver, 2019-08-19:
> Have you tried this procedure ?
> 
> > Press and hold the reset button on the device for +- 5 seconds (until the blue light starts blinking).
> > IMPORTANT: Release and start pressing the reset button every second. This keeps the device awake, otherwise pairing will fail!.

+ close and open one time the door/window sensor with magnet. That works guaranteed. I recently paired several sensors.

OliverGeist, 2019-08-19:
> > Have you tried this procedure ?
> > > Press and hold the reset button on the device for +- 5 seconds (until the blue light starts blinking).
> > > IMPORTANT: Release and start pressing the reset button every second. This keeps the device awake, otherwise pairing will fail!.
> 
> and at the end close and open one time the door/window sensor with magnet. That works guaranteed. I recently paired several sensors.

WoW, that's really really strange!
With your last hint I was able to register the first sensor directly. After 10 minutes of trying for another one I rebootet the Gateway an after another 5 minutes I had two sensores registered.

Thanks a lot for this help!

closed 2019-08-19 by OliverGeist

Sambaro1, 2019-08-25:
Worked perfectly, thanks

djhampson, 2020-01-26:
> and at the end close and open one time the door/window sensor with magnet. That works guaranteed. I recently paired several sensors.

This worked for me - Reset, move the sensor close to a magnet.

ReX1983, 2020-08-14:
Just wanted to share my experience, hoping this could avoid some frustration.

I have just migrated 63 sensors (a mix of door, motion, temperature, water leak  sensors) from 3 Xiaomi gateway to Deconz (connected to Home Assistant).

12 sensors didn't want to pair: I had tried all the possible pairing sequences but it just didn't work.

Eventually I decided to try with new batteries and magically they all joined at the first attempt, within seconds.
**My guess is that the pairing procedure requires much more power than the normal operation mode and hence fails with a half discharged battery.**

Keep this in mind before giving up.

mgrn0, 2021-06-09:
I have the very same problem, pairing two brand new Aqara door sensors - phoscon does not see them at all. Has this issue been addressed in the current stable version?

janerahere, 2022-02-06:
Hi,
I've tried all solutions provided here, but no success.
Although I can see the sensor sensing messages via deConz webApp software. How is that possible? cannot be paired through Phoscon, but I can see it on DeCONZ... Any ideas? Thanks!!
[image]

janerahere, 2022-02-08:
> 

Conbee2

Mimiix, 2022-02-08:
Please use the forum @janerahere