dresden-elektronik/deconz-rest-plugin #1753: Lights being forced to 366 (via Apple Homekit)

SimplyGardner, 2019-08-14:
I am having issues with changing Hue lights via HomeKit _(connected by Deconz + Conbee ii)_. When I try and change the lights, it looks like something is getting pushed back and the light are defaulting back to **366**. I have created a video to better explain https://cdn-b-east.streamable.com/video/mp4/c2hk1.mp4?token=uqlwxv4Hg7GH8WGCewntyQ&expires=1565813040

**Home Assistant Log:**
[image]

**Deconz Log:**
```21:39:13:963 Announced to internet
21:39:13:964 discovery server date: Wed, 14 Aug 2019 20:39:13 GMT
21:39:13:964 	 local time seems to be ok
21:40:12:788 Current channel 15
21:40:12:797 Device TTL 6540 s flags: 0x7
21:41:12:786 Current channel 15
21:41:12:795 Device TTL 6480 s flags: 0x7
21:41:18:809 dev /dev/ttyAMA0
21:41:18:810 GW firmware version: 0x264a0700
21:41:18:810 GW firmware version is up to date: 0x264a0700
21:41:55:928 0x001788010247FA7B error APSDE-DATA.confirm: 0xE9 on task
21:42:05:670 delay sending request 183 dt 0 ms to 0x001788010247FA7B, cluster 0x0300
21:42:05:675 	0x001788010247FA7B force poll (2)
21:42:05:711 	0x001788010247FA7B force poll (2)
21:42:05:868 send move to color temperature 366 to 0x001788010247FA7B
```

**Issues researched:**
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014
- https://github.com/ebaauw/homebridge-hue/issues/353

**Setup:**
- Home Assistant: 0.97.2
- Hass.io Core Add-on: deCONZ: 3.1
- Conbee ii

daanvb, 2019-09-02:
I have just set this up and I am also noticing the same behaviour. My lights go to 153 though...

SimplyGardner, 2019-09-02:
@daanvb - what lights do you use? I think mine goes to 366 as it is the default hue value.

daanvb, 2019-09-02:
Tried on 3 Philips Hue A19 White and Colour bulbs. All exhibit the same behaviour and go to colour temperature 153 when I try and change them in HomeKit. 

Also whenever I open the bulb settings in HomeKit to change colour, it defaults to the ‘temperature’ colour ring first which it didn’t do when connected to the Hue bridge.

ebaauw, 2019-09-03:
> This behaviour happens when using both Deconz (via Conbee ii) and also via the Philips Hue plugin (connected via official bridge) - only when using HomeKit.

I’m not sure what you’re saying here.  Are you using Home Assistent to expose the lights to HomeKit?  Is this happening both to Hue lights connected to deCONZ as well as to Hue lights connected to the Hue bridge?

daanvb, 2019-09-03:
> I’m not sure what you’re saying here. Are you using Home Assistent to expose the lights to HomeKit? Is this happening both to Hue lights connected to deCONZ as well as to Hue lights connected to the [...]

I can’t speak for @SimplyGardner but for me, I am exposing the lights via the HomeKit component in HomeAssistant. I have just moved all the lights over to use DeCONZ in HomeAssistant instead of the Philips Bridge (where the colour picker in HomeKit worked fine)

I am now experiencing the issue that @SimplyGardner is reporting.

ebaauw, 2019-09-03:
Didyou use the native HomeKit feature of the Hue bridge, or did you also expose the lights connected to the Hue bridge through HA?

daanvb, 2019-09-03:
> Did you use the native HomeKit feature of the Hue bridge, or did you also expose the lights connected to the Hue bridge through HA?

Previously I used the native HomeKit feature of the Hue bridge.

Fienberber, 2019-09-03:
I'm experiencing the same problem too. I'm using diyhue with some esp's and mi light bulbs. I think the problem is due to how the plugin (homebridge-hue) exposes the lights to homekit.
I'm experiencing jumping of colors in the apple home app if I try to set the lights with the temperature gradient.
But this is not due to some sort of communication problem because it occurs when I try to set the temperature in a scene. And when creating scenes homekit does not send data to the hue bridge (to my knowledge).
I have no problem with other RGBW lights that don't use homebridge-hue.

Hoping this can help.

ebaauw, 2019-09-03:
> I'm experiencing the same problem too

Are you using HA or homebridge-hue to expose your lights to HomeKit?  In the first case: it has nothing to do with homebridge-hue, in the second case: it's _not_ the same problem.

ebaauw, 2019-09-03:
@daanvb I hate to be fingerpointing, but it looks like an issue with HA.  Hue lights report back the closest `ct` value when setting a light colour through `xy`.  It seems like HA uses that as a hint to update the colour temperature, setting `ct`.  Then, the Hue light reports back the closest `xy` value on setting `ct`, which HA takes as a hint to update `xy`, etc.  Basically HA should check `colormode` and not take action on changed `ct` values unless `colormode` is `ct` (and similar for `xy`).

To confirm, you could try and set a yellow-white-ish colour and see high `ct` values (up to 500) and a blue-white-ish colour to see low `ct` values (down to 153).

As far as I know, only Hue lights report back their actual state (rather than the last state set), and only Hue lights update the attributes for another colour mode, so you won't see this behaviour for other lights.

daanvb, 2019-09-03:
@ebaauw I think you are correct that this is a HA issue. I've done a little more digging and found some threads raised regarding this.

https://github.com/home-assistant/home-assistant/issues/23394
https://github.com/home-assistant/home-assistant/issues/24513
https://github.com/home-assistant/home-assistant/issues/19251

someguythatcodes, 2019-09-03:
I have had the same problem described here with a Smart Life bulb (controlled through the Tuya component) and Philips Hue bulbs for which I use a HUSBZB-1 to control them.

dodepier, 2019-09-18:
I have exactly the same issue with homekit and Lifx on Hassio 0.98.1

someguythatcodes, 2019-09-20:
I've had a play around with this and it looks like when you pick a colour in Apple's Home colour picker, it sends the hue and saturation values followed by the colour temperature value to homeassistant. 

The reason the behaviour described in this issue is happening is due to the fact that the hue and saturation are sent first from the Home app, which home assistant acts upon and changes the colour to the selected colour. The problem occurs because the colour temperature value is sent after the hue and saturation which homeassistant acts upon and sets the colour temperature, which causes the light to return to the previously selected colour temperature.

This does only seem to be a problem with Apple's Home app, as I've tried using other apps such as "Controller for HomeKit" and "Eve for HomeKit" which both work fine

I've made a change to the "type_lights.py" in my developer environment to only change the colour temperature in homeassistant if the value is different to the previous value which seems to be working well.

daanvb, 2019-09-20:
@someguythatcodes can/will this be officially integrated into the plugin

SimplyGardner, 2019-09-20:
@someguythatcodes that's great - thanks for looking into this - its greatly appreciated. What would be the next steps on this? Wondering if a PR could be added to HA?

someguythatcodes, 2019-09-23:
I will look into doing a PR into HA for this soon

daanvb, 2019-09-23:
> 
> 
> I will look into doing a PR into HA for this soon

That would be amazing thank you!

someguythatcodes, 2019-09-25:
I think Apple may have fixed this in iOS 13.1 as it seems to be working for me after the update

daanvb, 2019-09-26:
I have tried it too and it does seem to be working for me as well

closed 2020-01-31 by stale[bot]