dresden-elektronik/deconz-rest-plugin #1754: Feller with EnOcean PTM 215Z

agners, 2019-08-15:
I have a Feller switch (the most prevalent manufacturer of classic light switch and plugs in Switzerland). They have a device with the EnOcean module PTM 215Z, also says DOLPHIN on it.

https://www.feller.ch/de/Produktangebot/Funktaster

The device can be paired by pressing a button 10 seconds down (left switch down). However, I actually have a hard time to pair the switch with Phoscon. I only managed twice so far. It does not appear on the deCONZ GUI (since it is a ZigBee Green Power device that is normal, I have read). It appears as "Hue Tap 19" in Phoscon. Assigning the switch to a lamp in Phoscon seems to work, but I am not able to control the lamp (doesn't turn on or off). I do not get any events according to the deCONZ log. When scanning with HomeAssistant the device is announced to HA:

```
             '19': {'config': {'on': True},                                                                                                                                                                                                                                                                                                                                                                                                                                                   
                    'ep': 242,                                                                                                                                                                                                                                                                                                                                                                                                                                                                
                    'etag': 'b6ddbd2792880d8318803eaf5fc9b523',                                                                                                                                                                                                                                                                                                                                                                                                                               
                    'manufacturername': 'PhilipsFoH',                                                                                                                                                                                                                                                                                                                                                                                                                                         
                    'modelid': 'FOHSWITCH',                                                                                                                                                                                                                                                                                                                                                                                                                                                   
                    'name': 'Hue Tap 19',                                                                                                                                                                                                                                                                                                                                                                                                                                                      
                    'state': {'buttonevent': None, 'lastupdated': 'none'},                                                                                                                                                                                                                                                                                                                                                                                                                    
                    'type': 'ZGPSwitch',                                                                                                                                                                                                                                                                                                                                                                                                                                                      
                    'uniqueid': '00:00:00:00:01:70:87:41-f2'},
```

This is the debugging log form my second successful pairing:

```
23:10:34:551 APS-DATA.request id: 94, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 0 len: 17 tx.options 0x00
23:10:34:551 send GP pairing
23:10:34:551 SensorNode 19: Hue Tap 19 reactivated
23:10:34:670 APS-DATA.confirm id: 94, status: 0x00 SUCCESS
23:10:34:670 aps request id: 94 finished, erase from queue
23:10:34:692 APS-DATA.request id: 97, addrmode: 0x01, addr: 0xfff0, profile: 0x0104, cluster: 0x0003, ep: 0x01 -> 0xFF queue: 0 len: 5 tx.options 0x00
23:10:34:707 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0003, lqi: 230, rssi: 0
23:10:34:707 FP indication 0x0104 / 0x0003 (0x00212EFFFF04239F / 0x0000)
23:10:34:707                       ...     (0x0000000000000000 / 0x0000)
23:10:34:773 APS-DATA.confirm id: 97, status: 0x00 SUCCESS
23:10:34:773 APS-DATA.confirm request id: 97 -> erase from queue
23:10:34:793 aps request id: 97 finished, erase from queue
23:10:35:033 Mgmt_Lqi_req zdpSeq: 138 to 0xD0CF5EFFFEC18742 start index 9
```

The module also has a QR code which reads the following string:

```
30S0170873E+Z8A8BDD1563665330E884CD6E2AEF902E+30PS3071-A215-FOH+2PDA01+S01504101035919
```

EnOcean provides a manual with some information what those fields exactly mean:

https://www.enocean.com/en/products/enocean_modules/ptm-215/

The modules manual can be found on FCC: https://www.fcc.gov/oet/ea/fccid (Grantee Code: "SZV", Product Code: "-PTM215Z").

The operational description also says: "The length of transmission is three pulses within 100ms, each being approximately 1ms long. Through manual operation radio transmissions of device
are fully random."

ebaauw, 2019-08-15:
> The device can be paired by pressing a button 10 seconds down (left switch down).

Which button you need to press actually depends on the ZigBee channel, see #1579.

> I do not get any events according to the deCONZ log.

The switch needs to be in direct range of the RaspBee or ConBee (for pairing as well as for operating); deCONZ doesn't yet configure the ZGP proxies (Hue lights) to forward the switch's ZGP commands over the ZigBee network.

agners, 2019-08-15:
> Which button you need to press actually depends on the ZigBee channel, see #1579.
Yes I realized that in between. It's A0 for me, channel 15.

> The switch needs to be in direct range of the RaspBee or ConBee (for pairing as well as for operating); deCONZ doesn't yet configure the ZGP proxies (Hue lights) to forward the switch's ZGP commands over the ZigBee network.

I am between 0.5 and 1m to the ConBee II. I never got a single message when pressing, even right after (presumably) successful pairing. It feels as if deCONZ pairs and rejects it immediately. Something seems to be wrong/different with those switches.

agners, 2019-08-15:
First of, A0 is left **up**, not what I previously stated... This confused me now for a while.

However, I found a strange behavior with deCONZ/those switches: When I pair a switch, then delete it, then try pairing a second time doesn't work.

I have two of those switches, when I pair the other switch, I then can pair the first switch again... It seems as if deCONZ doesn't allow pairing the same switch twice in a row. Anyway, when alternate with pairing/deleting I was successful in pairing both switches several times.

But I don't get any events from either of the two switches.

agners, 2019-09-30:
Is there an update to this switch? I do have a Philips Hue Hub 2 I could do some tests with. What could I do to help improving support for this switches?

eglia, 2019-10-05:
I'm having the same Problem, I can pair the Feller switch without any problems but then no events are received. The switch works fine on the Hue bridge.
I also tested a Hue Tap Switch, which works perfectly over deconz.

daendaen, 2019-10-11:
Me too, the same problem.

wvuyk, 2019-10-21:
Any progress here? Have been trying here too. Device joins, but no events are sent? Several users of my plugin are asking now

Wim

eglia, 2019-11-09:
I finally got my switch working! After pairing, you have to press the "bottom left" and "top right" keys at the same time. After that, all buttons (including long press) work as expected.

manup, 2019-11-09:
There is a new faq entry which describes the process:

https://phoscon.de/en/support#pairing-friends-of-hue-switch

CDRX2, 2019-11-18:
> The switch needs to be in direct range of the RaspBee or ConBee (for pairing as well as for operating); deCONZ doesn't yet configure the ZGP proxies (Hue lights) to forward the switch's ZGP commands over the ZigBee network.

Any chance of knowing whether forwarding ZGP commands over the network is on the roadmap? With luck, maybe there's even a timeline? Would love to move my RaspBee back to the basement and keep using my switches on the 1st floor ;-)

manup, 2019-11-18:
ETA's are tricky, but this is on the important todo list.

CDRX2, 2019-11-22:
@manup Thanks for the reply. Good enough for me, was not looking for a specific ETA.

closed 2020-03-28 by stale[bot]