dresden-elektronik/deconz-rest-plugin #1755: Contact sensor not available at rule list

alex2000-ha, 2019-08-16:
Hi
I connected sucessfully the aquara window/ door sensor. At phonscon sensors, they are shown as Xiaomi Sensor, open/close status is working. 
[image]

But I can not add them to the sensor control rule, because thy are not visible there.
[image]

Does anyone know how I can add them there?

thx

salopette, 2019-08-17:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1717

alex2000-ha, 2019-09-09:
firmware 2.05.69 ... no contact senor @rule list :(

closed 2020-01-14 by stale[bot]

mgrn0, 2020-04-28:
could this please be re-evaluated? it would really make sense for some people to have e.g. door sensors in rules.

Proche72, 2020-07-22:
Hi, is there any development on this?

SwoopX, 2020-07-23:
That is a question to be raised at the Phoscon repository, not here.

mgrn0, 2020-07-23:
@SwoopX I am confused since there was a lot discussion about this specific topic here, and not in the Phoscon repository. Can you point us in the right direction or should we open an identical issue there? is https://github.com/dresden-elektronik/phoscon-app-beta/issues the right place?

SwoopX, 2020-07-23:
Yes, the link you mention is the right place. DE has just recently re-opened the repo to keep the topics seperated.

mgrn0, 2020-07-23:
thanks @SwoopX , opened an issue there. (@Proche72 - you might want to jump in ;)

SwoopX, 2020-07-23:
Nope, since this is nothing I can help with :wink: