dresden-elektronik/deconz-rest-plugin #1757: Migrate to Docker?

FezVrasta, 2019-08-16:
My RPi 4 (4GB) is finally arrived and I'm getting ready to migrate my old Home Assistant + DeCONZ setup to it.

I want to use Docker, since right now everything is installed on the host system and it's a hot mess.

I was wondering what's the procedure to migrate my DeCONZ config to my new RPI, considering I'm going to use Docker this time? I would really like my to have to re-pair all my devices again.

manup, 2019-08-17:
Re-pairing isn't necessary. Basically you just need to transfer the zll.db file to the Docker environment. The easiest way is to create a backup file in the Phoscon App with your current setup and reload this configuration in the Phoscon App of your new Docker setup.

If something doesn't work right away please **don't** reset anything.

FezVrasta, 2019-08-18:
Thanks.

I'm uploading the backup through the web UI but it's been stuck on the "Loading backup - This can take a moment..." for quite a few minutes already.

What's the proper way to interrupt the process to try again?

FezVrasta, 2019-08-18:
This is what I see in the console:

[image]

FezVrasta, 2019-08-18:
I reloaded the page and tried again and I noticed this, I guess the backup restore process doesn't even start.

[image]

manup, 2019-08-18:
Hmm that's odd, I'll forward it to the developer of the backup page.
The 503 is however suspicious, is your setup working otherwise and the Zigbee network operational?

FezVrasta, 2019-08-18:
I have no idea, I haven't tried to use it yet, I don't want to pair stuff manually ever again

manup, 2019-08-18:
I mean is your Zigbee network working, are devices connected?

FezVrasta, 2019-08-18:
No, because I removed the RaspBee from my RPi2 and plugged it into the new RPi4 which is not yet configured.

But I just noticed the new RPi4 DeCONZ is not detecting the RaspBee, so I guess I first need to figure out why...

[image]

FezVrasta, 2019-08-18:
This is my docker-compose configuration:

```

  deconz:
    image: marthoc/deconz
    container_name: deconz
    restart: unless-stopped
    network_mode: host
    environment:
      - DECONZ_WEB_PORT=8080
      - TZ=Europe/Budapest
      - DECONZ_UPNP=0
    devices:
      - /dev/ttyAMA0
    volumes:
      - ./services/deconz/data/config:/root/.local/share/dresden-elektronic/deCONZ
```

manup, 2019-08-18:
Oh my bad just seeing the issue is about import (on the new machine I guess) not export.
I would say due how it looks the RaspBee isn't correctly installed yet and therefore "Not connected" is shown.

Have you configured the serial interface as described in the installation instructions?
https://phoscon.de/en/raspbee/install#docker

FezVrasta, 2019-08-18:
I'm using DietPi, during the initial setup it asked me if I wanted to keep the serial port enabled and I said yes, so I guess that's all I need?

manup, 2019-08-18:
I don't know about DietPi but the serial port should not be used for any system logging, as Raspbian does it per default. Does DietPi even provide the /dev/ttyAMA0 interface?

manup, 2019-08-18:
As a side note on Raspbian when the serial interface is configured properly `/dev/ttyS0` is used instead of `/dev/ttyAMA0`.

FezVrasta, 2019-08-18:
[image]

this is what I see

manup, 2019-08-18:
>this is what I see, so I think the ttyAMA0 and ttyS0 are okay

Not if the interface is still shared with the login shell, but I don't know if it is the case here. On Raspbian it needs to be explicitly disabled.

What is the output of `ls -l /dev/ttyAMA0`?

FezVrasta, 2019-08-18:
```
DietPi# ls -l /dev/ttyAMA0
crw--w---- 1 root tty 204, 64 Aug 18 11:43 /dev/ttyAMA0
```

manup, 2019-08-18:
That's interesting :)
So unless you are running deCONZ as root it can't work.

You need to place your user into the tty group and give the `/dev/ttyAMA0` group rw rights.

```
sudo gpasswd -a $USER tty
sudo chmod g+r /dev/ttyAMA0
```

Reboot afterwards. Still **if** the device is shared with the login shell it won't work.

FezVrasta, 2019-08-18:
I'm running everything as root since I'm using Docker to run the various services.

manup, 2019-08-18:
>I'm running everything as root since I'm using Docker to run the various services.

How do you start the container? Docker doesn't require to be run as root.

>I was reading this issue, is shell and console the same thing?
MichaIng/DietPi#343

I guess so, following looks interesting:
https://github.com/MichaIng/DietPi/issues/343#issuecomment-226946848

FezVrasta, 2019-08-18:
I setup a systemd service that starts my docker-compose config. It doesn't require root but I'm just using it for now to make the setup easier. Once that's done I'll try to run it with an unprivileged user and see how it goes.

FezVrasta, 2019-08-18:
Here's some log from the container...

```

deconz           | 12:04:44:464 dev /dev/ttyAMA0
deconz           | 12:04:45:055 device state timeout ignored in state 2
deconz           | 12:04:46:105 device state timeout ignored in state 2
deconz           | 12:04:47:156 device state timeout ignored in state 2
deconz           | 12:04:48:160 device state timeout ignored in state 2
deconz           | 12:04:49:160 device state timeout ignored in state 2
deconz           | 12:04:49:606 don't close database yet, keep open for 900 seconds
```

manup, 2019-08-18:
It seems that the serial communication doesn't work. My guess is the above mentioned issue.
You can also check which programs are accessing the device:

    sudo lsof /dev/ttyAMA0

Perhaps try to use `/dev/ttyS0` as alternative.

FezVrasta, 2019-08-18:
```
COMMAND   PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
hciattach 650 root    3u   CHR 204,64      0t0 1821 /dev/ttyAMA0
```

I'm starting to think it may be easier to use Raspbian 🙄

manup, 2019-08-18:
https://linux.die.net/man/8/hciattach

`hciattach` is related to the bluetooth service.

>I'm starting to think it may be easier to use Raspbian 🙄

I think it will work on DietPi but the setup requires extra steps which need to be figured out.

FezVrasta, 2019-08-18:
Ok so let me try to disable the BT for now (but I'll need to enable it later anyways)...

manup, 2019-08-18:
Bluetooth can be enabled, but it needs to be configured to use another serial interface.

FezVrasta, 2019-08-18:
Ok, for now it's disabled, at least we have to worry about one less thing, now I don't get anything back when I run `DietPi# lsof /dev/ttyAMA0`.

from `dietpi-config` I see the UART is enabled, but I kept the console off:
[image]

Still, DeCONZ can't detect the RaspBee 😩

manup, 2019-08-18:
Note you need to run `lsof` with sudo.

FezVrasta, 2019-08-18:
I'm not sure what I'm doing, but setting the container with DECONZ_DEVICE=/dev/ttyS0 now it detects RaspBee, but still it doesn't tell me anything about the firmware...

[image]

Would you consider this a progress? 😅

FezVrasta, 2019-08-18:
I'm not sure if it matters, but I can't even rename the gateway, it will always return a 403 error. Is that because it needs a working RaspBee connected or it's unrelated?

FezVrasta, 2019-08-18:
Oh dammit, I figured out I had a typo in the volume...

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1757#issuecomment-522306243

It should have been:

- ./services/deconz/data/config:/root/.local/share/dresden-elektronik/deCONZ

Now it shows the firmware correctly, but still I can't restore the backup with that 403 error I reported above...

FezVrasta, 2019-08-18:
Don't ask me how, but the configuration now appeared in the web app... 🤷‍♂️

FezVrasta, 2019-08-18:
I restarted Docker and I lost all the configuration 🙄 I don't yet know if it's my fault, but the backup import is now again not working properly.

manup, 2019-08-19:
>I restarted Docker and I lost all the configuration

How do you see all configuration is lost, are there no devices shown anymore?

Are you sure that the volume configuration is correct?
Your docker-compose configuration says:

```yaml
 deconz:
    image: marthoc/deconz
    container_name: deconz
    restart: unless-stopped
    network_mode: host
    environment:
      - DECONZ_WEB_PORT=8080
      - TZ=Europe/Budapest
      - DECONZ_UPNP=0
    devices:
      - /dev/ttyAMA0
    volumes:
      - ./services/deconz/data/config:/root/.local/share/dresden-elektronic/deCONZ
```

Shouldn't that be `/opt/docker-compose/services/deconz/data/config:/root/.local/share/dresden-elektronic/deCONZ`?

FezVrasta, 2019-08-19:
No, the volumes directive is correct, in fact I see files being written in the folder (I use the same format for all my containers).

I haven't restarted again, once I'll do so I'll see if it gets reset again.

Anyways, I'd keep this issue focused on the backup import feature which doesn't seem to work reliably. Is that okay to you?

FezVrasta, 2019-08-19:
I restarted the container and the config didn't get lost. So the only issue is the import backup feature.

closed 2019-12-24 by stale[bot]