dresden-elektronik/deconz-rest-plugin #1758: All Hue lights unreachable after coming back from holiday

imbaimbaimba, 2019-08-18:
Hi guys,
When I just got back from holiday, all my Hue bulbs are now unreachable in deCONZ. I checked the alt-click advanced page, and PANID, channel and network keys are all consistent - only ID and version changes between the 5 entries.

All my bulbs are hardwired, so I cannot really power cycle them. 

Any idea what is wrong here, and how I can fix it?

wvuyk, 2019-08-18:
Did you restart deCONZ, unscrew bulbs etc? Any way a restart sounds in place

imbaimbaimba, 2019-08-18:
> Did you restart deCONZ, unscrew bulbs etc? Any way a restart sounds in place

Restart did not work. Unscrewing bulbs did not work either. Even restarting my NAS did not work, and powering it down did not either. 

What I ended up doing is powering down the NAS, unplugging the ConBee stick for 5 minutes and then trying again, it then all came alive. So for future reference if someone experiences the same, that could be a solution.

closed 2019-08-18 by imbaimbaimba