dresden-elektronik/deconz-rest-plugin #1762: Cannot update RaspBee firmware

n2N8Z, 2019-08-19:
Raspberry Pi 4B
# grep Rev /proc/cpuinfo 
Revision        : c03111

# GCFFlasher_internal -l
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 

# systemctl stop deconz

# GCFFlasher_internal -f deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
deCONZ firmware version 26050500
action: reset device RaspBee
action: update firmware after 2 ms
could not sync with bootloader

I don't have any ZigBee devices yet, so can't confirm if the RaspBee works otherwise.
When deconz is running the little green light blinks.

Smanar, 2019-08-22:
Hello,
You have too, during connection at least, the Red led blinking or fixe ?
Have you checked serial interface setting ?
You are using headless version ? You are not using the GUI version ?
Are you sure about your power alimentation ?

xavi-dev, 2019-08-22:
Hello I'm getting the same message with the Raspberry pi 4 and the Raspbee hat.

I'm running docker deconz https://hub.docker.com/r/marthoc/deconz

I have configured the serial port (changed the bluetooth one) and the GUI is working fine. Once I run deCONZ software these messages appears on the log:

```
19:10:56:641 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
19:10:56:641 GW firmware version: 0x26050500
19:10:56:641 GW firmware version shall be updated to: 0x26330500
```
The light is always in red (only blinks 2 or 3 times in green when I turn on the Raspberry). I'm using the official power supply of 3A)

I've tried to run GCFFlasher from docker continer but I'm getting this error:

```
root@raspberrypi:/# GCFFlasher_internal -r
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
647: Error: failed to open device, status: Permission error while locking the device (2)
```

TheTimeWalker, 2019-08-22:
I can confirm that updating does not work on the Raspberry Pi 4 on my end too. After moving the module over to a spare Pi 3, update went without any issues. So it seems like there were some changes with the Pi 4 that blocks updating

n2N8Z, 2019-08-22:
> You have too, during connection at least, the Red led blinking or fixe ?
The red light is on, but doesn't blink.

> Have you checked serial interface setting ?
Yes, the serial port is configured as per the guide i.e. enabled with no login

> You are using headless version ? You are not using the GUI version ?
Headless.

> Are you sure about your power alimentation ?
I've also got the official power supply.

manup, 2019-08-23:
I can confirm the issue on Raspberry Pi 4. The problem is that the WiringPi version 2.50 which is shipped with Raspbian Buster is too old.

To fix the issue you need to manually install WiringPi 2.52 as described in
 http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/

```
cd /tmp
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
```

n2N8Z, 2019-08-23:
Upgrade to wiringpi 2.52 fixed it, thank you.

```
# GCFFlasher_internal -f deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_06 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
deCONZ firmware version 26050500
action: reset device RaspBee
action: update firmware after 2 ms

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
 (103 ms)
flashing 125684 bytes: |=============================|
verify: ....
SUCCESS
```

closed 2019-08-23 by n2N8Z

panhans, 2019-12-15:
Does anyone manged to install this using hass.io? I'll get the same error there!

n2N8Z, 2019-12-24:
Yes, I have 2.05.72 / 12/12/2019 installed via hass.io.