dresden-elektronik/deconz-rest-plugin #1766: Help.. "zigbee onoff controller" cannot connect

Allras82, 2019-08-21:
Hello

Anyone that can help.
I am not able to connect to mu zigbee onoff controller which is described here:
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1090](url)

I'm running a headless system at my RPI3 and for that reason i do not have the GUI access only web app.
I' using the system together with deCONZ-Domoticz.

Smanar, 2019-08-21:
Normally you don't need GUI for it. Wich one procedure have you used to pair it ? (Not sure it's the good notice, just find it on amazon)

> 1,Make sure the Zigbee OnOff Controller indicator is blinking. If the green LED indicator is on, press and hold the side button to make it blink. This time it is in the configuration state
> 2,If the Controller has been installed or you want to reconfigure it into the network, you can powered off and on for 5 times, This will make the Controller in the configuration state

Allras82, 2019-08-21:
Yes the Led flashes red... I have searched for new device both as bulb and switch, but it didn't show up!

Smanar, 2019-08-21:
It will be in light (bulb) section, so to go to "light" / "add new light" and try a power cycle.

Allras82, 2019-08-26:
I understand that i can choose to either use the reset button in the side og the onoff controller and hold it until LED start flashing red. or i can toggle power 5 timers. (i have tried both but so far not succeeded in the power toggel)

After watching the video i'm confused, do I need to do both? (holde the reset and afterwards toggel power?)

Smanar, 2019-08-26:
Lol, It's not realy clear, and I don't have the device to make tests, in the description I don't understadn the difference between "This time it is in the configuration state" and "This will make the Controller in the configuration state".
Try both, set the conbee in permit mode and try the both method, up you see something moving.

With headless deconz, you haven't GUI but perhaps you can enable some logs to check pairing phase ?

Allras82, 2019-09-04:
I Now think i got it working. Iv found an additional lamp in the deconcz.
Now my question is more related to the domoticz-deconz plugin.

The new lamp does not show up in the deconz-domoticz interface, why is that?

I can control it by adding the lamp to a group in deconz and then i can control the group in domoticz.

(sorry if this is the wrong plade to ask qusitons for the deconz-domoticz plugin!)

Smanar, 2019-09-04:
Lol, yep not the better place for that, you can use the domoticz forum or the other github.
Depend of the plugin version you have, but if you restart the plugin, it will update your device list with it, or at least print an error message for me (enable "debug info only')

But, by curiosity what is the trick to make it working ?

Allras82, 2019-09-04:
I didn't understand the procedure for putting the device into config mode. But finally I found out the I should push and hold the button at the side until light start flashing and immediately release. Before I was pushing until the red light were flashing...

Now it just found I right away😃

Thanks for your support😃

Smanar, 2019-09-04:
Lol, good to know, but instructions aren't realy clear.

BTW, tell me if you still have problem with this device and the domoticz plugin, from comments it will work on API, so it will work on domoticz too, else I need to correct something.

Allras82, 2019-09-04:
It still didn't show up in domoticz.
Iv just put the question in domoticz forum under bug and problems... I thought that was the right place for the specific domoticz topics...

https://www.domoticz.com/forum/viewtopic.php?f=6&t=29271

closed 2020-01-09 by stale[bot]