dresden-elektronik/deconz-rest-plugin #1767: IKEA lights (Conbee II) shows wrong state

martinkrulltott, 2019-08-21:
Hi guys,
I’m new to the home automation world and I’ve just set up a bunch of IKEA trådfri lights and switches using a Conbee II, raspberry pi (Raspbian Stretch), Deconz and the home assistant docker image.
My automations are working fine and I can control all my lights and switches. However from time to time the lights and switches are showing up in the wrong state.
Both today and yesterday all items are shown as on, even though they’re clearly off. Turning an item “off” doesn’t do anything (its already off), but turning it “on” again works and the item is back to working as normal and I can turn it off. Then a while later they might suddenly be back to reporting as “on” even though it’s still actually off and neither me nor the automations has ever set it to “on”.

So there seems to be a glitch with how the state is being reported. I'm primarily using Home Assistant but the Phoscon-GW UI is also reporting the same incorrect state, so I'm assuming that the issue lies here.

Has anyone else experienced these “fake on” states before?

Smanar, 2019-08-22:
Hello, do you have the ikea motion sensor ?

martinkrulltott, 2019-08-22:
@Smanar yes I do, but it's only connected to one specific lamp at night. During the day it shouldn't do anything.

Smanar, 2019-08-22:
Pls take a look here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676

closed 2019-12-27 by stale[bot]

boekabart, 2020-12-16:
Just letting everyone know that this is happening for me as well, for a very specific set of lights (which are not in 1 group) all the time. State reported as 'on' while they are off. 
The main problem is that deconz is being SMART and ignoring subsequent on commands... so switches don't work until you "turn them off and on again"

V 2.07.00 / 11/15/2020
26580700