dresden-elektronik/deconz-rest-plugin #1770: Type "On/Off light" is power plug?

Kane610, 2019-08-22:
So I got a report that a device identified as a light should actually be a power plug

Is type "On/Off light" only for power plugs?

{
                  'etag': '55d2bcea384ca9d521cea01dca55d6e5',
                  'hascolor': False,
                  'manufacturername': 'Jasco Products',
                  'modelid': '45856',
                  'name': 'Light 3',
                  'state': {'alert': 'none', 'on': False, 'reachable': True},
                  'swversion': '00000006',
                  'type': 'On/Off light',
                  'uniqueid': '00:22:a3:00:00:1c:ae:d6-01'
}

ebaauw, 2019-08-22:
No, "On/Off light" is for lights that only support On/Off.  Power plugs should be "On/Off plug-in unit" or "Smart plug".  Not all devices advertise the correct device type in their descriptor, though.

Kane610, 2019-08-22:
Are there any workarounds for this?

ebaauw, 2019-08-22:
In theory: yes.  We would have to whitelist the devices for which we would overwrite the advertised device type.  Currently, deCONZ doesn't whitelist lights.  I wouldn't want to introduce it before refactoring for API v2 and providing a better way to handle device capabilities (see #1688).

For homebridge-hue, I let the user specify which `/lights` (and `/groups`) resources should be exposed as outlet instead of lightbulb using a `/resourcelinks` resource.  Any logic I tried, based on device type, state attributes, ..., failed at some time.

Kane610, 2019-08-22:
Allright, thanks for your quick response @ebaauw

closed 2019-12-27 by stale[bot]