dresden-elektronik/deconz-rest-plugin #1771: Devices not 'meshing'?

chierichetto, 2019-08-22:
I'm using Deconz With a Conbee Stick 1 thorugh the Home Assistant Addon.
After i discovered the graphical user interface that let's me see the network, i see that none of the devices are meshing.

The Conbee stick is placed in the basement, and some sensors and lights are 2 stories up.

I also experience some devices become unavailable at random and need to be repaired.

How can i get meshing to work? And why don't i see any lines to any sensors?

[image]

ebaauw, 2019-08-22:
Only routers (yellow nodes) form the mesh network.  End devices (grey nodes) connect to a single "parent" router.  The lines are drawn when deCONZ reads the neighbour table of a node.  It might take up to 30 minutes after startup before all routers are polled.  End devices are asleep most of the time so it might take considerable longer before they respond.  If at all, some end-devices like Xioami don't respond at all.  I think they disabled polling of IKEA lights, because these lights didn't like that and would eventually hang.

Long story short: don't rely too much on the lines to check for meshing.

oleggold, 2019-08-28:
I have the same issue.
My routers are CC2531 flashed with the router firmware. I only use them so I can get Xiaomi sensors that are positioned far away from the Conbee II dongle to work.
The sensors seem to work, but the mesh network does not show any connected lines.
Also, the routers do not show any connection lines between them (no mesh).
Please let me know if you found the answer to the issue.

Here are my current map:
[image]

Smanar, 2019-08-29:
For me the @chierichetto is realy strange, for me it's impossible having ALL router with only 1 connection.
But for your situation, it's not same IDK for CC2531 bur for CC2530, connection line are invisible > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1428

skeptizzle, 2019-08-31:
I have this issue as well. The only router meshing is an innr SP 120 plug. 

[image]

sergiocaridi, 2019-09-01:
I have the same something worse.
Sunricher, those with black mark, are working as router.
The others (TOMA) do not mesh each other.
Some of are isolated because they are to far from the working routers.

[image]

skeptizzle, 2019-09-01:
I ended up trying to separate the Zigbee and Wi-Fi channels as much as possible Zigbee on ch 25 and Wi-Fi on 1 and 6.
It got better and now looks like this:
[image]

Smanar, 2019-09-01:
Haaa, ok so was for wifi perturbation ? You are using wifi on the same machije you have the zigbee gateway ? have you tried an usb extension cable, to take away the gateway ?

skeptizzle, 2019-09-01:
Nope. I use UniFi gear, no Wi-Fi on the NUC running HA and deCONZ.
I have USB extension cables on the way together with a ConBee II (Can't update firmware on my original ConBee).

sergiocaridi, 2019-09-01:
Thanks 
I'll try to:
- separate channels (wifi &zigbee) 
- use USB extension cables

but it looks like the problem is related to the devices. 
The working router are Sunrichter (Philips) all the others are Shenzen Toma 1012.

oleggold, 2019-09-01:
i am already using channel 25 for the gateway and channel 1 for my wifi.
I don't have any connecting lines between any of my CR2531 routers or Xiaomi endpoints unless the endpoint is connected directly to the controller.
I am not sure what else to try and would appreciate a Deconz engineer's response. 

Here is my current Deconz Map:
[image]

Please note that the Signal Monitor shows the LQI for Master Bedroom Router at 167 and LQI on the map for the same is equal to 84. That makes me believe that a connection exists between Main Hall Router East and Master Bdrm Router, but it is not depicted on the map.

Smanar, 2019-09-01:
When using CR2530, connections are invisible, but all working, I think it's same for CR2531
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1428

oleggold, 2019-09-01:
I think you are right @Smanar.
I am trying to identify and remove the routers I don't need. If the mesh map is not going to help, does anyone know what else I can do to identify what my endpoints and routers connections are?

skeptizzle, 2019-09-02:
I have some news, even though i can't tell if it's the change from the original Conbee to a Conbee 2 or if it's the 15cm extension i put on. Anyway my mesh seams to be a lot more active now and a lot more routes.
[image]

oleggold, 2019-09-02:
@skeptizzle, glad you feel better about you mesh and it does look good.

Could you please answer the following questions?
1. What are your devices that act as routers?
2. Do you have any Xiaomi endpoints connected?
3. Are Garageder, MultiX Badevaerelse, 0x1F0E endpoints working (they do not have any connection lines)?

Thank you.

skeptizzle, 2019-09-02:
Hi. @oleggold 

Sure thing!

1. My end devices are a mix of Ikea bulbs, plug and panels, i have a single innr SP 120 plug.
2. All my sensors are Xiaomi and they are working (connected) through deCONZ.
3. Garagedør, MultiX Badeværelse are reporting and working.
  - I have NO clue what 0x1F0E, lumi.sensor_switch 6 and Presence 49 is... 😃 

You are welcome. 😉

oleggold, 2019-09-02:
@skeptizzle, thank you very much for your response.

All my Xiaomi sensors are working as well, but no connection lines between my CR2531 routers and Xiaomi sensors or between CR2531 routers. At one point I tried Ikea plugs as routers but I had the same issue and that made me believe that the problem with mesh mapping is related to Xiaomi devices.
But you mesh shows Xiaomi devices connected to Ikea routers, so my conclusion was not correct.
Do you know if there is some other way I can determine what my mesh looks like?

Smanar, 2019-09-02:
@oleggold you haven't the "green lines" but your network have problems too ?
On your screen all your devices are connected and working, what is the problem ?

skeptizzle, 2019-09-02:
> Do you know if there is some other way I can determine what my mesh looks like?

I'm sorry but I have no idea. But if your network is working, don't worry to much about the software representation of the lines..

Cheers!

sergiocaridi, 2019-09-06:
here is an update.

[image]
the devices are the same used 5 days ago, same problem.

The device named "2w" in the corner is supposed to be isolated as confirmed by the App

[image]

Both deCONZ and the app do not see the device, but I'm still able to comand the "2w" device selecting "all". It means that the router are meshing, ins't it?

How can I solve the problem?

Smanar, 2019-09-06:
Perhaps not meshing, but the node is grayed on deconz, too, like for real unreachable device, but if the device respond to group request, its not something we can call "unreachable".

There is some others isssue with same bug, but no solutions yet.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1337
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/944

oleggold, 2019-09-06:
> @oleggold you haven't the "green lines" but your network have problems too ?
> On your screen all your devices are connected and working, what is the problem ?

My routers are CR2531, I have them connected to the outlets around the house and it they don' look very good. I want to remove the ones I don't need and I thought the Deconz mesh will help me with that.

sergiocaridi, 2019-09-09:
Hi. 

I tried to use a different pc and I have a different result. Now the devices I've used yesterday are meshing.

[image]

Is there a way to cear completely the pc. I tried to install the deCONZ and install it again but I have this error.

[image]

Smanar, 2019-09-09:
It's not because you have a better zigbee connection on the PC than the first machine ?

For your error message, I haven't find infos about your lib (ssleay32.dll) but for ICLSClinet > https://www.shouldiremoveit.com/iCLS-Client-140629-program.aspx

xbmcnut, 2019-09-11:
I have a similar issue. All routers are Xiaomi plugs, and all other sensors are Xiaomi too. Running deCONZ 2.05.69 under Hassio with Conbee II. Zigbee channel is 15 with closest UniFi AP on channel 1 but when I say close, it's less than 2m away from the Conbee. Everything works flawlessly though with the exception that device state syncing between the Xiaomi plugs and Conbee II is often out of sync. Turn the switch on and off and it resyncs for a while then it's out again a day later. 

Does that map look OK for a Xiaomi/Conbee II connection?
[image]

manup, 2019-09-11:
This is the expected picture.

Background: We don't query the neighbor tables of Xiaomi routers since this seems to cause stability problems where they go offline. Since this is just a visual thing it doesn't have any impact on functionality, **the mesh is there** but the lines won't be displayed.

xbmcnut, 2019-09-11:
Really good to know @manup, thank you. Should I log the Xiaomi plug syncing issue as a new item?

sergiocaridi, 2019-09-11:
Please help me!

last week I did the inclusion for the 32devices I've been writing about.
today I'm back and I've  restored the backup and... 
The controller is not able to see the devices anymore. 
If I execute the onoffx4 cicle and I search for lams, they come back in the sistem.
Who can explain to me where I'm wrong?

sergiocaridi, 2019-09-11:
how can I backup and restore different projects?

Smanar, 2019-09-11:
@sergiocaridi are you sure your 32th device was on your previous backup ?

sergiocaridi, 2019-09-11:
thank you 
yes I'm sure, here is what I did.
I opened the app, loaded the last backup. 
All the lamps included last time where in the project.
I powered the lamps but none of them where able to connect.
If I do the inclusion procedure on/off for 4 times and I scan for new ligths the lamps are back in comunication whith the same name, group ...
I do need to understand cause I have 4 building with 80 devices each

Smanar, 2019-09-11:
Ha ok, you have loose all your routers ? I haven't understand.
Was the same hardware and the same conbee ?
For me your procedure is good (But I m not a specialist), a thing you can try, just re-include some bulbs and take a look in the hidden zigbee configuration > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1849#issuecomment-530287817
Check if you haven't a difference beetween working and unworking device.

sergiocaridi, 2019-09-11:
2.05.69

Smanar, 2019-09-11:
And there is no change in the " hidden zigbee configuration" ?

BTW, the others people on the other issue have similar issue than you.

sergiocaridi, 2019-09-15:
hidden zigbee configuration?
what does it mean.
By the way it happened again.
this time using a raspberry + conbee

Yesterday Iinstalled deconz on a sdcard and, using a brand new  conbee
pen,  .I included the first lamp.
This morning the lamp is no more reachable.
I tried to include a second lamp and it works.
15 minutes ago I started the Signal Monitor plug in selecting the missing
in action lamp. 8 packets

sergiocaridi, 2019-09-15:
Maybe it is relevant
the problem appears when I move out of the network. In the sense that the coordinator is off line for hours. 
Here is the situation

[image]

sergiocaridi, 2019-09-15:
[image]

sergiocaridi, 2019-09-15:
sorry I've seen #1185 "All devices in phoscon are not reachable"
same problem I'll move to the other thread.
Thanks

manup, 2019-09-15:
>Maybe it is relevant
the problem appears when I move out of the network. In the sense that the coordinator is off line for hours.

I don't know these particular lights, normally it should be no problem when the coordinator is offline (even for days or weeks). But when I remember correctly there were a devices which would drop off the network in that case (I don't remember the brand).

Does the light come back only when you power-cycle it?

sergiocaridi, 2019-09-16:
thanks manup.
 the problem in relatet to the nvram of conbee. 
At the moment the solution is to access to the " hidden zigbee configuration" and restore the previus config.

Now I'd like to understand why some devices looks like they are not routing.
In the image L01cand L02c have a different FW from old1c and old2 c.

For exmple the L01c has only one connection with the coordinator

[image]

sergiocaridi, 2019-10-02:
Hi here is a knew problem. 
I have a system with more than 100 FFD (RGBW ballast) use for architectural lighting. 
At the very beginning I planned to have one system, nowaday I think in a different way,
I have 2 groups  and as a matter of facts the 2 groups are not in communication due to wall and distance. In the image all the devices connected to the controller. If I move to the other facade I can reach the greup B.
At the very beginning I planned to have one system, nowaday I think in a different way.

I'd like to clone the conbee II  and then delete from Conbee A the  the devices of group B  and from Combee B the devices of group A. 

Is this possible, do you have any suggestion?

What if I delete devices from the App while the router (ballast) is offline?
Is this operation usefull to speed for the communication and performances.

sergiocaridi, 2019-10-02:
[image]

Smanar, 2019-10-03:
I think it will be a little more complicated, you can't have 2 networks with same Network ID, same network key, ect ...

sergiocaridi, 2019-10-03:
even if they are not connected there are at list 40 meters in between

sergiocaridi, 2019-10-03:
I suppose that if the controller has so many lamps that are not reacheble would spend resouces in trying to connect with theese devices.

AdamWeglarz, 2019-12-07:
Hi,

I have the same issue.
All nodes are working correctly.
And whats more some nodes are missing in the list in GUI. They are available in Phoscon app and properly reported to other systems (I use domoticz)
**
[image]
**

Smanar, 2019-12-07:
@AdamWeglarz you realy have so much end-device for realy few routers.
IDK wich one repeater you have but on some of them connexion (green line) are not visible. I think it's for that you have sone device working but without connexion line.

closed 2020-04-12 by stale[bot]

mylesewen, 2020-07-25:
I too had the problem of devices not meshing after I migrated from a Raspberry Pi 4 based system to a PC running linux.  After a day or so of investigation and trying various things I stumbled upon a few posts which suggested the PC might be shielding or interfering with the Zigbee signal.  They suggested using a short male to female USB extension cable for the Conbee II.  I tried that and within about 5 minutes I started to see the mesh automatically rebuild.

ricricucit, 2020-12-10:
For those who will pass by here: for me, it helped A LOT, to connect things (gray nodes), while physically close to their target routers (yellow nodes).