dresden-elektronik/deconz-rest-plugin #1773: different nwk address 0x13E9 / 0x13EA

SimplyGardner, 2019-08-23:
Does this look normal? Specifically, the `different nwk address 0x13E9 / 0x13EA`

```
23:57:38:679 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:57:38:679 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:57:38:689 no button map for: no lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:57:38:689 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:57:44:390     * different nwk address 0x13E9 / 0x13EA
23:57:45:426 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:57:45:426 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:57:45:431 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:57:45:431 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:57:50:631     * different nwk address 0x13E9 / 0x13EA
23:57:56:868     * different nwk address 0x13E9 / 0x13EA
23:58:00:369 Current channel 25
23:58:00:378 Device TTL 1430 s flags: 0x7
23:58:03:100     * different nwk address 0x13E9 / 0x13EA
23:58:03:154 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:58:03:154 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:58:03:160 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:58:03:160 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:58:05:572 Announced to internet
23:58:05:573 discovery server date: Fri, 23 Aug 2019 22:58:05 GMT
23:58:05:573 	 local time seems to be ok
23:58:09:353     * different nwk address 0x13E9 / 0x13EA
23:58:15:588     * different nwk address 0x13E9 / 0x13EA
23:58:21:831     * different nwk address 0x13E9 / 0x13EA
23:58:28:071     * different nwk address 0x13E9 / 0x13EA
23:58:34:310     * different nwk address 0x13E9 / 0x13EA
23:58:40:549     * different nwk address 0x13E9 / 0x13EA
23:58:42:677 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:58:42:677 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:58:42:684 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:58:42:684 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:58:46:790     * different nwk address 0x13E9 / 0x13EA
23:58:48:329 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:58:48:329 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:58:48:335 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:58:48:335 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:58:53:029     * different nwk address 0x13E9 / 0x13EA
23:58:56:593 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:58:56:593 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:58:56:600 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:58:56:600 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:58:59:271     * different nwk address 0x13E9 / 0x13EA
23:59:00:369 Current channel 25
23:59:00:380 Device TTL 1370 s flags: 0x7
23:59:05:511     * different nwk address 0x13E9 / 0x13EA
23:59:10:882 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:10:882 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:10:892 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:10:892 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:11:752     * different nwk address 0x13E9 / 0x13EA
23:59:17:993     * different nwk address 0x13E9 / 0x13EA
23:59:23:086 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:23:086 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:23:096 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:23:096 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:24:231     * different nwk address 0x13E9 / 0x13EA
23:59:28:799 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:28:799 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:28:803 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:28:803 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:30:461     * different nwk address 0x13E9 / 0x13EA
23:59:35:244 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:35:244 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:35:254 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:35:254 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:36:712     * different nwk address 0x13E9 / 0x13EA
23:59:42:494 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:42:494 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:42:500 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:42:500 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:42:950     * different nwk address 0x13E9 / 0x13EA
23:59:48:035 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:48:035 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:48:045 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:48:045 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:49:190     * different nwk address 0x13E9 / 0x13EA
23:59:53:794 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
23:59:53:794 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
23:59:53:799 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
23:59:53:799 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
23:59:55:430     * different nwk address 0x13E9 / 0x13EA
00:00:00:370 Current channel 25
00:00:00:379 Device TTL 1310 s flags: 0x7
00:00:01:668     * different nwk address 0x13E9 / 0x13EA
00:00:02:260 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
00:00:02:260 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
00:00:02:270 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
00:00:02:270 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
00:00:07:489 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
00:00:07:489 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
00:00:07:493 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
00:00:07:493 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
00:00:07:900     * different nwk address 0x13E9 / 0x13EA
00:00:10:389 dev /dev/ttyAMA0
00:00:10:389 GW firmware version: 0x264a0700
00:00:10:389 GW firmware version is up to date: 0x264a0700
00:00:13:154 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
00:00:13:154 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
00:00:13:164 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
00:00:13:164 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01 
```

SimplyGardner, 2019-08-27:
updated with more details

Krocko, 2019-09-17:
Did you find a solution?
I have the same problem.
Help @manup.

SimplyGardner, 2019-09-18:
Nope @Krocko - still no solution.

I have also found my Deconz runs at 5% CPU. Seems like something is constantly doing something.

I only have a couple of Xiaomi Sensors.

Krocko, 2019-09-18:
But the sensors are working with deCONZ?

Can it be the usb extension cable?
I tried 3 different extension cable to get the ConBee II working.

I really hope @manup will help us.

manup, 2019-09-18:
The "different nwk address 0xB317 / 0xB318" message is just an information that a NWK address change of a device has been detected. This can happen for some reasons but should cause no harm.

The 5% percent CPU usage is likely related to the GUI, so if you run deCONZ headless or just minimise the deCONZ window the CPU usage should go down a bit.

SimplyGardner, 2019-09-23:
@manup - thanks for your reply.

It seems to be constantly reporting the network address change - many many times per minute. Here's a recent pull from he logs.

```20:48:57:683 	03 temperature 32 °C
20:48:57:684 	04 unknown 5032 (0x13A8)
20:48:57:684 	05 RSSI dB (?) 30 (0x001E)
20:48:57:684 	06 LQI (?) 4294967296 (0x0100000000)
20:48:57:684 	0a unknown 0 (0x0000)
20:48:57:684 	64 on/off 0
20:49:01:010     * different nwk address 0xD437 / 0xD438
20:49:11:568     * different nwk address 0xD437 / 0xD438
20:49:18:531 Current channel 25
20:49:18:539 Device TTL 4558 s flags: 0x7
20:49:22:129     * different nwk address 0xD437 / 0xD438
20:49:32:687     * different nwk address 0xD437 / 0xD438
20:49:43:249     * different nwk address 0xD437 / 0xD438
20:49:53:807     * different nwk address 0xD437 / 0xD438
20:50:04:359     * different nwk address 0xD437 / 0xD438
20:50:14:919     * different nwk address 0xD437 / 0xD438
20:50:18:530 Current channel 25
20:50:18:538 Device TTL 4498 s flags: 0x7
20:50:25:479     * different nwk address 0xD437 / 0xD438
20:50:36:048     * different nwk address 0xD437 / 0xD438
20:50:46:597     * different nwk address 0xD437 / 0xD438
20:50:48:616 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
20:50:48:616 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0000, ep 0x01
20:50:48:616 0x00158D0002E8CDD6 extract Xiaomi special attribute 0xFF01
20:50:48:616 	01 battery 3025 (0x0BD1)
20:50:48:616 	03 temperature 29 °C
20:50:48:616 	04 unknown 17320 (0x43A8)
20:50:48:616 	05 RSSI dB (?) 32 (0x0020)
20:50:48:616 	06 LQI (?) 64424509440 (0x0F00000000)
20:50:48:616 	0a unknown 15435 (0x3C4B)
20:50:48:616 	64 on/off 0
20:50:48:616 	0b lightlevel 0 (0x0000)
20:50:57:167     * different nwk address 0xD437 / 0xD438
20:51:07:726     * different nwk address 0xD437 / 0xD438
20:51:18:279     * different nwk address 0xD437 / 0xD438
20:51:18:521 Current channel 25
20:51:18:523 Device TTL 4438 s flags: 0x7
20:51:23:558 dev /dev/ttyAMA0
20:51:23:558 GW firmware version: 0x264a0700
20:51:23:558 GW firmware version is up to date: 0x264a0700
20:51:28:849     * different nwk address 0xD437 / 0xD438
20:51:39:409     * different nwk address 0xD437 / 0xD438
20:51:49:959     * different nwk address 0xD437 / 0xD438
20:52:00:528     * different nwk address 0xD437 / 0xD438
20:52:11:088     * different nwk address 0xD437 / 0xD438
20:52:18:530 Current channel 25
20:52:18:538 Device TTL 4378 s flags: 0x7
20:52:21:649     * different nwk address 0xD437 / 0xD438
20:52:32:198     * different nwk address 0xD437 / 0xD438
20:52:42:758     * different nwk address 0xD437 / 0xD438
20:52:51:190 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
20:52:51:190 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
20:52:51:204 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
20:52:51:204 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
20:52:52:015 Set sensor check interval to 100 milliseconds
20:52:53:327     * different nwk address 0xD437 / 0xD438
20:52:59:792 ZCL attribute report 0x0017880110481E50 for cluster 0x0001, ep 0x02
20:53:03:888     * different nwk address 0xD437 / 0xD438
20:53:14:448     * different nwk address 0xD437 / 0xD438
20:53:18:532 Current channel 25
20:53:18:540 Device TTL 4318 s flags: 0x7
20:53:24:998     * different nwk address 0xD437 / 0xD438
20:53:35:560     * different nwk address 0xD437 / 0xD438
20:53:46:127     * different nwk address 0xD437 / 0xD438
20:53:53:415 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
20:53:53:415 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
20:53:53:429 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
20:53:53:429 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
20:53:56:686     * different nwk address 0xD437 / 0xD438
20:54:07:239     * different nwk address 0xD437 / 0xD438
20:54:17:809     * different nwk address 0xD437 / 0xD438
20:54:18:534 Current channel 25
20:54:18:542 Device TTL 4258 s flags: 0x7
20:54:23:949 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
20:54:23:949 ZCL attribute report 0x00158D0002E98F53 for cluster 0x0402, ep 0x01
20:54:23:957 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
20:54:23:957 ZCL attribute report 0x00158D0002E98F53 for cluster 0x0405, ep 0x01
20:54:23:959 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
20:54:23:959 ZCL attribute report 0x00158D0002E98F53 for cluster 0x0403, ep 0x01
20:54:28:368     * different nwk address 0xD437 / 0xD438
20:54:39:400     * different nwk address 0xD437 / 0xD438
20:54:50:448     * different nwk address 0xD437 / 0xD438
20:55:01:489     * different nwk address 0xD437 / 0xD438
20:55:12:543     * different nwk address 0xD437 / 0xD438
20:55:12:592 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
20:55:12:592 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0400, ep 0x01
20:55:12:601 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
20:55:12:601 ZCL attribute report 0x00158D0002E8CDD6 for cluster 0x0406, ep 0x01
20:55:18:531 Current channel 25
20:55:18:539 Device TTL 4198 s flags: 0x7
20:55:19:168 Announced to internet
20:55:19:168 discovery server date: Mon, 23 Sep 2019 19:55:19 GMT
20:55:19:168 	 local time seems to be ok
20:55:23:539 dev /dev/ttyAMA0
20:55:23:539 GW firmware version: 0x264a0700
20:55:23:539 GW firmware version is up to date: 0x264a0700
20:55:23:573     * different nwk address 0xD437 / 0xD438
20:55:27:938 New websocket 192.168.1.6:56800 (state: 3) 
20:55:34:128     * different nwk address 0xD437 / 0xD438
20:55:44:678     * different nwk address 0xD437 / 0xD438```

SimplyGardner, 2019-09-23:
```22:19:55:816     * different nwk address 0x924B / 0x924C
22:20:06:368     * different nwk address 0x924B / 0x924C
22:20:16:935     * different nwk address 0x924B / 0x924C
22:20:18:520 Current channel 25
22:20:18:522 Device TTL 5700 s flags: 0x7
22:20:27:486     * different nwk address 0x924B / 0x924C
22:20:38:056     * different nwk address 0x924B / 0x924C
22:20:48:617     * different nwk address 0x924B / 0x924C
22:20:59:177     * different nwk address 0x924B / 0x924C
22:21:09:729     * different nwk address 0x924B / 0x924C
22:21:18:532 Current channel 25
22:21:18:540 Device TTL 5639 s flags: 0x7
22:21:20:287     * different nwk address 0x924B / 0x924C
22:21:30:855     * different nwk address 0x924B / 0x924C
22:21:41:407     * different nwk address 0x924B / 0x924C
22:21:51:966     * different nwk address 0x924B / 0x924C
22:22:02:537     * different nwk address 0x924B / 0x924C
22:22:13:098     * different nwk address 0x924B / 0x924C
22:22:18:523 Current channel 25
22:22:18:525 Device TTL 5579 s flags: 0x7
22:22:23:656     * different nwk address 0x924B / 0x924C
22:22:34:214     * different nwk address 0x924B / 0x924C
22:22:44:766     * different nwk address 0x924B / 0x924C
22:22:55:336     * different nwk address 0x924B / 0x924C
22:23:05:887     * different nwk address 0x924B / 0x924C
22:23:16:448     * different nwk address 0x924B / 0x924C
22:23:18:531 Current channel 25
22:23:18:539 Device TTL 5519 s flags: 0x7
22:23:23:560 dev /dev/ttyAMA0
22:23:23:560 GW firmware version: 0x264a0700
22:23:23:560 GW firmware version is up to date: 0x264a0700
22:23:24:260 ZCL attribute report 0x0017880110481E50 for cluster 0x0001, ep 0x02
22:23:27:016     * different nwk address 0x924B / 0x924C
22:23:35:175     * different nwk address 0x924B / 0x924C
22:23:45:734     * different nwk address 0x924B / 0x924C
22:23:56:295     * different nwk address 0x924B / 0x924C
22:24:06:845     * different nwk address 0x924B / 0x924C
22:24:17:406     * different nwk address 0x924B / 0x924C
22:24:18:530 Current channel 25
22:24:18:538 Device TTL 5459 s flags: 0x7
22:24:27:975     * different nwk address 0x924B / 0x924C
22:24:38:538     * different nwk address 0x924B / 0x924C
22:24:49:096     * different nwk address 0x924B / 0x924C
22:24:59:656     * different nwk address 0x924B / 0x924C
22:25:10:217     * different nwk address 0x924B / 0x924C
22:25:18:532 Current channel 25
22:25:18:540 Device TTL 5399 s flags: 0x7
22:25:18:645 discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable
22:25:20:777     * different nwk address 0x924B / 0x924C
22:25:31:348     * different nwk address 0x924B / 0x924C
22:25:41:885     * different nwk address 0x924B / 0x924C
22:25:52:456     * different nwk address 0x924B / 0x924C
22:26:03:008     * different nwk address 0x924B / 0x924C
22:26:13:576     * different nwk address 0x924B / 0x924C
22:26:18:521 Current channel 25
22:26:18:523 Device TTL 5339 s flags: 0x7
22:26:24:135     * different nwk address 0x924B / 0x924C
22:26:30:264 0x001788010247EB91 error APSDE-DATA.confirm: 0xE9 on task
22:26:34:696     * different nwk address 0x924B / 0x924C
22:26:45:255     * different nwk address 0x924B / 0x924C
22:26:55:816     * different nwk address 0x924B / 0x924C
22:27:06:378     * different nwk address 0x924B / 0x924C
22:27:16:936     * different nwk address 0x924B / 0x924C```

manup, 2019-09-24:
I see looks a bit busy. In the core application is a tick handler which is processing maintenance tasks, depending on the network size the message is shown more often. I think the code can be improved to detect and print the nwk address change only once.

SimplyGardner, 2019-09-25:
Thanks @manup - is there anything I should do or is this an improvement on the central code?

andras-tim, 2019-10-05:
**I hadn't problems while I bought the dongle (3 months ago), but I had this issue yesterday morning.**

FYI: my network is made of

* 25+ sensors (so, 25+ endpoints)
* 2 IKEA repeaters, 1 IKEA LED lamp (so, 3 routers)

A day before yesterday, the 3 routers + the dongle created a full-graph and the sensors were connected to the closest router (almost...). But on yesterday morning one IKEA repeater and the IKEA LED lamp are gray on the map, and all child devices.

The log contains some errors like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1773#issuecomment-534293594

	01:09:52:352     * different nwk address 0x778F / 0x7790
	01:09:52:352     * different nwk address 0x9100 / 0x9101
	01:09:52:830     * different nwk address 0x2DF5 / 0x2DF6
	01:09:53:296 discovery for zombie 0x000d6ffffe9a1078 dropped, last try was 372 s ago
	01:09:53:776 discovery for zombie 0x90fd9ffffe13e3c3 dropped, last try was 336 s ago
	01:10:03:855 discovery for zombie 0x000d6ffffe9a1078 dropped, last try was 383 s ago
	01:10:04:335 discovery for zombie 0x90fd9ffffe13e3c3 dropped, last try was 346 s ago
	01:10:05:816 Current channel 11
	01:10:05:824 Device TTL 2399 s flags: 0x7
	01:10:05:960 discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable
	01:10:10:834 GW firmware version: 0x264a0700
	01:10:10:835 GW firmware version is up to date: 0x264a0700
	01:10:14:384 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
	01:10:14:384 ZCL attribute report 0x00158D0004141BBF for cluster 0x0000, ep 0x01
	01:10:14:384 0x00158D0004141BBF extract Xiaomi special attribute 0xFF01
	01:10:14:384 	01 battery 3045 (0x0BE5)
	01:10:14:384 	03 temperature 27 °C
	01:10:14:384 	04 unknown 17320 (0x43A8)
	01:10:14:384 	05 RSSI dB (?) 48 (0x0030)
	01:10:14:385 	06 LQI (?) 64424509440 (0x0F00000000)
	01:10:14:385 	0a unknown 0 (0x0000)
	01:10:14:385 	64 on/off 0
	01:10:14:385 	0b lightlevel 0 (0x0000)
	01:10:16:355     * different nwk address 0x778F / 0x7790
	01:10:16:833     * different nwk address 0x643E / 0x643F
	01:10:16:833     * different nwk address 0x2DF5 / 0x2DF6
	01:10:17:295 discovery for zombie 0x000d6ffffe9a1078 dropped, last try was 396 s ago
	01:10:17:775 discovery for zombie 0x90fd9ffffe13e3c3 dropped, last try was 360 s ago
	01:10:22:549 0x00158D0002B612FA: 0x0101/0x0505: vibration strength: 15
	01:10:22:549 no button handler for: lumi.vibration.aq1 ep: 0x01 cl: 0x0101 cmd: 0x0A pl[0]: 0x05
	01:10:22:549 ZCL attribute report 0x00158D0002B612FA for cluster 0x0101, ep 0x01
	01:10:27:855 discovery for zombie 0x000d6ffffe9a1078 dropped, last try was 407 s ago
	01:10:28:335 discovery for zombie 0x90fd9ffffe13e3c3 dropped, last try was 370 s ago
	01:10:40:351     * different nwk address 0x778F / 0x7790
	01:10:40:830     * different nwk address 0x643E / 0x643F
	01:10:40:830     * different nwk address 0x2DF5 / 0x2DF6
	01:10:41:776 discovery for zombie 0x90fd9ffffe13e3c3 dropped, last try was 384 s ago

For details:

* I use `marthoc/deconz:amd64-2.05.69` docker image
* Gateway firmware: `264A0700`

andras-tim, 2019-10-05:
On the other hand, I'm missing one `lumi.sensor_motion` sensor from the deCONZ app, but this still existing on the Phoscon App's UI.

What happened 2 days ago?! And how can I solve it?

andras-tim, 2019-10-06:
Ohhh, I had to unplug the bad routers from the wall socket, now it started to work again, but the connections are still quite bad + I'm still missing the motion sensor

[image]

andras-tim, 2019-10-06:
for the complete story, I waited ~30 minutes and now the connection quality is quite good now, but the graph is still absolute sub-optimized:
[image]

SimplyGardner, 2019-10-06:
@andras-tim Is your issue related to my issue? If not maybe you should open your issue

andras-tim, 2019-10-06:
Dear @SimplyGardner, I got lots of `different nwk address` messages in the log, but I didn't get any before... so, I don't know.

maximm07, 2019-10-09:
I have the same issue..log is being filled with this messages...i already tried to reset GW and pair devices again, without success.

2:26:39:169 	06 LQI (?) 4294967296 (0x0100000000)
12:26:39:169 	0a unknown 0 (0x0000)
12:26:39:169 	64 on/off 0
12:26:39:623     * different nwk address 0x4CDF / 0x4CE0
12:26:43:942     * different nwk address 0x4CDF / 0x4CE0
12:26:48:254     * different nwk address 0x4CDF / 0x4CE0
12:26:52:583     * different nwk address 0x4CDF / 0x4CE0
12:26:56:904     * different nwk address 0x4CDF / 0x4CE0
12:27:01:224     * different nwk address 0x4CDF / 0x4CE0
12:27:05:544     * different nwk address 0x4CDF / 0x4CE0
12:27:06:713 Current channel 11
12:27:06:721 Device TTL 3777 s flags: 0x7
12:27:09:852     * different nwk address 0x4CDF / 0x4CE0
12:27:14:174     * different nwk address 0x4CDF / 0x4CE0
12:27:18:493     * different nwk address 0x4CDF / 0x4CE0
12:27:22:825     * different nwk address 0x4CDF / 0x4CE0
12:27:27:144     * different nwk address 0x4CDF / 0x4CE0
12:27:31:454     * different nwk address 0x4CDF / 0x4CE0
12:27:35:775     * different nwk address 0x4CDF / 0x4CE0
12:27:40:103     * different nwk address 0x4CDF / 0x4CE0
12:27:44:423     * different nwk address 0x4CDF / 0x4CE0
12:27:48:742     * different nwk address 0x4CDF / 0x4CE0
12:27:53:062     * different nwk address 0x4CDF / 0x4CE0
12:27:57:382     * different nwk address 0x4CDF / 0x4CE0
12:28:01:694     * different nwk address 0x4CDF / 0x4CE0
12:28:06:022     * different nwk address 0x4CDF / 0x4CE0
12:28:06:714 Current channel 11
12:28:06:722 Device TTL 3717 s flags: 0x7
12:28:10:334     * different nwk address 0x4CDF / 0x4CE0
12:28:14:661     * different nwk address 0x4CDF / 0x4CE0
12:28:18:982     * different nwk address 0x4CDF / 0x4CE0
12:28:23:293     * different nwk address 0x4CDF / 0x4CE0
12:28:27:621     * different nwk address 0x4CDF / 0x4CE0
12:28:31:934     * different nwk address 0x4CDF / 0x4CE0
12:28:36:264     * different nwk address 0x4CDF / 0x4CE0
12:28:40:583     * different nwk address 0x4CDF / 0x4CE0
12:28:44:903     * different nwk address 0x4CDF / 0x4CE0
12:28:49:222     * different nwk address 0x4CDF / 0x4CE0
12:28:53:542     * different nwk address 0x4CDF / 0x4CE0
12:28:57:864     * different nwk address 0x4CDF / 0x4CE0
12:29:02:172     * different nwk address 0x4CDF / 0x4CE0
12:29:06:503     * different nwk address 0x4CDF / 0x4CE0
12:29:06:713 Current channel 11
12:29:06:721 Device TTL 3657 s flags: 0x7
12:29:10:822     * different nwk address 0x4CDF / 0x4CE0
12:29:15:142     * different nwk address 0x4CDF / 0x4CE0
12:29:19:463     * different nwk address 0x4CDF / 0x4CE0
12:29:23:783     * different nwk address 0x4CDF / 0x4CE0
12:29:28:104     * different nwk address 0x4CDF / 0x4CE0
12:29:32:423     * different nwk address 0x4CDF / 0x4CE0
12:29:36:744     * different nwk address 0x4CDF / 0x4CE0
12:29:41:054     * different nwk address 0x4CDF / 0x4CE0
12:29:45:374     * different nwk address 0x4CDF / 0x4CE0
12:29:49:693     * different nwk address 0x4CDF / 0x4CE0
12:29:54:015     * different nwk address 0x4CDF / 0x4CE0
12:29:58:334     * different nwk address 0x4CDF / 0x4CE0
12:30:02:654     * different nwk address 0x4CDF / 0x4CE0
12:30:06:715 Current channel 11
12:30:06:723 Device TTL 3597 s flags: 0x7
12:30:06:986     * different nwk address 0x4CDF / 0x4CE0
12:30:07:441 Announced to internet
12:30:07:442 discovery server date: Wed, 09 Oct 2019 10:30:07 GMT
12:30:07:442 	 local time seems to be ok
12:30:11:303     * different nwk address 0x4CDF / 0x4CE0
12:30:11:765 dev /dev/ttyAMA0
12:30:11:765 GW firmware version: 0x264a0700
12:30:11:765 GW firmware version is up to date: 0x264a0700
12:30:15:624     * different nwk address 0x4CDF / 0x4CE0
12:30:19:941     * different nwk address 0x4CDF / 0x4CE0
12:30:24:254     * different nwk address 0x4CDF / 0x4CE0
12:30:28:584     * different nwk address 0x4CDF / 0x4CE0
12:30:32:892     * different nwk address 0x4CDF / 0x4CE0
12:30:37:221     * different nwk address 0x4CDF / 0x4CE0
12:30:41:534     * different nwk address 0x4CDF / 0x4CE0
12:30:45:865     * different nwk address 0x4CDF / 0x4CE0
12:30:50:183     * different nwk address 0x4CDF / 0x4CE0
12:30:54:493     * different nwk address 0x4CDF / 0x4CE0
12:30:58:823     * different nwk address 0x4CDF / 0x4CE0
12:31:03:141     * different nwk address 0x4CDF / 0x4CE0
12:31:06:714 Current channel 11
12:31:06:722 Device TTL 3537 s flags: 0x7
12:31:07:462     * different nwk address 0x4CDF / 0x4CE0
12:31:11:782     * different nwk address 0x4CDF / 0x4CE0
12:31:16:102     * different nwk address 0x4CDF / 0x4CE0
12:31:20:413     * different nwk address 0x4CDF / 0x4CE0
12:31:24:744     * different nwk address 0x4CDF / 0x4CE0
12:31:29:052     * different nwk address 0x4CDF / 0x4CE0
12:31:33:381     * different nwk address 0x4CDF / 0x4CE0
12:31:37:701     * different nwk address 0x4CDF / 0x4CE0
12:31:42:026     * different nwk address 0x4CDF / 0x4CE0
12:31:46:342     * different nwk address 0x4CDF / 0x4CE0
12:31:50:661     * different nwk address 0x4CDF / 0x4CE0
12:31:54:983     * different nwk address 0x4CDF / 0x4CE0
12:31:59:303     * different nwk address 0x4CDF / 0x4CE0
12:32:03:624     * different nwk address 0x4CDF / 0x4CE0
12:32:06:706 Current channel 11
12:32:06:708 Device TTL 3477 s flags: 0x7
12:32:07:942     * different nwk address 0x4CDF / 0x4CE0
12:32:12:254     * different nwk address 0x4CDF / 0x4CE0
12:32:16:585     * different nwk address 0x4CDF / 0x4CE0

Viald, 2019-11-09:
Hi. Running the last release 2.05.69, I have the same issue.
`09:41:42:630 Current channel 20
09:41:42:634 CTRL got nwk update id 2
09:41:42:638 Device TTL 755 s flags: 0x7
09:41:51:319 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
09:41:51:319 ZCL attribute report 0x00158D0003F11509 for cluster 0x0402, ep 0x01
09:41:51:328 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
09:41:51:328 ZCL attribute report 0x00158D0003F11509 for cluster 0x0405, ep 0x01
09:41:51:332 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
09:41:51:332 ZCL attribute report 0x00158D0003F11509 for cluster 0x0403, ep 0x01
09:41:53:655 Websocket disconnected 192.168.1.7:62144 (state: 0) 
09:41:59:785 New websocket 192.168.1.7:62234 (state: 3) 
09:42:01:590 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
09:42:01:590 ZCL attribute report 0x00158D0003F11509 for cluster 0x0402, ep 0x01
09:42:01:598 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
09:42:01:598 ZCL attribute report 0x00158D0003F11509 for cluster 0x0405, ep 0x01
09:42:01:602 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
09:42:01:602 ZCL attribute report 0x00158D0003F11509 for cluster 0x0403, ep 0x01
09:42:11:864 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
09:42:11:864 ZCL attribute report 0x00158D0003F11509 for cluster 0x0402, ep 0x01
09:42:11:874 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
09:42:11:874 ZCL attribute report 0x00158D0003F11509 for cluster 0x0405, ep 0x01
09:42:11:876 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
09:42:11:876 ZCL attribute report 0x00158D0003F11509 for cluster 0x0403, ep 0x01
09:42:42:632 Current channel 20
09:42:42:636 CTRL got nwk update id 2
09:42:42:640 Device TTL 695 s flags: 0x7
09:42:52:245 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
09:42:52:245 ZCL attribute report 0x00158D0003F11509 for cluster 0x0402, ep 0x01
09:42:52:252 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
09:42:52:252 ZCL attribute report 0x00158D0003F11509 for cluster 0x0405, ep 0x01
09:42:52:256 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
09:42:52:256 ZCL attribute report 0x00158D0003F11509 for cluster 0x0403, ep 0x01
09:43:22:597 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
09:43:22:597 ZCL attribute report 0x00158D0003F11509 for cluster 0x0402, ep 0x01
09:43:22:606 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
09:43:22:606 ZCL attribute report 0x00158D0003F11509 for cluster 0x0405, ep 0x01
09:43:22:610 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
09:43:22:610 ZCL attribute report 0x00158D0003F11509 for cluster 0x0403, ep 0x01
09:43:42:631 Current channel 20
09:43:42:635 CTRL got nwk update id 2
09:43:42:639 Device TTL 635 s flags: 0x7
09:44:42:631 Current channel 20
09:44:42:635 CTRL got nwk update id 2
09:44:42:639 Device TTL 575 s flags: 0x7
09:44:42:639 Device reset watchdog 7200 s
09:45:42:632 Current channel 20
09:45:42:636 CTRL got nwk update id 2
09:45:42:640 Device TTL 7140 s flags: 0x7
09:46:42:632 Current channel 20
09:46:42:636 CTRL got nwk update id 2
09:46:42:640 Device TTL 7079 s flags: 0x7
09:47:24:976 no button map for: lumi.weather ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
09:47:24:976 ZCL attribute report 0x00158D00044A4720 for cluster 0x0000, ep 0x01
09:47:24:976 0x00158D00044A4720 extract Xiaomi special attribute 0xFF01
09:47:24:976 	01 battery 3005 (0x0BBD)
09:47:24:976 	04 unknown 5032 (0x13A8)
09:47:24:976 	05 RSSI dB (?) 79 (0x004F)
09:47:24:976 	06 LQI (?) 4294967296 (0x0100000000)
09:47:24:976 	64 temperature 1999
09:47:24:976 	65 humidity 5803
09:47:24:976 	66 pressure 1001
09:47:24:976 	0a unknown 0 (0x0000)
09:47:42:630 Current channel 20
09:47:42:633 CTRL got nwk update id 2
09:47:42:637 Device TTL 7019 s flags: 0x7
09:47:44:415 Announced to internet
09:48:42:629 Current channel 20
09:48:42:633 CTRL got nwk update id 2
09:48:42:637 Device TTL 6959 s flags: 0x7
[... 27 lines cut ...]
09:51:28:677     * different nwk address 0xC6EC / 0xC6ED
09:51:33:478     * different nwk address 0xC6EC / 0xC6ED
09:51:38:277     * different nwk address 0xC6EC / 0xC6ED
09:51:42:599     * different nwk address 0xC6EC / 0xC6ED
09:51:42:633 Current channel 20
09:51:42:637 CTRL got nwk update id 2
09:51:42:641 Device TTL 6779 s flags: 0x7
09:51:46:916     * different nwk address 0xC6EC / 0xC6ED
09:51:51:235     * different nwk address 0xC6EC / 0xC6ED
09:51:55:558     * different nwk address 0xC6EC / 0xC6ED
09:51:59:877     * different nwk address 0xC6EC / 0xC6ED
09:52:04:198     * different nwk address 0xC6EC / 0xC6ED
09:52:08:516     * different nwk address 0xC6EC / 0xC6ED
09:52:12:836     * different nwk address 0xC6EC / 0xC6ED
09:52:17:156     * different nwk address 0xC6EC / 0xC6ED
09:52:21:476     * different nwk address 0xC6EC / 0xC6ED
09:52:25:798     * different nwk address 0xC6EC / 0xC6ED
09:52:30:117     * different nwk address 0xC6EC / 0xC6ED
09:52:34:437     * different nwk address 0xC6EC / 0xC6ED
09:52:38:758     * different nwk address 0xC6EC / 0xC6ED
09:52:42:630 Current channel 20
09:52:42:634 CTRL got nwk update id 2
09:52:42:638 Device TTL 6719 s flags: 0x7
09:52:43:078     * different nwk address 0xC6EC / 0xC6ED`
`

closed 2020-03-15 by stale[bot]