dresden-elektronik/deconz-rest-plugin #1775: WiringPi will be depreceated soon

TheTimeWalker, 2019-08-24:
The author of WiringPi anounced that his library will be depreceated soon according to the latest blog post from two weeks ago: http://wiringpi.com/wiringpi-deprecated/

Since this library depends on it for the Raspbee, I'd like to create a discussion on how to go on with it in the future as there's always a chance future revisions like a new Raspberry Pi release could break something. We already saw with the issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1762 that updating on Pi 4 didn't work without manually updating to the latest version. So the risk can get pretty high with hardware changes or full upgrades.

manup, 2019-08-24:
The WiringPi library is used to toggle the RaspBee reset pin to reboot the firmware into bootloader mode for firmware updates. The plan is to remove the dependency and use the new GPIO linux library `libgpiod` in future versions of GCFFlasher. This should also be portable to Raspberry Pi clones with similar header interface.

https://www.acmesystems.it/libgpiod

closed 2019-12-29 by stale[bot]

quotschmacher, 2020-07-29:
any progress on this topic? i just wanted to install http://deconz.dresden-elektronik.de/raspbian/gcfflasher/gcfflasher-latest.deb but it yields an error, that wiringpi is not available. as this deb-package is from 2017 i just wanted to ask...

SwoopX, 2020-07-29:
GCFFlasher comes with deconz package already and is more up to date (3.1.3 afaik). Not sure though if the dependency still exists.

quotschmacher, 2020-07-29:
you are right. but with the actual version of the iobroker zigbee adapter allows you to use the raspbee without deconz ( https://github.com/Koenkk/zigbee-herdsman/issues/72 ). so i only wanted to install the flasher to update my raspbee. but the deb-file requires 'wiringPi' which is not available - 'wiringpi' is preinstalled in raspbian buster (i think it is raspberry os now). so i was looking for a standalone, which seems to be pretty outdated.