dresden-elektronik/deconz-rest-plugin #1776: Innr ZHAConsumption

wvuyk, 2019-08-24:
I have one user of my plugin having a weird issue with the plugin. He is using deCONZ 2.05.66 and the power consuption is not reporting in decimals, but in power notation? is this a change that has been done in 2.05.66? As I am using 2.05.65 due to issue in .66 and this one would worry me....

"manufacturername":"innr","modelid":"SP 120","name":"Consumption 18","state":{**"consumption":1.06933e+06**,"lastupdated":"2019-08-24T16:38:50"},"swversion":"2.0","type":"ZHAConsumption","uniqueid":"00:15:8d:00:03:44:47:13-01-0702"}

ebaauw, 2019-08-24:
The largest `state.consumption` value on any of my innr plugs is ~200000 (Wh), so that's not going to help us.  I cannot reproduce this particular scientific notation output with a CLIPConsumption sensor:
```
$ ph put -v /sensors/13/state '{"consumption": 1.06933e+06}'
{
  "consumption": 1069330
}
$ ph get /sensors/13
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "495609a4dbf5279dd9880d2aa652cf1b",
  "manufacturername": "Test",
  "modelid": "CLIPConsumption",
  "name": "Test",
  "state": {
    "consumption": 1069330,
    "lastupdated": "2019-08-24T20:44:56"
  },
  "swversion": "1.0",
  "type": "CLIPConsumption",
  "uniqueid": "test"
}
```
But the database contains: `{"consumption":1.06933e+06,"lastupdated":"2019-08-24T21:06:03"}`.

With larger values, I do get the scientific notation in the API output.
```
$ ph put -v /sensors/13/state '{"consumption": 1234567890}'
{
  "consumption": 1234570000
}
```
As well as in the web socket notification:
```
Aug 24 22:59:17 pi1 dc_eventlog[310]: /sensors/13/state: {"consumption":1234570000,"lastupdated":"2019-08-24T20:59:17"}
```
After this, the database contains: `{"consumption":1.23457e+09,"lastupdated":"2019-08-24T21:02:47"}`.

Even after patching the database to `1234567890`, the API continues to report the scientific notation.

One could argue that the scientific notation is still valid JSON, and any API client should be able to handle it.  However, the loss of precision is definitely a bug.  I don't think it's introduced recently, but surfaces now, as the consumption value exceeds six digits.  If I were to guess, it's probably an issue with `ResourceItem` and/or `QVariant` not supporting `uint64` properly.

wvuyk, 2019-08-25:
Erik,

You did put me on track here. thanks. I was using integer and indeed should have expected doubles.
Is there an overview on that datatype used  per attribute in the REST API? I searched but could nt find any, not even in the ussue list?

Thanks,

Wim

ebaauw, 2019-08-25:
> I was using integer and indeed should have expected doubles.

Javascript and consequently JSON only know `number`; they don't make the distinction between integer and float / double.  The ZigBee attribute _Current Summation Delivered_ is `uint48`; the resource item `RStateConsumption` is `uint64`.

> Is there an overview on that datatype used per attribute in the REST API?

Not in the documentation.  Use the Source, Luke: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/dab8ea9ea44e9a74bac742ec3872909a6a9a81cc/resource.cpp#L143.

closed 2019-12-30 by stale[bot]