dresden-elektronik/deconz-rest-plugin #1779: How can a switch be removed from a group?

jo-me, 2019-08-25:


Smanar, 2019-08-26:
I have make tries on my side, and if I select the switch I have "delete key asssignement" that appear on the right top corner.

jo-me, 2019-08-28:
Yes, I can remove the key assignment but I cannot remove the switch altogether (so that it disappears from the list on the left).

Smanar, 2019-08-28:
On the left side ""Xiaomi_Bedroom"  ? It'a group name ? Try delete the group.

jo-me, 2019-08-28:
No, the group has two remotes / switches assigned "xiaomi_bedroom" and "dimmschalter..." .
I can maintain the key assignments but I cannot remove the switches from the group once I added them to the list on the left.

Smanar, 2019-08-29:
Lol, right, I have same problem too ^^.

I will try with API

JackPoint, 2019-11-19:
- From the main page, select the group
- Click manage lights
- Remove lights from the left part of the page

Smanar, 2019-11-20:
And this procedure remove the switch too ?

JackPoint, 2019-11-20:
This is to remove it from a group, not completely remove it

noerstad, 2019-12-14:
Delete all key assignments, then go to the group setting. The switch automagically dissapears when you have no key assignments.
NOTE: I also removed one light from the group, might be that forced the group to update

closed 2020-04-19 by stale[bot]