dresden-elektronik/deconz-rest-plugin #1780: "Toggle

jo-me, 2019-08-25:


manup, 2019-08-27:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_67

The issue is fixed in 2.05.67, but you need to update or re-create the switch button assignment.

jo-me, 2019-08-28:
Seems to work fine after the update :) 
Thank you!

closed 2019-08-28 by jo-me

LJSven, 2019-09-05:
What does it mean? What to do?

„update or re-create the switch“

manup, 2019-09-05:
You need to edit the related switch assignment in the Phoscon App switch editor.
Easiest way:
* Change toggle to "turn on"
* "turn on" to toggle again
* Press the Save button

That way the related rules will be rewritten and corrected.

LJSven, 2019-09-05:
Is it also necessary, if i´m using it for ioBroker? I don´t want to control the lights / scenes in the Phoscon app.

manup, 2019-09-05:
The issue is only related to the switches which are configured via Phoscon App switch editor.