dresden-elektronik/deconz-rest-plugin #1781: Homepage linkgs to old deconz software

bjornfor, 2019-08-25:
(Hi, sorry if this is the wrong place, but I couldn't find any better.)

After getting my ConBee II, I found the download page for the deCONZ software here: https://www.dresden-elektronik.de/funktechnik/products/software/pc-software/deconz/?L=1. The problem is it links to an *old version* of the software (v2.5.20). That version does not even work with my ConBee II. I wasted some time on this (using strace etc.) until I found the docker image (https://github.com/marthoc/docker-deconz/) which has the latest version (v2.5.66) and fetches it from a better location (human readable and meaningful URL).

Please update the software download links on the homepage.

manup, 2019-08-25:
The ConBee II package should contain a small inlet which points to https://phoscon.de/conbee2

Unfortunately, the dresden elektronik homepage isn't fully updated yet with the correct links,  these will be updated in the coming weeks. The plan is also to create a proper deCONZ page on phoscon.de.

bjornfor, 2019-08-26:
> The ConBee II package should contain a small inlet which points to https://phoscon.de/conbee2

Right. Although that page has many links (e.g. to the docker image), it does not have the link to the Ubuntu .deb package. To be fair, it does link to installation instructions for Ubuntu, but I was actually looking for the .deb *file* (because I don't run apt), and for that, I landed on https://www.dresden-elektronik.de/funktechnik/products/software/pc-software/deconz/ (probably via Google).

> Unfortunately, the dresden elektronik homepage isn't fully updated yet with the correct links, these will be updated in the coming weeks. The plan is also to create a proper deCONZ page on phoscon.de.

Thanks.

manup, 2019-08-26:
>Right. Although that page has many links (e.g. to the docker image), it does not have the link to the Ubuntu .deb package. 

The package files are currently only available via the links provided in:
https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages-for-manual-installation

manup, 2019-08-28:
The old links are now removed and additional links to the new phoscon.de website were added.
The links to the package files will be added on phoscon.de in the next days. Closing the issue for now.

closed 2019-08-28 by manup

bjornfor, 2019-08-28:
AFAICS, the homepage (https://www.dresden-elektronik.de/funktechnik/products/software/pc-software/deconz/?L=1) only links to phoscon.de under the "Matching hardware" section (checked just now). The whole page is about the deCONZ *software*, so why not have some links where to download it? Landing on that homepage now would still have me "stuck", since I would not think of clicking the hardware links, seeing I already have hardware.

To sum up my experience:
* Old situation: homepage leads to old software (user=sad).
* New situation: homepages leads to no software (user=even more sad).