dresden-elektronik/deconz-rest-plugin #1782: IKEA Dimmer only exposing battery_level through API

martinkrulltott, 2019-08-26:
Hey,
I'm controlling a bunch of IKEA trådfri bulbs and switches using a Conbee II, raspberry pi (Raspbian Stretch), Deconz and Home Assistant (through Docker)
My switches and bulbs are working fine, but the IKEA Trådfri Wireless Dimmer (the simple round one) isn't working.
When paired to a light directly ti's dimming as it should. But when I've paired it with Conbee through Phoscon, it's only being exposed to Home Assistant as "dimmer_battery_level" with device class: battery. At first I thought the battery level = dimmer level, but when turning the dimmer the value doesn't change. For some reason it's always stuck at 16.
In the Phoscon UI it's displayed as a dimmer (version 1.1.1.1-5.7.2.0).

I'm still trying to figure out if this is a Home Assistant issue or a Deconz issue, but as it seems to be exposed through the API but as the wrong device class and value, I'm assuming that the issue lies on the API side?

Has anyone else experienced something similar?

Kane610, 2019-08-26:
Remotes are only exposed as events in HASS, see https://www.home-assistant.io/components/deconz/#step-up-and-step-down-input-number-with-wireless-dimmer as an example on how to use it.

You can use the event listener in developer tools to see what signals you get from the dimmer. Just listen to deconz_event and then rotate the device and you should see what signals to use in automations.

martinkrulltott, 2019-08-26:
Thanks @Kane610, I used the guide you sent and it works great!

closed 2019-08-26 by martinkrulltott