dresden-elektronik/deconz-rest-plugin #1784: 2.05.67 missing deb package from website

phdelodder, 2019-08-27:
I'm trying to download the latest release: 2.05.67 and it's missing from: https://www.dresden-elektronik.de/rpi/deconz/beta/

manup, 2019-08-27:
The RPi package is now online but I've just notices the Ubuntu build failed :/ ... I'm on it

phdelodder, 2019-08-27:
Package deconz_2.05.66-debian-stretch-beta_arm64.deb is missing on https://www.dresden-elektronik.de/rpi/deconz/alpha/

@manup thanks

manup, 2019-08-27:
The files are online now :)

https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.67-qt5.deb

https://www.dresden-elektronik.de/rpi/deconz/alpha/deconz_2.05.67-debian-stretch-beta_arm64.deb

The APT-repository is rebuilding so in a few hours the remaining files for updating via apt-get will be online.

closed 2019-08-27 by phdelodder