dresden-elektronik/deconz-rest-plugin #1786: IKEA color temperature is missing after the update.

salopette, 2019-08-27:
IKEA color temperature is missing after the update.

[image]

MacJawa, 2019-08-28:
@salopette , after update to which version? What bulb do you have exactly? Because I'm in similar situation. I bought new IKEA E14 CWS Bulb, which according to description should be able to change brightness, colors as well as temperature. I'm using version 2.05.65 (which states it's up to date) and I my screen looks exactly as yours. I can change brightness, color and saturation, but Color temperature button does nothing.

ebaauw, 2019-08-28:
I don’t know the E14, but the regular IKEA CWS bulb does not support colour temperature.  They (used to?) fake colour temperature support in the REST API plugin, translating the `ct` value into `xy`.  The E14 bulb probably needs to be whitelisted for this feature.

> according to description

Which description?  By whom?

salopette, 2019-08-28:
[image]
[image]

MacJawa, 2019-08-28:
> > according to description
> 
> Which description? By whom?

Well, I bought mine in Slovakia and according to description on the website, you should be able to change color and temperature as well. With IKEA remote you should be able to choose from 4 temperatures and 5 colors. With IKEA gateway you have even more options. [NL page](https://www.ikea.com/nl/nl/p/tradfri-led-lamp-e14-600-lumen-draadloos-dimbaar-kleur-en-wit-spectrum-opaalwit-50411565/) states the same if I'm not mistaken (I only used google translator). What more, the E27 version have the [same description](https://www.ikea.com/nl/nl/p/tradfri-led-lamp-e27-600-lumen-draadloos-dimbaar-kleur-en-wit-spectrum-opaalwit-00408612/).
This is [german site](https://www.ikea.com/de/de/p/tradfri-led-leuchtmittel-e14-600-lm-kabellos-dimmbar-farb-und-weissspektrum-opalweiss-50411565/) for E14 version and all three have the same product number - 504.115.65 and the same description.

I have little to zero experience in this area, because I only started with deconz and smart lights, but I'm a fast learner and I feel like something is not right in deConz about this bulb. Actually I noticed the issue in openHAB, where this E14 bulb is reported as 0x0200 thing type using Connbee 2 and deConz, while it should be 0x0210 I believe. Because of that I can't control the bulb from openHAB. So for a past few days I'm trying to learn more about this stuff and I ended up here, because (as guy from software development environment) I think the issue is somewhere in deConz REST API, which reports this bulb as "Color light" instead of "Extended Color light", which I believe it is.

There is probably one way how to test my theory, buy the IKEA gateway and connect it to openHAB to see how their own gateway will report this bulb. But to be honest this is the last option, because I bought ConnBee just because I don't want to buy each vendor's gateway :)

Right now I don't know where is the real root of this issue, I can only sumarize what I found so far:
- my bulb has the same firmware as @salopette  has - 1.3.009
- my bulb could be controlled in some way using Phoscon web application, but I cannot change the color temperature, only brightness, color and saturation
- I can set brightness, color and saturation, but I when I check the bulb state using REST client, 'hue' and 'sat' are always zero, only brightness and x,y change their values (probably because of the "hack" @ebaauw  mentioned)
- I did not try to set hue and sat values using REST client yet, not sure if this will help, but I plan to do that anyway
- this bulb is reported as Color Light in REST API, so logically the same way openHAB sees this thing, but it cannot be controlled from openHAB. This could be also issue on another level, but while openHAB uses general Philips HUE binding to control lights via deConz and ConnBee, I guess that part knows how to control Color Light bulb and the issue is somewhere on deConz side
- I also have IKEA E14 WS bulbs, these works just fine, I can change the brightness and color temperature as expected from both Phoscon and openHAB
- just a sidenote - I believe WS states for White Spectrum and CWS for Color White Spectrum, therefore this one should be Extended Color Light with temperature as well. But it's only my guess :)

Just to be on the same level, I don't blame deConz for anything bad, I just written down some notes I gathered until now. Actually I'm pretty happy this option exists and I merely want to contribute to the project to make IKEA and ConnBee the perfect combination :)

ebaauw, 2019-08-28:
> my bulb has the same firmware as @salopette has - 1.3.009

I only have the original E27 colour light, which does not support colour temperature (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/218).  Could very well be that they changed that for the newer models - I think these have different firmware.

> Actually I noticed the issue in openHAB, where this E14 bulb is reported as 0x0200 thing type using Connbee 2 and deConz, while it should be 0x0210 I believe
>  I think the issue is somewhere in deConz REST API, which reports this bulb as "Color light" instead of "Extended Color light", which I believe it is.

The REST API simply translates the ZigBee device type as advertised by the bulb to text.  Can you double-check what device type is shown in the deCONZ GUI?  Is this a ZLL light, or ZigBee 3.0 (shown as ZHA in the GUI).   Could you post (several) screenshots of the _Cluster Info_ panel showing the _Color Control_ attributes.  Please press _Read_, so the attributes reflect the light.  Of particular interest is 0x400A, which advertises the colour capabilities.

Could you try, still in the deCONZ GUI, if the light reacts to the _Move to color temperature_ command?

> I can set brightness, color and saturation, but I when I check the bulb state using REST client, 'hue' and 'sat' are always zero, only brightness and x,y change their values 

Unlike Hue lights, IKEA lights don't update the other colour attributes when setting the colour.  I'm not sure of this light support (Enhanced) Hue/Sat, see 0x400A above, and check whether it responds to the _Move to Hue/Sat_ or _Move to Enhanced Hue/Sat_.  `hue` and `sat` are utterly useless anyway, as they show different colours for different types of lights.  You want to use `xy` to set colours.

> this bulb is reported as Color Light in REST API, so logically the same way openHAB sees this thing, but it cannot be controlled from openHAB. 

I don't know openHAB, but do they use `xy` or `hue`/`sat`?

> just a sidenote - I believe WS states for White Spectrum and CWS for Color White Spectrum, therefore this one should be Extended Color Light with temperature as well. But it's only my guess :)

That probably gets lost in the translation to Swedish...

> I don't blame deConz for anything bad, I just written down some notes I gathered until now. Actually I'm pretty happy this option exists and I merely want to contribute to the project to make IKEA and ConnBee the perfect combination

From past experience, IKEA tends to have a non-traditional interpretation of the ZigBee "standard".  You'll probably run into the same issues pairing the bulb with a Hue bridge.  The cool thing about deCONZ is that the REST API plugin is open source, and we can work around the IKEA idiosyncrasies.

MacJawa, 2019-08-30:
Thank you Erik for your reply, which helps me to get deeper :) I have a lot to learn.

> The REST API simply translates the ZigBee device type as advertised by the bulb to text. Can you double-check what device type is shown in the deCONZ GUI? Is this a ZLL light, or ZigBee 3.0 (shown [...]

It's ZLL according to GUI:
[image]

Here are Color Control attributes:
[image]
[image]
[image]
[image]

What the value of 0x400A means?

> 
> Could you try, still in the deCONZ GUI, if the light reacts to the _Move to color temperature_ command?

I can't try it right now, probably later today I will be able to test it and post the results.

> Unlike Hue lights, IKEA lights don't update the other colour attributes when setting the colour. I'm not sure of this light support (Enhanced) Hue/Sat, see 0x400A above, and check whether it [...]

I understand that I want to set colors by 'xy'. But what about temperature of white light? I still don't understand if it is possible to set _warm_ white and _cold_ white using 'xy' as well. If the answer is yes and color temperature is just the subset of color spectrum, then it's only about the controls we use to set the color/temperature values, right? If I know the 'xy' of Cold White (6000K), I can set it directly, can I?

> I don't know openHAB, but do they use `xy` or `hue`/`sat`?

I have to check this, but first I wanted to understand the topic, so I know where to focus. I think there was a discussion on openHAB forum about converting some values to 'xy', I have to find it again.

> From past experience, IKEA tends to have a non-traditional interpretation of the ZigBee "standard". You'll probably run into the same issues pairing the bulb with a Hue bridge. The cool thing about deCONZ is that the REST API plugin is open source, and we can work around the IKEA idiosyncrasies.

That's why I've chosen deCONZ over proprietary gateways :)

ebaauw, 2019-08-30:
> What the value of 0x400A means?

That it only supports _Current X_ and _Current Y_.  That's reassuringly consistent with not supporting the (greyed out) attributes for: _Hue_, _Saturation_, _Color Temperature_, _Color Loop_, and _Enhanced Hue_.  If you double-click on the _Color Capabilities_ attribute, the deCONZ GUI pops up the _Attribute Editor_ window, with checkboxes decoding the bitmap value.

> But what about temperature of white light? I still don't understand if it is possible to set warm white and cold white using 'xy' as well.

Yes, the colour temperatures (`ct` values) can be translated to `xy` values, see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/c6981385c63d2728808f45a9aa7dc20b9c94028a/colorspace.cpp#L1351

The algorithm is beyond my understanding, though.

How well the warm or cool white is displayed by the bulb actually depends on the range of colours (the gamut) supported by the bulb's hardware.  Different types of bulbs use different coloured LEDs to mix colours.  

Colours are created by mixing RGB LEDs.  However, whites usually don't come out well when mixing RGB.  That's why most 3-channel bulbs (like the Hue gamut-A and the Trådfri) don't support `ct`.  Some bulbs (like the Hue gamut-C) use additional WW/CW LEDs to mix colour temperatures (just like colour temperature lights).  On these five-channel bulbs, setting `ct` will drive the WW/CW channels, whereas setting the corresponding `xy` values will drive the RGB channels.  Other three-channel bulbs (like the Hue gamut-B) optimised their RGB LEDs for displaying whites, at the expense of deep blue and green colours.

Some bulbs, like Hue and OSRAM, actually report the colours of their RGB LEDs through attributes 0x0010 - 0x001B.  Trådfri doesn't.  The WW and CW LEDs should correspond to _Colour temperature max_ and _Color temperature min_.

See also https://developers.meethue.com/develop/get-started-2/core-concepts/, https://developers.meethue.com/develop/hue-api/supported-devices/ and https://developers.meethue.com/develop/application-design-guidance/color-conversion-formulas-rgb-to-xy-and-back/.

MacJawa, 2019-08-30:
@ebaauw , thank you very much for all your explanations and links provided. It was very helpful to me.

At the end, it looks like I was completely wrong about IKEA bulbs. These are not _Extended Color Light_, but _Color Light_ type of bulbs. Please excuse my newbie mistake. 

Now the question is, if Color temperature button in Phoscon app should do nothing as OP originaly mentioned or if there is some bug causing this behaviour. I cannot compare to older version nor to another vendor bulbs, so I leave the answer to someone else.
My mission now would be to find out why I can't control IKEA CWS bulb from openHAB.

RoadRunner120485, 2019-11-06:
Hey there. After my Tradfri Gateway stops working once a day and I don't like the need for Cloud of my SmartThings Hub I recently bought a Conbee II and I recognized too, that the Color Temperature slider for the Tradfri E27 CWS opal 600lm does not appear when clicking on the icon. 

I don't know the implementation details, about when and why the Color Temperature switch should work in Phoscon. But in the old "Wireless Light Control" UI I can set Color as well as Color Temperature for the IKEA bulb. And as the older UI could set the color temperature (maybe using any type of conversion) it feels more like a bug to me than a feature that the toggle on Phoscon does not work.
[image]

Flurkmark, 2019-11-06:
There is something making IKEA lights (all three types I have at least) stop reporting color temperature after you change it. It presents as a xy light instead. If you change color temperature with ct_inc instead it does not happen. It was supposed to be fixed in 2.05.70.

ReekyMarko, 2019-11-07:
Same issue here, I discovered a workaround for home assistant users and posted about it on the home assistant issue tracker a workaround

https://github.com/home-assistant/home-assistant/issues/14554#issuecomment-550899713

sildehoop, 2020-02-18:
Hello I have the same issue, and the workaround does not work for me ether.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

MaxWinterstein, 2020-07-24:
Mhm, no working color support for me here, too. 
So my wish to get rid of the tradfri gateway seems not possible :(

johnekarlsson, 2020-08-02:
Same problem for me, still.

Mimiix, 2020-08-03:
@MaxWinterstein check: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2526

boelle, 2021-01-08:
@Mimiix still an issue

i can change from phoscon but not on the remote, and i did try to change mode on the remote but no luck

i looked at #2526 but i did not figure out what is wrong (maybe me stupid?)