dresden-elektronik/deconz-rest-plugin #1788: Auth problems on non-80 http port

bjornfor, 2019-08-28:
If I run the deconz / Phoscon App on a non-80 http port (e.g. --http-port=8080), the auth mechanism start having problems. After the initial "Set login password" page, the app works. But if I log out, and try to login again, I cannot. The app constantly redirects me to the "Set login password" page and rejects my password.

Steps to reproduce:

1. run deCONZ with --http-port=8080
2. do the initial login / set password at http://localhost:8080 (it redirects to the proper /pwa/login.html page)
3. logout
4. try to login... instead of getting the expected single-field password page, it shows the initial login page and rejects the (correct) password

Tested using the amd64 Dockerfile from https://github.com/marthoc/docker-deconz (commit a64c032797bf0e2346a2f880f67e805ef4993d0d), I only changed DECONZ_WEB_PORT in the Dockerfile. I used Firefox 68 and Chromium 76 for testing.

bjornfor, 2019-12-26:
Not stale.

pgardiner, 2020-02-05:
Is there a workaround for this? Currently I have to recreate the Docker container and restore before making any changes. I had the same behaviour as @bjornfor, except I'm using port 10080, and am using.

bjornfor, 2020-06-05:
Still important to me.

bjornfor, 2020-06-05:
Ping @manup.

Mimiix, 2020-06-12:
This is more a docker issue than a deCONZ issue, You'd be better off in the marthoc docker git.

bjornfor, 2020-06-12:
@Mimiix: What makes you say it's a docker issue?

I was thinking that using docker would make it easy to reproduce and fix. (Btw, I don't actually plan to run it in docker, but on NixOS + systemd.)

Mimiix, 2020-06-12:
Ah wait, i was blinded on the comment of @pgardiner  and your comment about the dockerfile. Sorry, need some more coffee this morning.

So where would this issue be. In Phoscon or the API?

bjornfor, 2020-06-12:
> So where would this issue be. In Phoscon or the API?

I don't know.

bjornfor, 2020-06-14:
I reproduced the issue with deconz-2.05.77 using the latest https://github.com/marthoc/docker-deconz/.

This is a real blocker for me. I don't have a dedicated device to run deconz on and I already use 80 for a webserver. So I have to run deconz on non-80 port. The current state is I'm unable to use my zigbee hardware for almost a year now, due to a software auth issue.

bjornfor, 2020-07-05:
Still relevant.

SwoopX, 2020-07-05:
Just stop using localhost/127.0.0.1 and be happy

bjornfor, 2020-07-05:
@SwoopX: You mean just use a dedicated machine where deconz can run on port 80? (Because it's not just localhost that seem buggy with this software, it's the port too.) I could do that, but find it whasteful. The software even has a CLI option to select which port to run on...

SwoopX, 2020-07-05:
No, I mean what I wrote. Stop using lo interface for access.

Btw, I run deconz on port 8080 as well just fine since 2 years.

closed 2020-08-02 by stale[bot]