dresden-elektronik/deconz-rest-plugin #1792: Trying to change password: "Service not available. Try again later."

bjornfor, 2019-08-28:
This is not 100 % reproducible (unfortunately), but I often get this error when trying to change the password: "Service not available. Try again later." (this shows up below the Confirm Password field).

Tested using the amd64 Dockerfile from https://github.com/marthoc/docker-deconz (commit a64c032797bf0e2346a2f880f67e805ef4993d0d, which has deconz-2.5.66). I used Firefox 68 and Chromium 76 for testing. Before each test I remove the state (~/.local/share/dresden-elektronik/deCONZ) and the browser Local Storage (somewhere in developer tools: Ctrl-Shift-I), to start clean.

bjornfor, 2019-08-28:
It seems the front-end shows that error in response to this HTTP traffic:

```
-> PUT http://192.168.1.161/api/F616747C41/config/password
 with params: {"username":"delight","oldhash":"ZGVsaWdodDphZG1pbjEyMTI=","newhash":"ZGVsaWdodDphZG1pbjEyMzQ="}
<- 403 Forbidden
 with params: [{"error":{"address":"/config/password","description":"unauthorized user","type":1}}]
```

It's weird to get "unauthorized user" when I *just* logged in (and did not log out!).

Could this have something to do with the server name/IP being used? I always type http://localhost/ into the URL bar, but sometimes the app finds the IP address of my machine and presents that instead on the /pwa/login.html page. I'm asking since the PUT the above trace has the machine IP address instead of localhost. Does the auth mechanism use the servername?

Smanar, 2019-08-28:
Have you try to unlock the gateway ? It not usefull is all situation, but you can try with that.

bjornfor, 2019-08-28:
Unlock? What does that do and how do I do it?

I'm just using the Phoscon App in my browser, I have not started testing 3rd party access using "Authenticate App", if that's what you meant.

Smanar, 2019-08-28:
Yep it's that. If you don't do that you can have this error message 403 "unauthorized user".

bjornfor, 2019-08-29:
> Does the auth mechanism use the servername?

At least it seems to have special case for the localhost address: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1279#issuecomment-510738795.

bjornfor, 2019-08-29:
But using the external IP address of my machine did not fix the issue.

Smanar, 2019-08-30:
Even with "Authenticate App" ?
IDK for changing password but to reset it there is too a time limit after power On.

bjornfor, 2019-09-01:
No, I haven't tried "Authenticate App". I cannot imagine a system designed so that you'd have to give 3rd party apps access in order to change the admin password.

bjornfor, 2019-09-01:
@Smanar: Does changing the password work for you?

manup, 2019-09-01:
The error indicates that the user is currently not logged in.
Can you please try to logout and login again and then change the password?

bjornfor, 2019-09-01:
I did the following steps, with the same setup as in the first post:

[0. wipe browser cache, rm -rf ~/.local/share/dresden-elektronik, start deconz / phoscon app]
1. initial login
2. try to change password -> "Service not available. Try again later."
3. logout
4. login
5. try to change password -> "Service not available. Try again later."

I did all this on http://localhost.

Smanar, 2019-09-02:
Error 403
[
  {
    "error": {
      "address": "/config/password",
      "description": "unauthorized user",
      "type": 1
    }
  }
]

I have reproduce your message, using bad API key. Are you sure for the key ?

bjornfor, 2019-09-02:
Yes, I'm sure about the password.

Smanar, 2019-09-02:
Not the password, the "F616747C41" part.
Try a GET with http://192.168.1.161/api/F616747C41/config/ to see if you have the full config. With bad api key the result is shorter.

bjornfor, 2019-09-03:
```
$ curl http://localhost/api/F616747C41/config/
{"apiversion":"1.16.0","bridgeid":"00212EFFFF04898E","datastoreversion":"60","factorynew":false,"mac":"f4:06:69:34:af:8f","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.5.66"}
```

Note that I wipe `~/.local/share/dresden-elektronik` quite frequently, since I'm still only evaluating this software. (I cannot really use it until I've figured out these issues.) That means the key above is wrong, since (I really hope) deconz doesn't use a hardcoded key, and instead creates one randomly when initializing its state.

Seeing as the above output looks kind of "full" to me, I tried using some other random keys. And look at that, it completely ignores the key:

```
$ curl http://localhost/api/really-not-a-key/config/
{"apiversion":"1.16.0","bridgeid":"00212EFFFF04898E","datastoreversion":"60","factorynew":false,"mac":"f4:06:69:34:af:8f","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.5.66"}
```

I also tried with the external IP address of my machine (not "localhost"), same result:

```
$ curl http://192.168.43.66/api/really-not-a-key/config/
{"apiversion":"1.16.0","bridgeid":"00212EFFFF04898E","datastoreversion":"60","factorynew":false,"mac":"f4:06:69:34:af:8f","modelid":"deCONZ","name":"Phoscon-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.5.66"}
```

In fact, after setting the initial password, the phoscon app accepts _any_ password at the (re)login page. Well, the initial login page also accepts any password, but that's expected :-)

Smanar, 2019-09-03:
So, I can confirm, your api key is bad. And no, its not full ^^, on the full version you will have lot of more information, like all memorised api key.

bjornfor, 2019-09-03:
> So, I can confirm, your api key is bad.

Thanks.

> Try to make a new one, using the api.

If the app creates a bad api key (or somehow messes up the key handling along the way), why should I work around it by making another key? There seems to be an issue here and it looks easy to reproduce: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1792#issuecomment-526939693.

Smanar, 2019-09-03:
No, the app is not here to create key for you use it, take a look here https://dresden-elektronik.github.io/deconz-rest-doc/configuration/#aquireapikey

bjornfor, 2019-09-04:
Thanks for the link.

But... are we talking past each other?

> No, the app is not here to create key for you use it (phoscon create a key for its use),

Yes, that's my understanding too: when I do the initial login in the phoscon app (in my web browser), the app creates an API key for its own use (hence why I see it in the network trace -- it's not my password I see there). When I continue to use my web browser to access the phoscon app, I would expect things to just work. I don't expect having to create a key for 3rd party access, in hopes of that allowing continued use of the app which I _just_ logged on to.

Are you saying this is not a bug: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1792#issuecomment-526939693?

bjornfor, 2019-09-04:
To repeat: I'm not (yet) interested in the REST API. This issue is about the Phoscon App.

Hm, I just realized that perhaps I've mistaken what this repo is about. Is it _just_ the REST back-end and not the Phoscon App front-end? If so, where is the front-end (Phoscon App) located? (https://github.com/dresden-elektronik/phoscon-app-beta is no longer maintained, since it's no longer "beta", as the README says.)

Smanar, 2019-09-04:
Oups, sorry, my bad, as I have read API request, I thought you were using it too.

bjornfor, 2020-01-02:
The issue exists until confirmed fixed.

Mimiix, 2020-06-06:
@bjornfor Can you confirm yourself it still exist? Instead of waiting for confirmation from DE.

bjornfor, 2020-06-12:
@Mimiix: trying deconz again has been on my TODO for a while, although low priority since the first (and last) time I tried it I quickly hit two blocking auth issues, one of which is this issue. (The other is https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1788.)

I didn't respond quickly to your ping because I was hoping to actually find some time to reproduce instead of replying "yes, I'll try and reproduce when I have time".

Mimiix, 2020-06-12:
@bjornfor I'll take it ;)

I'm just trying to find out if these old issues are still actual or not. It sometimes is  hard to see for me.

I'll keep it open, if you have addition info, please provide!

Only need a pinky swear from you that you'll close it if it is resolved. You can do that for me ;)?

bjornfor, 2020-06-12:
@Mimiix: Thanks! That increases my motivation to try deconz again :-) Yes, I'll try my best to remember this issue and close it if I don't see it anymore.

Mimiix, 2020-06-12:
The bot will remind you every 21 days ;)

If you need more motivation, join the discord.

bjornfor, 2020-06-18:
Reproduced in deconz 2.05.77:

1. set initial password ("passA").
2. change password to ("passB").
3. change back to first password, get error "Service not available. Try again later."

bjornfor, 2020-06-18:
...
4. Logging out. (This might be important.)
5. Logging back in with current password ("passB").
6. Trying to change to any new password and get error "Service not available. Try again later." Goto 4, repeat.

So to change the password more than one time it seems one must reset the service (wipe all files and restore from backup).

bjornfor, 2020-07-10:
Hey maintainers. This stale bot is really annoying!

Mimiix, 2020-07-10:
For this issue it might be, but for 9/10 it is not :) I put it on backlog so it stays open.

Mimiix, 2020-07-10:
Also: I'll check with some devs, i might have to move this to the phoscon repo.

Smanar, 2020-07-10:
Someone else can reproduce it ?