dresden-elektronik/deconz-rest-plugin #1795: Build fails on ubuntu-18.04

bjornfor, 2019-08-29:
```
$ qmake && make
[...]
make -f Makefile.Release
make[1]: Entering directory '/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DGW_SW_VERSION=\"2.05.67\" -DGW_SW_DATE=1566988495 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"c6981385c63d2728808f45a9aa7dc20b9c94028a\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26330500 -DGW_MIN_R21_FW_VERSION=0x26490700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/firmware_update.o firmware_update.cpp
firmware_update.cpp: In member function 'void DeRestPluginPrivate::queryFirmwareVersion()':
firmware_update.cpp:353:25: error: 'FW_ONLY_R21_BOOTLOADER' was not declared in this scope
             fwVersion = FW_ONLY_R21_BOOTLOADER;
                         ^~~~~~~~~~~~~~~~~~~~~~
firmware_update.cpp:353:25: note: suggested alternative: 'FW_ONLY_AVR_BOOTLOADER'
             fwVersion = FW_ONLY_R21_BOOTLOADER;
                         ^~~~~~~~~~~~~~~~~~~~~~
                         FW_ONLY_AVR_BOOTLOADER
firmware_update.cpp:367:86: error: 'FW_ONLY_R21_BOOTLOADER' was not declared in this scope
         else if (((fwVersion & FW_PLATFORM_MASK) == FW_PLATFORM_R21) || fwVersion == FW_ONLY_R21_BOOTLOADER)
                                                                                      ^~~~~~~~~~~~~~~~~~~~~~
firmware_update.cpp:367:86: note: suggested alternative: 'FW_ONLY_AVR_BOOTLOADER'
         else if (((fwVersion & FW_PLATFORM_MASK) == FW_PLATFORM_R21) || fwVersion == FW_ONLY_R21_BOOTLOADER)
                                                                                      ^~~~~~~~~~~~~~~~~~~~~~
                                                                                      FW_ONLY_AVR_BOOTLOADER
firmware_update.cpp:501:81: error: 'FW_ONLY_R21_BOOTLOADER' was not declared in this scope
         if (((fwVersion & FW_PLATFORM_MASK) == FW_PLATFORM_R21) || fwVersion == FW_ONLY_R21_BOOTLOADER)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
firmware_update.cpp:501:81: note: suggested alternative: 'FW_ONLY_AVR_BOOTLOADER'
         if (((fwVersion & FW_PLATFORM_MASK) == FW_PLATFORM_R21) || fwVersion == FW_ONLY_R21_BOOTLOADER)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
                                                                                 FW_ONLY_AVR_BOOTLOADER
Makefile.Release:680: recipe for target 'release/firmware_update.o' failed
make[1]: *** [release/firmware_update.o] Error 1
make[1]: Leaving directory '/deconz-rest-plugin'
Makefile:40: recipe for target 'release' failed
make: *** [release] Error 2
```

Steps to reproduce:

1. Ubuntu 18.04 (amd64) in a VM, container or bare metal.
2. Install deconz (stable) packages according to https://phoscon.de/en/conbee2/install#ubuntu.
3. Install the deconz-dev package (and [build-essential](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1794))
4. Clone this repo (currently at commit c6981385c63d2728808f45a9aa7dc20b9c94028a)
5. Start building: ```qmake && make```

manup, 2019-08-31:
The stable version 2.05.65 is missing the FW_ONLY_R21_BOOTLOADER define.
This was meanwhile fixed in version 2.05.67 beta version.

To fix your build, please use the APT beta repository:

```
sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz \
            $(lsb_release -cs)-beta main' > \
            /etc/apt/sources.list.d/deconz.list"
```

Or as alternative upgrade the deconz-dev package manually:

    wget https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-dev-2.05.67.deb
    dpkg -i deconz-dev-2.05.67.deb

bjornfor, 2019-09-01:
Switching to beta fixed it. Thanks!

closed 2019-09-01 by bjornfor