dresden-elektronik/deconz-rest-plugin #1796: Heimann CO sensor becomes unresponsive after some time

timbru31, 2019-08-29:
I've got a Heimann CO detector (HS1CA, support should be added via #1542). After some time (~ 1 day) it becomes unavailable though. The sensor is still on (LED flashes, function test works, too) - but in order to have a value in deCONZ (2.05.67, 0x26330500) again I need to reboot it.

Unfortunately(?) I wasn't able to trigger a carbon monoxide alarm yet.

This is the **unresponsive** API response:
```
  "18": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": false
    },
    "ep": 1,
    "etag": "7628cf9359e71c5227199b76fd1e9196",
    "manufacturername": "Heiman",
    "modelid": "COSensor-EM",
    "name": "CarbonMonoxide 18",
    "state": {
      "carbonmonoxide": null,
      "lastupdated": "none",
      "lowbattery": null,
      "tampered": null
    },
    "swversion": "1.1.1",
    "type": "ZHACarbonMonoxide",
    "uniqueid": "XXXX"
  }
```

This is the **responsive** API response:
```
  "18": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "c41be9ae6a56b27ed46e792a49b3f8db",
    "manufacturername": "Heiman",
    "modelid": "COSensor-EM",
    "name": "CarbonMonoxide 18",
    "state": {
      "carbonmonoxide": null,
      "lastupdated": "none",
      "lowbattery": null,
      "tampered": null
    },
    "swversion": "1.1.1",
    "type": "ZHACarbonMonoxide",
    "uniqueid": "XXXX"
  }
```

Node info is below:
[image]

Let me know if I can assist debugging.

PeteCondliffe, 2019-09-20:
Damn :( got the same issue with this.
For me pressing the test button on the unit beings it back online.

timbru31, 2019-09-20:
Yeah unfortunately I’d like permanently mount it, so pressing the test button on a daily basis is not really possible

PeteCondliffe, 2019-09-20:
CO alarms want regular testing anyway (weekly?) and mounted somewhere near optimum position (head height) meaning pressing the button should be no issue with it permanently mounted.

I definitely agree daily is a little extreme and going unavailable in HA is going to make it impossible to see if its online and monitor battery. But because it comes back up in a on state when the test button is pressed any automations looking for the alarm being activated will still work.

The main point of my last post was feedback that you are not alone with this issue and a little more information that the test button brings it back online for anyone kind enough to look into the underlying issue for us.

timbru31, 2020-01-19:
Issue still exists and renders the device useless. Therefore, not stale.

timbru31, 2020-03-03:
Either with the change https://github.com/dresden-elektronik/deconz-rest-plugin/commit/33d8a8b349c9f4967e8b94ed2657e038406317c8 or the new test firmware (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-590511276) the state does not become unavailable anymore and stays off. Can you confirm this behavior change @PeteCondliffe

closed 2020-07-03 by stale[bot]

rnsc, 2022-04-17:
This is a recurrent issue for me too.
I'm considering buying a device from a different brand to alleviate this.