dresden-elektronik/deconz-rest-plugin #1797: [Feature] "advanced" zigbee API

Equidamoid, 2019-08-29:
Just as an idea, how about adding more low-level API closer to the actual zigbee messages?

For example I'd like to send `OnWithTimedOff` message so I'll be sure the device will turn off even in case of some zigbee connectivity problem or even a nicely timed cat bite to some wires.

Kane610, 2019-08-30:
Being able to send this similar to bri_inc would be nice. Why keep timers in automations when the light can keep it self

ebaauw, 2019-08-30:
I appreciate the call for a “raw” API, closer to ZigBee, see #1688.

Last January, I already implemented _On with Timed Off_ through `state.ontime`, see https://github.com/dresden-elektronik/deconz-rest-plugin/commit/6ed48e272450beff7555a428ab5467d4da11b7a8.

Equidamoid, 2019-08-30:
@ebaauw 
Oh, it's nice to hear that such idea also comes "from inside"!
Thanks for the `ontime` hint, that will do in the meantime

closed 2020-01-04 by stale[bot]