dresden-elektronik/deconz-rest-plugin #1798: Conbee II shows up as Raspee in Phoscon

andordavoti, 2019-08-30:
I have used Deconz with the Conbee II for a couple of months now. Currently switching from Xiaomi Aqara Hub. But I after updating the firmware through the Phsocon GUI I had some pretty chaotic issues.

After updating it just didn’t want to work anymore. Just died. So I unplugged it from my Intel NUC and plugged it into my Windows laptop to do an update manually using this guide:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows

I have hassio installed in docker in Ubuntu Server 18.04, but I didn’t understand the docker guide so I did the update on Windows changing the IP address of the NUC to something else and the IP of my laptop to the previous IP of the NUC.

After doing the update, I restored a previous firmware to get back all my paired devices. Then my Conbee II now shows up as a Raspee in Phoscon. The issue now is that in HA the event listener randomly stops, don’t know if it’s a problem in HA or deconz (probably my conbee II that shows up as a raspee, but I don’t know). I also can’t update it now, because it thinks it’s a different device then it actually is. Does anyone have any advice for what I should do? Don’t really want to have to buy another Conbee II…

Smanar, 2019-08-30:
What is the result of the command "GCFFlasher_internal -l"
Have you tried the update by the command line ?

manup, 2019-08-31:
>Does anyone have any advice for what I should do? Don’t really want to have to buy another Conbee II…

The following FAQ  entry hopefully has some hints to resolve the issue.

https://phoscon.de/en/support#conbee2-connection-issues

In any way, you don't need to buy another one, firmware can be updated (it's not possible to brick the device) and messy configurations can be fixed.

The most important thing usually is to have a backup created in the Phoscon App, it's not required but prevents the worst case of repairing all devices.

xbmcnut, 2019-09-01:
@andordavoti From memory with the deCONZ add-on <3.1, the Conbee II showed up as a Raspbee on my Hassio NUC install also. This is fixed in 3.3 and it now shows up as a USB stick. Are you running firmware 26490700 on the Conbee? Latest 3.3 add-on update should also have you running deCONZ 2.05.67 / 4/22/2019

andordavoti, 2019-09-01:
@xbmcnut I'm running v3.3 of the addon and firmware 26490700 on Conbee II (which shows up as raspee in Phoscon). So the update has sadly not fixed anything for me.
[image]

xbmcnut, 2019-09-01:
Maybe try changing the USB path to the specific device ID one as I have?
[image]

Munsio, 2019-09-02:
I have the same problem.
Running conbee 2 stick on intel nuc with home-assistant.
The addon mentioned is this one: 
https://github.com/home-assistant/hassio-addons/tree/master/deconz

I configured the addon to use the serial by-id path which shows correctly that it is an conbee 2 stick
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DEXXXXX

But the Phoscon App showing a RaspBee with following version + firmware
Version 2.05.67 / 4/22/2019
Firmware 264A0700

pvizeli, 2019-09-05:
Is fixed with latest version of Add-on

Munsio, 2019-09-09:
Can confirm the device is now correctly detected as conbee 2 usb stick.

pvizeli, 2019-09-09:
issue can be closed

manup, 2019-09-09:
Cool, thanks for verify

closed 2019-09-09 by manup

sebastien-k, 2020-01-26:
I actually have the same issue; my Conbee II suddenly became a RaspBee last night, I can't control anything anymore. Does anyone found the solution have his Conbee II detected as a Conbee II ?

I'm using on a :
- HassOS 3.8
- Home Assistant 0.104.3
- deCONZ 5.1
- Hass.io 195

(on a Raspberry Pi 3 B+) 

Thanks for your help!