dresden-elektronik/deconz-rest-plugin #1799: Wrong brightness is send over the websocket

Xento, 2019-08-30:
Version 2.05.67 / 22.4.2019
Firmware 26330500 

I'm using phoscon in the version specified above.
When I send a brightness to an osram lightify gardenpole I get multiples messages back from the websocket. I'm using nodered for this.

I want to set the Brightness to about 3%.
It is 8 in the range from 0 to 255.
As you can see the command is accepted and executed but during the change it seems that values during the transition are send.
I have a slider attached via mqtt and this looks really ugly when it moves to nearly 100% and than again to the correct value.
```
30.8.2019, 13:54:15 node: 8ddc8cc1.2315c
/Deconz/lights/1/set : msg.payload : string[79]
"[{"success":{"/lights/1/state/on":true}},{"success":{"/lights/1/state/bri":8}}]"
30.8.2019, 13:54:16 node: 40b9dbe1.7d0664
msg.payload.state.bri : number
8
30.8.2019, 13:54:17 node: 40b9dbe1.7d0664
msg.payload.state.bri : number
249
30.8.2019, 13:54:17 node: 40b9dbe1.7d0664
msg.payload.state.bri : number
238
30.8.2019, 13:54:23 node: 40b9dbe1.7d0664
msg.payload.state.bri : number
8
```

ebaauw, 2019-08-30:
The first message is issued when deCONZ updates it's cache because of the command to set the _target_ brightness; the other messages are caused by the bulb sending attribute reports for the _current_ brightness.

You should have exposed two sliders: one write-only for the target brightness, and one read-only for the actual brightness.  This is a fundamental design flaw in the REST API and in all home automation systems I've seen: there's no distinction between current brightness and target brightness.  Changes are assumed to happen instantaneously, not taking into account the transition time.  For a re-design of the API, I proposed to use `action.bri` for the target brightness and `state.bri` for the current brightness.

To prevent this behaviour in your current setup using only one slider, either specify a `transitiontime` of 0 when setting `bri` (which leads to ugly changes) or change the attribute reporting for the bulb in the GUI, setting _Reportable Change_ to 0.

Xento, 2019-08-30:
In openhab you normaly only have one slider for current value and target value.
```
Type dimmer : dimmer "Dimmer"           [ stateTopic="/Deconz/lights/1", transformationPattern="JSONPATH:$.bri",
                                                      commandTopic="/Deconz/lights/1/set", formatBeforePublish="{\"bri\":%s}" ]
```

I now worked around by ignoring change events during the next 2 seconds from this device.
I noticed when I change the brightnes with Phoscon the transition events are sent but the final value is send several seconds later.
For example when I move the slider in Phoscon from 100 to 30.
Updates are send going down from 100 to 50 and several seconds later 30.

alialrahma, 2019-09-02:
Thanks for mentioning this issue and a fix. I have a similar issue in my Homebridge setup. How am I able to publish these changes to my setup?

Xento, 2019-09-25:
This is my NodeRed Flow: https://pastebin.com/Qr6ewnRc
Don't know how you connected homebridge.
Maybe you could create two things. One that sends command and one that receives the updates.
Then create a rule that sets a variable when you change the  sending thing and resets it after 2 seconds. Than a second rule that when the receiving thing is updated that checks if the variable is set and when not update the state of the sending thing.

closed 2019-09-25 by Xento